adafruit/adafruit-beaglebone-io-python #80: v4.1.x + overlay migration notes...

RobertCNelson, 2015-05-07:
linux repo: https://github.com/beagleboard/linux/tree/4.1
overlay repo: https://github.com/RobertCNelson/bb.org-overlays

slots:
```
/sys/devices/platform/bone_capemgr/slots
```

enable/disable cmdline:
```
beaglebone-capemgr.enable_partno=
beaglebone-capemgr.disable_partno=
```

uart:
```
target = <&uart(X)>; = target = <&uart(X-1)>;
```

tdicola, 2015-05-07:
Oh cool, is that the new mainline support for the bone capemanager with dynamic device tree overlays?  Thanks for the heads up as we'll likely need to make some changes to support it.

Do you have a BBB Debian version with the 4.1 kernel support yet to start testing it out with? :)

RobertCNelson, 2015-05-07:
Hi Tony,

Trying to get a *.deb out, but we need to upgrade gcc on:
http://builds.beagleboard.org/builders/build-kernel

I'll post a test *.deb later tonight.

But, v4.1.0-rc3 this weekend will default to it and be in the repo.

btw, feel free to fork the overlay repo and add your guy's *.dts's i'm going to be having that repo installed by default on the bb.org images..

Regards,

RobertCNelson, 2015-05-07:
Correct, this is the "new" capemgr panto has been working on getting mainline:

https://github.com/pantoniou/linux-beagle-track-mainline/commits/bbb-overlays

Regards,

tdicola, 2015-05-07:
Awesome, thanks for all the work as always to make the latest releases available!  Just curious too, is there any timeline for when you might switch to this 4.1-based kernel as the 'official' Debian release for BBB users?  Want to make sure we can get the fixes in, etc. for it.

RobertCNelson, 2015-05-07:
I'd say, just as soon as we test/confrm all these capes:

https://github.com/beagleboard/devicetree-source/tree/master/arch/arm/boot/dts

I've got a lot of them, for some I'll be hitting up ebay!  But I'm trying to get the bb.org community and all downstream v3.8.x users involved.. That way when we switch, everything works 100%!  The lcd's are still going to be a pain thou.

Regards,

ivanchou0826, 2015-06-22:
Adafruit-BBIO==0.0.30 support the latest kernel of 4.1.0-rc7-bone7

when I try to use ADC utility, the following error popups, but GPIO utility of the library is okay.
anything wrong?

>>> import Adafruit_BBIO.ADC as ADC
>>> ADC.setup()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper

I update the latest kernel, and install bb.org-overlays. I enable ADC in uEnv.txt by

optargs=dtb=am335x-boneblack-overlay.dtb
optargs=bone_capemgr.enable_partno=BB-ADC bone_capemgr.disable_partno==BB-BONELT-HDMI,BB-BONELT-HDMIN

My kernel message about capemgr is the following. Related ADC sysfs is available at 
/sys/bus/iio/devices/iio\:device0/
I cannot get adc value by reading the path, but using adafruit-beaglebone-io-python does not work~~

root@beaglebone:~/python# dmesg | grep capemgr
[    0.000000] Kernel command line: console=tty0 console=ttyO0,115200n8 bone_capemgr.enable_partno=BB-ADC bone_capemgr.disable_partno==BB-BONELT-HDMI,BB-BONELT-HDMIN root=UUID=9e07e85d-795b-4480-ad05-8258e4a23995 ro rootfstype=ext4 rootwait fixrtc quiet init=/lib/systemd/systemd
[    3.237676] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,00C0,3214BBBK1346'
[    3.237696] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black
[    3.292912] bone_capemgr bone_capemgr: slot #0: No cape found
[    3.352912] bone_capemgr bone_capemgr: slot #1: No cape found
[    3.412909] bone_capemgr bone_capemgr: slot #2: No cape found
[    3.472908] bone_capemgr bone_capemgr: slot #3: No cape found
[    3.478930] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-ADC' VER 'N/A' PR '0'
[    3.478941] bone_capemgr bone_capemgr: slot #4: override
[    3.478954] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[    3.478968] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-ADC'
[    3.479220] bone_capemgr bone_capemgr: initialized OK.
[    3.495394] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-ADC-00A0.dtbo' loaded; overlay id #0

shaunmulligan, 2015-08-05:
Is there any update on this? I would be very interested in this and willing to help where I can, even if its just to help testing.

mgaggero, 2015-09-08:
I had the same issue. The problem seems to lie in the way Adafruit-BBIO library tries to access the adc devices using the old 3.8.x kernels device tree. Few months ago I've submitted a pull request for a patched version of Adafruit-BBIO library that works with the new >=3.14 device tree structure, but so far I've not received any comments. It's just a starting point and it needs further development.

10man, 2015-09-26:
Hello,
I'm having a tough time getting the capemgr to work on the BBB using the 4.1.8-bone16 kernel. Everything is good out of the box on 3.8.13-bone74, but I was hoping to get an image setup with the later kernel. I'm using Debian 8.2 console (9-20-15) and I require the UARTs. The "update-kernel.sh" script kicks back to the 3x kernel. if I then update the kernel manually the capemgr remains working. I'm sure there is something I'm not doing that is preventing me from using the new kernel with the cape manager.

grizmio, 2015-11-17:
Maybe it's too late, it has been months since this thread started, but it could help someone like me when i came here the first time.
I'm using https://github.com/adafruit/adafruit-beaglebone-io-python libraries with ubuntu xenial with kernel 4.1.12-ti-r29
and to get ADC working i used this uglies patches:
https://gist.github.com/grizmio/bc752dd070652f21266b#file-gistfile1-txt

Then:
# python -c "import Adafruit_BBIO.ADC as ADC;ADC.setup();value=ADC.read_raw('P9_40');print(value)"
3070.0
# python -c "import Adafruit_BBIO.ADC as ADC;ADC.setup();value=ADC.read('P9_40');print(value)"
1.73000001907

Seems to work :-)

iandow, 2015-12-02:
I'm also having trouble using Adafruit_BBIO on a beagleboneblack running Ubuntu 14.04.3 and kernel 4.1.12.  That patch you referenced looks helpful, but does it require that I recompile my kernel?

Seem strange that I would need to patch and recompile my kernel just to get basic I/O to work.  Might be easier just to revert to default Angstrom, since I don't have a kernel development environment setup.

grizmio, 2015-12-02:
@iandow 
There's no need to recompile the kernel, you could just download this repo, apply the changes and then run python setup.py install
Newer kernel versions use different device file paths, so i just fixed those paths, but this just fix ADC support maybe even broking PWM, so PWM should need copy/paste some patch code to get it works too.

mattmaherpeterson, 2015-12-09:
How do apply this patch?  Never done that before?

mattmp, 2015-12-09:
How do I apply this patch?

claudiofantacci, 2015-12-10:
Hello everyone,

I'm migrating a 3.8 overlay UART + PPS (pulse per second) to kernel 4.1.*-ti + RT, please find attached the dts file of it.

I have edited the **target=<&uart2>** to be **target=<&uart1>**, but I'm not sure whether or not 
the *fragment@2* describing the PPS need to be changed as well.
In particular, as far as the PPS uses *GPIO1_16* it seems that **gpios = <&gpio2 16 0>** should also become **gpios = <&gpio1 16 0>**. Is it correct?

What I get after compiling the dts and loading it is the following from dmesg:
> 48024000.serial: ttyS2 at MMIO 0x48024000 (irq = 189, base_baud = 3000000) is a 8250
> pps pps0: new PPS source ocp:pps.-1
> pps pps0: Registered IRQ 106 as PPS source

Problem is that the serial should be ttyS1!
If, from the same dts, I remove the parts of the pps the correct serial is loaded.
Could you please provide me any insight?
Thanks!

[UART-PPS.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/57940/UART-PPS.txt)

grizmio, 2015-12-11:
Right now i dont have my laptop but if i remeber correctly its something líke
git apply patch_file
Or you can get the repo from @mgaggero

spmaniato, 2016-03-19:
> How do I apply this patch?

Concise instructions (starting from scratch) for people landing here from other issues / webpages:
```
git clone https://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
wget https://gist.githubusercontent.com/grizmio/bc752dd070652f21266b/raw/336b5ab5ad14ef28341edfeff6cbf20d959d2f70/gistfile1.txt
git apply gistfile1.txt
sudo python setup.py install
cd ..
sudo rm -rf adafruit-beaglebone-io-python # Unless you want to keep it around ofc
```

julianhille, 2016-04-08:
tried it with and without the packes on neweset 8.3 image from BB.org
unable to get the PWM working on my BBB. after patching the errors were gone but still not possible to use the PWMs.

jstoner, 2016-04-27:
I can say the same. I also did have to monkey with it a bit--Python couldn't find the new egg, so I unzipped it and copied the code to my virtualenv's site-packages. 

But yeah, no PWM in Linux 4.1. I'll be switching back to 3.8 til someone gets it to work.

keerthanamanivannan, 2016-06-21:
Hi, i'm trying to apply the patch and get the PWM working. But I get this error:

`error: patch failed: source/c_adc.c:30
error: source/c_adc.c: patch does not apply
error: patch failed: source/common.c:361
error: source/common.c: patch does not apply
error: patch failed: source/common.h:39
error: source/common.h: patch does not apply`

 Could you guys resolve it? How to get it working?

pdp7, 2016-06-21:
@keerthanamanivannan I am working on getting PWM in Adafruit_BBIO to work with Linux kernel 4.1+.  I'm keeping track of my progress here: #103

pdp7, 2016-08-20:
@keerthanamanivannan PWM support for Linux kernel 4.1+ has now been merged from #108

pdp7, 2016-08-20:
@RobertCNelson I believe with #96 and #108 now merged, this library works OK with Linux kernel 4.1+.

keerthanamanivannan, 2016-08-31:
@pdp7 awesome. thanks!

zerrcol, 2016-09-19:
hi @pdp7 ,

I am working with Linux beaglebone 4.4.21-bone13 . And I have went through the post and several sites. Basically for PWM. But After applying all patches. (and sources from the merged libraries ) I am still not able to load pwm's. 
When I do settings in cape manager , then i do sync, after when i reboot. 
I always getting warning as 
'failed to load cape manager' . So for that My PWM's are not working.

pdp7, 2016-09-20:
@zerrcol Could you please paste the commands and/or code that you are using?

Also, please post output of:
```
cat /sys/devices/platform/bone_capemgr/slots

cat /boot/uEnv.txt

cat /etc/debian_version
```

zerrcol, 2016-09-20:
Hi @pdp7 , 
Thanks for replay. 
Please check below outputs of commands you suggested. 

root@beaglebone:/lib/firmware# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   3 Override Board Name,00A0,Override Manuf,BB-UART4
 5: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-UART5
 6: P-O-L-   1 Override Board Name,00A0,Override Manuf,BB-UART2
 7: P-O-L-   2 Override Board Name,00A0,Override Manuf,am33xx_pwm

------------------------------------------------------------------------------------------------------
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.21-bone13
#uuid=
dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
#dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

cmdline=quiet cape_universal=enable
#coherent_pool=1M quiet cape_universal=enable

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M quiet cape_universal=enable video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
cape_enable=capemgr.enable_partno=BB-UART4,BB-UART5,BB-UART2,am33xx_pwm,bone_pwm_P8_13,bone_pwm_P9_28,bone_pwm_P9_42

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
cape_enable=bone_capemgr.enable_partno=BB-UART4,BB-UART5,BB-UART2,am33xx_pwm,bone_pwm_P8_13,bone_pwm_P9_28,bone_pwm_P9_42

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
-------------------------------------------------------------------------------------------------
cat /etc/debian_version
8.4

--------------------------
I have managed to get UART working ( you can suggest the correct way if I not did properly). 
But I need to bring up all peripherals. including PWM

pdp7, 2016-09-20:
@zerrcol I think you'll need to disable `cape_universal` so that it doesn't conflict.

In `/boot/uEnv.txt`, change `cmdline` from:
```
cmdline=quiet cape_universal=enable
```
to 
```
cmdline=quiet
```

You should then be able to load the device tree overlays for PWM.

Alternatively, you could keep `cape_universal` enabled and use the `config-pin` utility to mux the pwm pins.  Check out the [beaglebone-universal-io README](https://github.com/cdsteinkuehler/beaglebone-universal-io) by @cdsteinkuehler

zerrcol, 2016-09-20:
hi @pdp7 ,

I have did all the changes described on that link only. 
Also I have edited the uEnv.txt as you said. and rebooting the board.
still getting error while applying patches from given link.

error is " 
[294.834495]bone_capemgr bone+capemgr :slot #19 : failed toi resilve tree

zerrcol, 2016-09-20:
also @pdp7 
attaching logs of uEnv.txt

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.21-bone13
#uuid=
dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
#dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

cmdline=quiet
# cape_universal=enable
#coherent_pool=1M quiet cape_universal=enable

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M quiet cape_universal=enable video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
cape_enable=capemgr.enable_partno=BB-UART4,BB-UART5,BB-UART2,am33xx_pwm,bone_pwm_P8_13,bone_pwm_P9_28,bone_pwm_P9_42

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
cape_enable=bone_capemgr.enable_partno=BB-UART4,BB-UART5,BB-UART2,am33xx_pwm,bone_pwm_P8_13,bone_pwm_P9_28,bone_pwm_P9_42

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

zerrcol, 2016-09-21:
Hi @pdp7 ,
Have got any solutions for this ?

pdp7, 2016-09-22:
>  still getting error while applying patches from given link

@zerrcol I'd like to get a better understanding exactly of what you are typing and what the output is.  

Could you post the entire transcript of the terminal or make screenshots of the terminal?

Is there Python code that you are running before that error occurs?

zerrcol, 2016-09-22:
Hi @pdp7 ,
Thanks for the replay. I am following all the steps that are mentioned over the link.
I am using kernel version 4.4.x , So as per the steps I am applying them on respective directory. 
Not doing /running anything else.
I will record the terminal activity and ll post.

pdp7, 2016-09-22:
@zerrcol are you trying to apply the patch that @grizmio created?  

PWM support for Linux kernel 4.1+ has now been merged from #108.  Have you installed the latest version of this repo on your BeagleBone?

zerrcol, 2016-09-22:
Yes @pdp7 ,
I am doing that only.

pdp7, 2016-09-22:
@zerrcol Have you installed the latest version of `Adafruit_BBIO` from `master` branch of this repo?

No patches should be necessary anymore.  Linux 4.1+ support has been merged into `master`:
* PWM support merged from #108 by @MatthewWest
* ADC support merged from #96 by @PeteLawler

zerrcol, 2016-09-23:
hi @pdp7 ,

I have downloaded adafruit bbio only from master. and directly I am using that only. No separate patches That I am using.

pdp7, 2016-09-23:
@zerrcol Ah ok, that is good to know.

You previously wrote that you were trying to load PWM and saw this error:
```
[294.834495]bone_capemgr bone+capemgr :slot #19 : failed to resolve tree
```

Does that occur during bootup?

I'd suggest commenting out the  `cape_enable=` line in `/boot/uEnv.txt`:
```
#cape_enable=bone_capemgr.enable_partno=BB-UART4,BB-UART5,BB-UART2,am33xx_pwm,bone_pwm_P8_13,bone_pwm_P9_28,bone_pwm_P9_42
```

Then try loading those capes individually from the command line.  Please copy and paste the contents of your terminal after doing that, including any errors from `dmesg`.

zerrcol, 2016-09-23:
@pdp7 
Yes That occurred during boot up
And If i comment out that line in uEnv.txt. that uart and other peripherals ,I think will not work good.??

pdp7, 2016-09-23:
@zerrcol The reason I think it would be good to comment out `cape_enable` is so that the results can be observed of loading each cape individually from the command line.

jcdammeyer, 2016-10-17:
Running uname -a
Linux beaglebone 4.1.4-ti-r8 #1 SMP PREEMPT Thu Aug 6 21:01:16 UTC 2015 armv7l GNU/Linux
$SLOTS is

 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 7: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-W1-P8.11

where my BB-W1-P8.11 is a dts for DS1822/DS18B20 one wire temperature sensors.
I have python program reporting the temperature.  In order to load the BB-W1-P8.11 dtbo I had to disable cape_universaln in the uEnv.txt with

##Enable the One Wire interface on P8-11
cape_disable=bone_capemgr.disable_partno=cape-universaln
cape_enable=bone_capemgr.enable_partno=BB-W1-P8.11

The cape_enable doesn't work.  I still have to echo the dtbo out to the slots file but without the disable there's a pin conflict from cape_universaln.

Trying python script with 
import Adafruit_BBIO.GPIO as bbb 

But I'm not able to write out to a GPIO pin.  The Python code does the following:

bbb.setup("P9_12",bbb.OUT)   # Set GPIO P9_12 control to output
bbb.output("P9_12", bbb.LOW) 
bbb.output("P9_12", bbb.HIGH)

Is this still the approach from the 3.8 kernel?  
I can control the output by doing 
echo 60 > export
which creates the gpio60 folder and inside there I can set the direction to output and then set the value to 1 or 0.  The scope on P9_12 verifies that the commands work.

What am I missing in the python program?  All the books I have and most web pages don't appear to address the newer device tree approach.

RobertCNelson, 2016-10-17:
@jcdammeyer compare your w1 dts file with this version:

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts

Regards,

jcdammeyer, 2016-10-17:
Thanks Robert,
Mine and yours are virtually identical with the exception that yours uses include files to hide the absolute constants.  So first thing I had to do is find them.  Might be worthwhile putting the entire path 
 
/opt/source/bb.org-overlays/include/dt-bindings   
 
into a comment in front of the three include statements.
 
root@beaglebone:/# find -name dt-bindings
./usr/src/linux-headers-4.1.4-ti-r8/include/dt-bindings
./opt/source/bb.org-overlays/include/dt-bindings
./opt/source/dtb-4.1-ti/include/dt-bindings
 
The second one on my BBB was the location of your include files.
 
Neither mine, which is derived from TI, nor yours, have enough comments to explain why the fragment@1 uses target =<&ocp>;  verses target_path=”/”;
 
Not sure why mine uses onewire@0 in fragment@1 while yours uses just onewire.
 
And GPIO_ACTIVE_HIGH is certainly more descriptive than ‘0’ but one has to find the include file to learn what it is.
 
My ultimate goal is to embed this information into the EEROM of a cape along with several other GPIO definitions.  Having never done that it’s a new area for me.  I suspect I have to create one .dts file that has all of the definitions in it.  So if I want to keep using “P8.11” for my One Wire but want to add a LED drive for Cape Active to “P9.12” I imagine it’s just added to the fragment 0 and 1 structures.
 
Thanks for the feedback.
John

 

https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-W1-P9.12-00A0.dts

jcdammeyer, 2016-10-17:
My dts file was stripped in the reply.  I've added a .txt to it.    Still not sure why the cape_enable in uEnv.txt doesn't work nor can I find out in which script cape-universaln is loaded.
[BB-W1-P8.11.dts.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/534249/BB-W1-P8.11.dts.txt)

Also, interesting point.  Sign into Cloud9 and run this simple Python example and my scope shows the output toggling.  Very peculiar.

import Adafruit_BBIO.GPIO as GPIO
import time
GPIO.setup("P9_12", GPIO.OUT)
while True:
    GPIO.output("P9_12", GPIO.HIGH)
    time.sleep(0.5)
    GPIO.output("P9_12", GPIO.LOW)
    time.sleep(0.5)

And my Python tcpServer.py running in the background is still serving up temperatures from the DS1822 One-Wire device to an Android Tablet running a Client app written in Embarcadero Delphi.

[tcpServer.py.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/534265/tcpServer.py.txt)

pdp7, 2016-10-17:
@jcdammeyer could you please open up a new issue?  That will make it easier to keep track of the specific problem that you are experiencing.

closed 2016-10-19 by pdp7