adafruit/adafruit-beaglebone-io-python #82: The method Adafruit_I2C.reverseByteOrder() in Adafruit_I2C.py is broken

phrogger, 2015-06-20:
The method Adafruit_I2C.reverseByteOrder() in Adafruit_I2C.py does not work.

It tries to automatically detect the presence of a 16 bit or 32 bit value
by detecting the first byte containing non-zero bits.  This fails whenever
there are valid leading bytes containing zeros.

------------------------------------ CLIP --------------------------------------
#!/usr/bin/python

## Demonstration of bug in Adafruit_I2C.reverseByteOrder()

from Adafruit_I2C import *

## Initialize the class; address and bus number, must be that of a valid I2C device
## I am using an Adafruit MCP9808 temp sensor on bus 2.
interface = Adafruit_I2C(0x18 , busnum=2)

for i in range( 0x03FD, 0x0404):
    y = ((i << 8 & 0xFF00) | (i >> 8 & 0X00FF)) # Reverse 16 bit byte order
    z = interface.reverseByteOrder(y)
    print 'Value = %04X, Input = %04X, Result = %04X' % (i, y, z)

---------------------------------- END CLIP -------------------------------------

## Resulting demonstration
## Result should equal Value.  
## Input is the byte reversal of Value, and is correct by inspection

root@BBB1:~# python bugdemo.py
Value = 03FD, Input = FD03, Result = 03FD
Value = 03FE, Input = FE03, Result = 03FE
Value = 03FF, Input = FF03, Result = 03FF
Value = 0400, Input = 0004, Result = 0004    # <<<<<<<<<< FAIL
Value = 0401, Input = 0104, Result = 0401
Value = 0402, Input = 0204, Result = 0402
Value = 0403, Input = 0304, Result = 0403

--------------------------------------------------------------------------------

Suggested fix:
I think it is impossible to make Adafruit_I2C.reverseByteOrder() work as written.
It is trying to determine the size of a value by detecting the presence of "ones" 
to mark the first valid byte. 
This fails when a leading valid byte is 0x00.

I suggest that Adafruit_I2C.reverseByteOrder() be replaced by two separate methods:

    def reverseByteOrder16(self, data):
        "Reverses the byte order of a 16-bit value"
        val = ((data << 8 & 0xFF00) | (data >> 8 & 0X00FF))
        return val

    def reverseByteOrder32(self, data):
        "Reverses the byte order of a 32-bit value"
        val = ((data << 24 & 0xFF000000) | (data << 8 & 0xFF0000) | (data >> 8 & 0xFF00) | (data >> 24 & 0xFF))
        return val

==

phrogger, 2015-06-20:
Didn't mean to "shout".  
The git markdown is interpreting the Python comments as headings, or something.
--- Graham

pdp7, 2016-08-20:
@phrogger is this still an issue?

phrogger, 2016-08-20:
Yes.

It also appears in multiple Adafruit python modules.

--- Graham

==

pdp7, 2016-08-21:
@phrogger Thanks for the update.  Would you be able to create a Pull Request with the fix you suggested?  

Thanks,
Drew

pdp7, 2016-08-21:
@phrogger I think this is same issue, right? https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/issues/121

phrogger, 2016-08-21:
> @phrogger <https://github.com/phrogger> I think this is same issue,
> right? adafruit/Adafruit-Raspberry-Pi-Python-Code#121

Yes. Same code, same issue.

I think I have seen it in other places too.

I can generate a pull request for those two classes, but I have no idea
where all the higher level calling methods are that would break when I
split that piece of code into two separate methods, or re-wrote it as a
single method with an additional parameter that tells the method what size
data it is dealing with.

So yes if you want a pull request for those two classes, but I don't have
time to scrub Adafruit's code base for everything the change will break
above it.

--- Graham

==

pdp7, 2016-08-21:
@phrogger Thanks for explaining.

@jwcooper did comment in [adafruit-beaglebone-io-python/Adafruit_I2C.py](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/Adafruit_I2C.py#L8):

> Adafruit_I2C.py is essentially a fork of the Adafruit Raspberry Pi I2C module. Any pull requests for this module should be directed to the following, and I can pull them. I'd rather not deviate from the original: https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/tree/master/Adafruit_I2C

However, it looks `Adafruit_I2C` has now moved to `Adafruit_GPIO` by @tdicola:
[Adafruit_Python_GPIO/Adafruit_GPIO/](https://github.com/adafruit/Adafruit_Python_GPIO/)

`reverseByteOrder()` is located in [Adafruit_GPIO/I2C.py](https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/I2C.py#L28)

I've create a new issue in `Adafruit_Python_GPIO`:
https://github.com/adafruit/Adafruit_Python_GPIO/issues/48

pdp7, 2016-10-19:
Closing this issue.  See  adafruit/Adafruit_Python_GPIO#48 for updates.

closed 2016-10-19 by pdp7