adafruit/adafruit-beaglebone-io-python #84: gpio not updated

tiblop, 2015-11-26:
Hello, I've been trying to use your library on my beaglebone , it works perfectly on almost all GPIO pins, However I have problem with P8_40 and P8_44, P8_32 and P8_42 . setting them up to high or low doesn't affect them at all. I have the last version of beaglebone debian image. I don't know how I can fix this?!
Also you say we can try to control witht the names of the pin instead of the key (P8_40 ... ) but I don't know where to find the names... The only name_key map I have is this one and it doesn't fit with your example so I think it might not be a good map: 
http://elinux.org/images/9/91/4.5_serial_UARTs.PNG
Thank you for your help

silver2row, 2015-11-26:
Hello...

You can try to go to adafruit.com and look up BBIO in their search tab. This will be a very good source on what can and cannot be done while using their system. 

Sincerely,

Seth

P.S. If you cannot find the correct info, maybe we can help each other and I can try to solve your questions then.

pdp7, 2016-08-20:
@tiblop is this still an issue?

closed 2016-10-19 by pdp7