adafruit/adafruit-beaglebone-io-python #85: Python.h in source dir

peidong, 2015-11-27:
Hello, I find it no "Python.h" in [common.c](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L31)

cnobile2012, 2015-11-28:
Mr. Chen,

This is an error you get if you do not have the the Python development
libraries installed on your machine. I don't know what platform you are
using, but the package manager on your system should provide a way to find
the correct package and install it. Then just install and compile the
package again and it should work.

~Carl

SimplicityGuy, 2016-05-05:
To clarify, you need to do the following:
```
sudo apt-get install -yy build-essential python-dev
```

pdp7, 2016-08-20:
@peidong are you still having an issue?

closed 2016-10-19 by pdp7