adafruit/adafruit-beaglebone-io-python #86: Inconsistent naming of Adafruit_BBIO

Mausy5043, 2015-12-08:
I'm using a BeagleBone Black with pre-installed Debian wheezy.
The problem I had was that in Python I could not import Adafruit_BBIO:
```
$ python
Python 2.7.3 (default, Mar 14 2014, 17:55:54)
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.GPIO as GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named Adafruit_BBIO.GPIO
>>> import Adafruit-BBIO.GPIO as GPIO
  File "<stdin>", line 1
    import Adafruit-BBIO.GPIO as GPIO
                   ^
SyntaxError: invalid syntax
```
But `pip` told me it was there:
```
$ sudo pip freeze
Adafruit-BBIO==0.0.30
PyBBIO==0.10
argparse==1.2.1
distribute==0.7.3
gyp==0.1
numpy==1.6.2
pyserial==2.7
requests==2.8.1
serbus==1.0.4
smbus==1.1
urwid==1.0.1
wicd==1.7.2.4
wsgiref==0.1.2
```
**note the minus-sign in the module name i.s.o. an underscore**

So, I uninstalled the module...
```
$ sudo pip uninstall Adafruit-BBIO
Uninstalling Adafruit-BBIO:
Proceed (y/n)? y
  Successfully uninstalled Adafruit-BBIO
```
... and had to `rm -rf /usr/local/lib/python2.7/dist-packages/Adafruit_BBIO` and `sudo rm -rf /usr/local/lib/python2.7/dist-packages/Adafruit_BBIO` to get rid of it.

Then I re-installed the module:
```
$ sudo pip  install --upgrade Adafruit_BBIO
Downloading/unpacking Adafruit-BBIO
  Downloading Adafruit_BBIO-0.0.30.tar.gz
  Running setup.py egg_info for package Adafruit-BBIO

Installing collected packages: Adafruit-BBIO
  Running setup.py install for Adafruit-BBIO
    building 'Adafruit_BBIO.GPIO' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv7l-2.7/source/py_gpio.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/event_gpio.c -o build/temp.linux-armv7l-2.7/source/event_gpio.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/py_gpio.o build/temp.linux-armv7l-2.7/source/event_gpio.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/GPIO.so
    building 'Adafruit_BBIO.PWM' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_pwm.c -o build/temp.linux-armv7l-2.7/source/py_pwm.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/c_pwm.c -o build/temp.linux-armv7l-2.7/source/c_pwm.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/py_pwm.o build/temp.linux-armv7l-2.7/source/c_pwm.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/PWM.so
    building 'Adafruit_BBIO.ADC' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_adc.c -o build/temp.linux-armv7l-2.7/source/py_adc.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/c_adc.c -o build/temp.linux-armv7l-2.7/source/c_adc.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/py_adc.o build/temp.linux-armv7l-2.7/source/c_adc.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/ADC.so
    building 'Adafruit_BBIO.SPI' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/spimodule.c -o build/temp.linux-armv7l-2.7/source/spimodule.o -Wno-format-security
    source/spimodule.c:39:0: warning: "PyLong_Check" redefined [enabled by default]
    /usr/include/python2.7/longobject.h:14:0: note: this is the location of the previous definition
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/spimodule.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/SPI.so
    building 'Adafruit_BBIO.UART' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_uart.c -o build/temp.linux-armv7l-2.7/source/py_uart.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/c_uart.c -o build/temp.linux-armv7l-2.7/source/c_uart.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/py_uart.o build/temp.linux-armv7l-2.7/source/c_uart.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/UART.so

Successfully installed Adafruit-BBIO
Cleaning up...
```
Things are working fine now. However, notice in the above output that there is an inconsistency in the naming of the module. Both `Adafruit-BBIO` and `Adafruit_BBIO` are used. Could this have caused the "misspelling" of the module on my system?

Mausy5043, 2015-12-08:
Addendum:

While `pip` currently still insists the modulename has a minus-sign in it:
```
$ sudo pip freeze
Adafruit-BBIO==0.0.30
PyBBIO==0.10
argparse==1.2.1
distribute==0.7.3
gyp==0.1
numpy==1.6.2
pyserial==2.7
requests==2.8.1
serbus==1.0.4
smbus==1.1
urwid==1.0.1
wicd==1.7.2.4
wsgiref==0.1.2
```

Python is happy to accept the underscore in the name:
```
$ python
Python 2.7.3 (default, Mar 14 2014, 17:55:54)
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO
>>>
```

pdp7, 2016-08-20:
@Mausy5043 is this still an issue?  Maybe related to #107?

Mausy5043, 2016-08-20:
@pdp7 : This isn't an issue for me anymore because I've abandoned the Beaglebone platform in favour of the Raspberry Pi.

 I won't mind if you close this issue.

closed 2016-08-21 by pdp7