adafruit/adafruit-beaglebone-io-python #87: IOError when calling PWM.set_duty_cycle

munframed, 2015-12-21:
Hi,
I saw two related but closed topics on this. I'm using the latest Debian on a BBB, and everything works so far but this. `PWM.start` and `PWM.stop` work, but when I use `PWM.set_duty_cycle`, I get `IOError: [Errno 2] No such file or directory: '/slots'`. I probably should mention that I'm working in puTTY.
Is it about the device-tree-compiler? and if so how to fix it? I did some tests like this:
`root@beaglebone:/sys/devices/platform/bone_capemgr# more slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln`

and also there's no dts here
`root@beaglebone:/lib/firmware# ls *pwm* -al
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-ehrpwm1-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-ehrpwm2-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-epwmss0-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-epwmss1-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-epwmss2-00A0.dtbo`

RobertCNelson, 2015-12-21:
First, disable:
```
4: P-O-L- 0 Override Board Name,00A0,Override Manuf,cape-universaln
```
by removing "cape_universal=enable" from /boot/uEnv.txt

You'll need to rebuild "PyBBIO-ehrpwm1-00A0.dtbo" with the dtc found at /usr/local/bin/dtc (this supports 4.1.x)

Then modify the slots location stated here:
https://github.com/adafruit/adafruit-beaglebone-io-python/issues/80

munframed, 2015-12-21:
Thank you so much for your prompt reply. Since I'm still a newbe can you explain a bit more about how to rebuild "PyBBIO-ehrpwm1-00A0.dtbo" with the dtc?

RobertCNelson, 2015-12-21:
Oh, and "PyBBIO" is actually for:

https://github.com/graycatlabs/PyBBIO

So wrong 'base' project.. I know Alexander is working on v4.1.x changes..

munframed, 2015-12-21:
Can it be because `slots` is actually in `/sys/devices/platform/bone_capemgr` instead of `/sys/devices/bone_capemgr.8`? And if so how to fix it?

RobertCNelson, 2015-12-21:
The format of the compiled *.dtbo also changed, so even if you echo it to the new location, it won't work..

munframed, 2015-12-21:
Gotcha. It was kinda funny to ask that question as it has been addressed in the link you sent. Thanks. Now I don't have that 4th slot and I'm gonna skip the second step you mentioned and try the third. I'll get back to you. Thanks for your replies.

munframed, 2015-12-21:
I'm trying to run `dtc`. Does it take too much? as it's stuck there doing nothing. I have these listed
```
root@beaglebone:/usr/local/bin# ls -al
total 556
drwxr-xr-x  2 root staff   4096 Nov 12 21:17 .
drwxrwsr-x 10 root staff   4096 Nov 12 20:44 ..
lrwxrwxrwx  1 root staff     46 Nov 12 21:17 config-pin -> /opt/source/beaglebone-universal-io/config-pin
-rwxr-xr-x  1 root root   62944 Nov 12 21:17 convert-dtsv0
-rwxr-xr-x  1 root root  228984 Nov 12 21:17 dtc
-rwxr-xr-x  1 root root     636 Nov 12 21:17 dtdiff
-rwxr-xr-x  1 root root   40264 Nov 12 21:17 fdtdump
-rwxr-xr-x  1 root root   63088 Nov 12 21:17 fdtget
-rwxr-xr-x  1 root root   91896 Nov 12 21:17 fdtput
-rwx------  1 root staff    492 Nov 12 21:14 jekyll
-rwx------  1 root staff    500 Nov 12 21:12 kramdown
-rwx------  1 root staff    492 Nov 12 21:14 listen
-rwxr-xr-x  1 root staff  27447 Nov 12 21:14 miniterm.py
-rwx------  1 root staff    490 Nov 12 21:13 rougify
-rwx------  1 root staff    504 Nov 12 21:13 safe_yaml
-rwx------  1 root staff    484 Nov 12 21:13 sass
-rwx------  1 root staff    492 Nov 12 21:13 sass-convert
-rwx------  1 root staff    484 Nov 12 21:13 scss
```
Should I try the `config-pin`? Should I just stop being impatient?

RobertCNelson, 2015-12-21:
/usr/local/bin/dtc is the compiler, you need rebuild PyBBIO-ehrpwm1-00A0.dts..

BTW, you really should create a ticket here:
https://github.com/graycatlabs/PyBBIO/issues

That way you don't spam adafruit/adafruit-beaglebone-io-python
Regards,

munframed, 2015-12-21:
I'm sorry I didn't mean to spam, but I'm trying to do this tutorial https://learn.adafruit.com/controlling-a-servo-with-a-beaglebone-black/the-python-console, and it's Adafruit_BBIO and not PyBBIO, and here's where I first found few threads about the exact same issue ( #23, #6, #74 and #46 as examples),   . Should I still go creat a ticket there or I'm in a right place?

RobertCNelson, 2015-12-21:
Nope, your in correct place..  Your actually using adafruit's dtbo..

While these are from PyBBIO, (you just had confused me..)
```
root@beaglebone:/lib/firmware# ls *pwm* -al
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-ehrpwm1-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-ehrpwm2-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-epwmss0-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-epwmss1-00A0.dtbo
-rw------- 1 root root 353 Nov 12 21:14 PyBBIO-epwmss2-00A0.dtbo
```

munframed, 2015-12-21:
These are what I have under firmware! and `Adafruit_BBIO` is the only thing I have installed on this BBB.
Anyways, down under `/opt/source/beaglebone-universal-io/` I found a few dts
```
root@beaglebone:/opt/source/beaglebone-universal-io# ls -al
total 404
drwx------  3 debian debian   4096 Nov 12 21:17 .
drwx------ 14 debian debian   4096 Nov 12 21:18 ..
drwx------  8 debian debian   4096 Nov 12 21:17 .git
-rw-------  1 debian debian  18026 Nov 12 21:17 LICENSE
-rw-------  1 debian debian    884 Nov 12 21:17 Makefile
-rw-------  1 debian debian   4750 Nov 12 21:17 README.md
-rw-------  1 debian debian  10268 Nov 12 21:17 cape-univ-audio-00A0.dts
-rw-------  1 debian debian  15473 Nov 12 21:17 cape-univ-emmc-00A0.dts
-rw-------  1 debian debian  35515 Nov 12 21:17 cape-univ-hdmi-00A0.dts
-rw-------  1 debian debian  76647 Nov 12 21:17 cape-universal-00A0.dts
-rw-------  1 debian debian 120733 Nov 12 21:17 cape-universala-00A0.dts
-rw-------  1 debian debian  69152 Nov 12 21:17 cape-universaln-00A0.dts
-rwx------  1 debian debian  29196 Nov 12 21:17 config-pin
```
Though it might help. It's my last comment unless you need me to do something, this way I spam less :)

spmaniato, 2016-03-18:
@RobertCNelson in your first reply here, you said
> Then modify the slots location stated here: #80

In your notes there, I don't see instructions on how to move the location of slots. I'm on `Linux version 4.1.17-ti-rt-r47` and my slots location is: `/sys/devices/platform/bone_capemgr/slots`

What do I need to do in order to to fix the `No such file or directory: '/slots'` error in the original issue? Is it an OS fix or a change to this module's source code? Thank you :smile:

pdp7, 2016-08-20:
Please try again now that #108 has been merged.

closed 2016-10-19 by pdp7