adafruit/adafruit-beaglebone-io-python #90: ADC

sergiosrl, 2016-01-26:
Hi. I'm trying to read values from a sensor (analog). 
I have a simple code very similar to the one that reads te photoresistor

**test_analog.py**
import Adafruit_BBIO.ADC as ADC
import time
 
sensor_pin = 'P9_40'

ADC.setup()

while True:
    reading = ADC.read(sensor_pin)
    print('%f ' % (reading))
    time.sleep(1) 

I run _root@arm:/home/ubuntu# **python test_analog.py**_ and I get:
Traceback (most recent call last):
  File "teste_analog.py", line 6, in <module>
    ADC.setup()
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper

I have Ubutu 14.04.3 LTS, and my kernel is 4.1.15-ti-rt-r40.

The GPIOs work just fine

jeffesquivels, 2016-03-02:
I'm having this same issue using the latest Ubuntu Snappy image with "classic" mode. This is my kernel version:

```
(classic)ubuntu@localhost:~$ uname -a
Linux localhost.localdomain 4.4.0-2-generic #16-Ubuntu SMP Thu Jan 28 15:47:36 UTC 2016 armv7l armv7l armv7l GNU/Linux
```

I can confirm GPIO work correctly for me too.

jonhp, 2016-03-31:
I am having the exact same problem on Debian 8.3, Linux beaglebone 4.1.15-ti-rt-r43 #1 SMP PREEMPT RT Thu Jan 21 20:13:58 UTC 2016 armv7l GNU/Linux. Same result run as "root" or "debian".

hoffmanjon, 2016-04-16:
I am having the same issue as well, are there any plans to fix this?

pdp7, 2016-06-05:
@sergiosrl Support for ADC with Linux kernel 4.1+ has recently been merged: #96 

Please re-test and let us know how it worked.

pdp7, 2016-08-20:
@ladyada @jwcooper: Please close as there has been no response in over 2 months.

closed 2016-08-20 by ladyada