adafruit/adafruit-beaglebone-io-python #94: SPI not opening on BBB w/ Debian 4.1 and BB SPI cape

chorca, 2016-04-27:
Installed the library, and am currently loading BB-SPIDEV0 'cape' to enable HW SPI.

When trying to create the spi object, python appears to not be able to find the device?

```
debian@beaglebone:~$ ls -l /dev/spi*
crw-rw---- 1 root spi 153, 1 Apr 22 01:16 /dev/spidev1.0
crw-rw---- 1 root spi 153, 0 Apr 22 01:16 /dev/spidev1.1
debian@beaglebone:~$ sudo python
Python 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from Adafruit_BBIO.SPI import SPI
>>> spi = SPI(0,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
IOError: [Errno 2] No such file or directory
>>> 
```

SimplicityGuy, 2016-05-05:
I'm hitting the same exact thing.

Generally speaking, the instructions for enabling SPI on Debian Jessie with the 4.1 kernel are not very clear. Here's what I managed to cobble together:
```
1. clone this repo: https://github.com/beagleboard/bb.org-overlays
2. follow steps in that README
3. reboot
4. sudo sh -c "echo 'BB-SPIDEV1' > /sys/devices/platform/bone_capemgr/slots"
5. sudo sh -c "echo 'BB-SPIDEV0' > /sys/devices/platform/bone_capemgr/slots"
6. cat /sys/devices/platform/bone_capemgr/slots (see both SPI0 and SPI1 enabled and loaded)
7. Update uEnv.txt appropriately (adding BB-SPIDEV0, BB-SPIDEV1 to the enabled list, disabling HDMI)
```

```
~$ ls -al /dev/spi*
crw-rw---- 1 root spi 153, 0 May  4 21:38 /dev/spidev1.0
crw-rw---- 1 root spi 153, 1 May  4 21:38 /dev/spidev1.1
crw-rw---- 1 root spi 153, 2 May  4 21:38 /dev/spidev2.0
crw-rw---- 1 root spi 153, 3 May  4 21:38 /dev/spidev2.1
```

@chorca is this what you did too?

RobertCNelson, 2016-05-05:
For spi, you need to have them loaded via the kernel parameter in /boot/uEnv.txt

echoing it to the slots while in userspace is too late..

SimplicityGuy, 2016-05-05:
I added the following to my uEnv.txt:
```
cape_disable=bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN
cape_enable=bone_capemgr.enable_partno=BB-SPIDEV0,BB-SPIDEV1
```
however, I still hit the same issue:
```
>>> from Adafruit_BBIO.SPI import SPI
>>> spi = SPI(0,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
IOError: [Errno 2] No such file or directory
>>>
```

RobertCNelson, 2016-05-05:
The spidev overlays are correctly loaded. The issue is in the adafruit library, probably misreading the status of the slot file.

chorca, 2016-05-05:
My SPI overlays are also being loaded in at boot via uEnv.txt.

SimplicityGuy, 2016-05-05:
Thanks @RobertCNelson for confirming that the overlay is loaded correctly.

Given the error message it looks like the path is wrong to begin with... Looking at common.c on line 367:
```
           build_path("/sys/devices", "bone_capemgr", ctrl_dir, sizeof(ctrl_dir));
```

The initial path should be `/sys/devices/platform`.

Is the path that is there now on line 367 a legacy capemgr path? Or perhaps there should be a symlink to that path?

SimplicityGuy, 2016-05-05:
Here's a fix:
```C
diff --git a/source/common.c b/source/common.c
index 8911a9c..2ea7da9 100644
--- a/source/common.c
+++ b/source/common.c
@@ -334,9 +334,9 @@ int build_path(const char *partial_path, const char *prefix, char *full_path, si
 
 int get_spi_bus_path_number(unsigned int spi)
 {
-  char path[50];
+  char path[256];
   
-  build_path("/sys/devices", "ocp", ocp_dir, sizeof(ocp_dir));
+  build_path("/sys/devices/platform", "ocp", ocp_dir, sizeof(ocp_dir));
 
   if (spi == 0) {
     snprintf(path, sizeof(path), "%s/48030000.spi/spi_master/spi1", ocp_dir);
@@ -361,10 +361,10 @@ int get_spi_bus_path_number(unsigned int spi)
 int load_device_tree(const char *name)
 {
     FILE *file = NULL;
-    char slots[40];
+    char slots[64];
     char line[256];
 
-    build_path("/sys/devices", "bone_capemgr", ctrl_dir, sizeof(ctrl_dir));
+    build_path("/sys/devices/platform", "bone_capemgr", ctrl_dir, sizeof(ctrl_dir));
     snprintf(slots, sizeof(slots), "%s/slots", ctrl_dir);
 
     file = fopen(slots, "r+");
@@ -394,11 +394,11 @@ int load_device_tree(const char *name)
 int unload_device_tree(const char *name)
 {
     FILE *file = NULL;
-    char slots[40];
+    char slots[64];
     char line[256];
     char *slot_line;
 
-    build_path("/sys/devices", "bone_capemgr", ctrl_dir, sizeof(ctrl_dir));
+    build_path("/sys/devices/platform", "bone_capemgr", ctrl_dir, sizeof(ctrl_dir));
     snprintf(slots, sizeof(slots), "%s/slots", ctrl_dir);
 
     file = fopen(slots, "r+");
diff --git a/source/common.h b/source/common.h
index 2600b40..947fb2d 100644
--- a/source/common.h
+++ b/source/common.h
@@ -39,8 +39,8 @@ SOFTWARE.
 int gpio_mode;
 int gpio_direction[120];
 
-char ctrl_dir[35];
-char ocp_dir[25];
+char ctrl_dir[64];
+char ocp_dir[128];
 
 int get_gpio_number(const char *key, unsigned int *gpio);
 int get_pwm_key(const char *input, char *key);
```

Results:
```python
~$ python
Python 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from Adafruit_BBIO.SPI import SPI
>>> s = SPI(0,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
IOError: [Errno 13] Permission denied: '/sys/devices/platform/bone_capemgr/slots'
>>> exit()
~$ sudo python
Python 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from Adafruit_BBIO.SPI import SPI
>>> s = SPI(0,0)
>>> # success!
... 
>>> exit()
```

I'm going to clean up this fix a bit, since this will not work the for the 3.8 kernel. However, if you want to unblock yourself for now, give this a try. Once ready, I'll issue a PR for the change.

rolandschmidt, 2016-06-08:
In the master-branch from 08.06.2016 i just have to change line 346 from 
´build_path("/sys/devices", "ocp", ocp_dir, sizeof(ocp_dir));´
to
´build_path("/sys/devices/platform", "ocp", ocp_dir, sizeof(ocp_dir));´
to get the SPI object to work.
Additional code in line 346 like 
```
#ifdef BBBVERSION41
     build_path("/sys/devices/platform", "ocp", ocp_dir, sizeof(ocp_dir));
#else
     build_path("/sys/devices", "ocp", ocp_dir, sizeof(ocp_dir));
#endif
```
could be sufficient for the kernels > 4.1

pdp7, 2016-06-08:
@rolandschmidt thanks.  Could you please create a pull request with that fix?

pdp7, 2016-08-20:
There have been several fixes made for Linux 4.1+ support in #108. Please try again now that #108 has been merged.

kozga, 2016-09-30:
In version 1.0 declared path length is too low (50 bytes) 

```
  char path[50];

#ifdef BBBVERSION41
  strncpy(ocp_dir, "/sys/devices/platform/ocp", sizeof(ocp_dir));
#else
  build_path("/sys/devices", "ocp", ocp_dir, sizeof(ocp_dir));
#endif

  if (spi == 0) {
      snprintf(path, sizeof(path), "%s/48030000.spi/spi_master/spi1", ocp_dir);
  } else {
      snprintf(path, sizeof(path), "%s/481a0000.spi/spi_master/spi1", ocp_dir);
  }
```
But for kernels 4.1 path length is 54 bytes:
```
$ echo -n "/sys/devices/platform/ocp/48030000.spi/spi_master/spi1" | wc -c
54
```
In result, path is shortened to '/sys/devices/platform/ocp/48030000.spi/spi_master/'  and function always return 1, even if you try to open /dev/spidev2.x.

```
DIR* dir = opendir(path);
if (dir) {
    closedir(dir);
    //device is using /dev/spidev1.x
    return 1;
} else if (ENOENT == errno) {
    //device is using /dev/spidev2.x
    return 2;
} else {
    return -1;
}
```

pdp7, 2016-10-19:
see comments in #115

closed 2016-10-19 by pdp7