adafruit/adafruit-beaglebone-io-python #99: Initialising camera prroblem

GingerMuda, 2016-05-26:
Hi,

I am trying to initialize a camera when thee PIR sensor detects motion, see below my code and the error i am getting;
`import cv2
import Adafruit_BBIO.GPIO as GPIO
import time

#create variable called PIR, which refers to the P8_11 pin
PIR = "P8_11"

#Initialize the pin as an INPUT
GPIO.setup(PIR, GPIO.IN)
GPIO.add_event_detect(PIR, GPIO.RISING)

#capture from camera at location 0
cap = cv2.VideoCapture(0)
#set the width, height and frame rate.
cap.set(3,320)
cap.set(4,240)
cap.set(5, 30)

while (True):

    if (GPIO.event_detect(PIR)):
        ret, img = cap.read()
        cv2.imshow("input", img)

        key = cv2.waitKey(10)
        if key == 27:
            break
    time.sleep(0.05)

cv2.destroyAllWindows()
cap.release()
`

`root@beaglebone:~/Project/Testing# python video_capture.py
Traceback (most recent call last):
  File "video_capture.py", line 22, in <module>
    if (GPIO.event_detect(PIR)): 
AttributeError: 'module' object has no attribute 'event_detect'
root@beaglebone:~/Project/Testing# 
`

pdp7, 2016-07-01:
@GingerMuda I think the issue is that:
```
if (GPIO.event_detect(PIR)):
```
should be:
```
if (GPIO.event_detected(PIR)):
```

pdp7, 2016-08-20:
@ladyada @tdicola @jwcooper: no reply in over 1 month.  please close.

closed 2016-08-20 by ladyada