adafruit/adafruit-beaglebone-io-python #100: Beaglebone PWM at 50Hz not working (P8_19 P9_16)

cagdaseckin, 2016-05-28:
I wrote a code which needs 50Hz PWM. The code is given below. PWM P9_14 ,P9_42 and P8_13 works fine at 50 Hz but P8_19 (also P9_16) doesnt start any dedicated Hz PWM it only works with default. I couldnt find solution. Please help me to find what is the problem?

Pot driven PWM outputs code in Python.

```
import Adafruit_BBIO.ADC as ADC
import Adafruit_BBIO.PWM as PWM
# analog input setup
ADC.setup()
analogPin="P9_33"
# pwm start with default which works fine
""" passive
PWM.start("P9_14", 10.0)
PWM.start("P9_42", 10.0)
PWM.start("P8_13", 10.0,)
PWM.start("P8_19", 10.0)
""" 
# dedicated frequency for PWM. It doesnt start (Also 490Hz not working)
PWM.start("P9_14", 10.0,50)
PWM.start("P9_42", 10.0,50)
PWM.start("P8_13", 10.0,50)
PWM.start("P8_19", 10.0,50)

gir=raw_input("begin")
while (1):
        #read pot and write to pwm port
        potVal=ADC.read(analogPin)*100
        print potVal
        PWM.set_duty_cycle("P9_14",potVal)
        PWM.set_duty_cycle("P9_42",potVal)
        PWM.set_duty_cycle("P8_13",potVal)
        PWM.set_duty_cycle("P8_19",potVal)
        if potVal==0:
               break
PWM.stop("P9_14")
PWM.stop("P9_21")
PWM.stop("P8_13")
PWM.stop("P8_19")
PWM.cleanup()
```
Error respond with dedicated frequency like below but it start:

```
PWM.set_duty_cycle("P8_19",potVal)
RuntimeError: You must start() the PWM channel first
```
Configuration: Beaglebone Black RevC, Debian Wheezy

ThatSnail, 2016-06-01:
Getting a similar problem with the following test case:

    import Adafruit_BBIO.PWM as PWM

    pins = ['P9_21',
            'P9_22']

    for pin in pins:
        PWM.start(pin, 0, 1000)
        print pin, 'set'

    for pin in pins:
        PWM.set_duty_cycle(pin, 0)
        print pin

What version of PWM are you using?

    import Adafruit_BBIO.PWM as PWM
    print(PWM.VERSION)

cagdaseckin, 2016-06-01:
Verison : 0.0.20

cagdaseckin, 2016-06-01:
I discover that it didnt do same thing in differnt directory.  for example it givers error on /home/debian but it work fine in /home/debian/pb directory. it is happened in same device. Also I tried to do boot code from cron. But when code is in /home/debian/pb  code didnt work properly. It have to read UART but it didnt. Here is my codes

pwmpb.py

```
import Adafruit_BBIO.PWM as PWM
import numpy as np
class pwmpb():
    def __init__(self, ports=["P9_22","P9_31","P9_21","P9_29","P9_42","P9_14","P8_36","P9_16","P8_34","P8_19","P8_45","P8_13","P8_46","P9_28"]):
        self.ports=ports
        self.scale=np.arange(5.5,9.5,0.02)
        for p in self.ports:
            PWM.start(p, self.scale[50],50)
    def write(self,port,value):
        print self.ports[port], self.scale[value]
        PWM.set_duty_cycle(self.ports[port], self.scale[value])
    def stop(self):
        for p in self.ports:
            PWM.stop(p)
        PWM.cleanup()
```

xbeepb.py

```
# -*- coding: utf-8 -*-
import serial
import Adafruit_BBIO.UART as UART
class xbeepb():
    def __init__(self,port="/dev/ttyO5",baud=9600):
        self.port=port
        self.baud=baud
        self.uname="UART"+str(port[-1])
        self.uart=UART.setup(self.uname)
        self.ser= serial.Serial(port = self.port, baudrate=self.baud)
        self.ser.close()
        self.ser.open()
        self.status=False
        self.p1=0
        self.p2=0
        self.p3=0
        self.p4=0
        self.p5=0
        if self.ser.isOpen():
            self.satus=True
            self.ser.write("Pilot")
            self.ser.flush()
    def read(self):
        xbeein=self.ser.read()
        print xbeein
        pstr="~"
        if "~" in xbeein:
                print "Inputs:",
                codein=True
                while(codein):
                        xbeein=self.ser.read()
                        if  "}" in xbeein:
                                print pstr
                                parts=pstr
                                self.p1=ord(parts[1])
                                self.p2=ord(parts[2])
                                self.p3=ord(parts[3])
                                self.p4=ord(parts[4])
                                self.p5=ord(parts[5])
                                break
                        pstr=pstr+str(xbeein)
    def write(self,send):
        print "Gonderilen:",send
        self.ser.write(send)
```

main.py

```
import pwmpb
import xbeepb
import time

pwm=pwmpb.pwmpb()
xbee=xbeepb.xbeepb()

while(1):
    xbee.read()
    thrust=99-float(xbee.p1)
    pitch=float(xbee.p2)
    roll=float(xbee.p3)
    yaw=float(xbee.p4)
    oper=int(xbee.p5)
    if oper==0:
        print  thrust,pitch,roll, yaw
        pwm.write(0,thrust)
        pwm.write(1,pitch)
        pwm.write(2,roll)
        pwm.write(3,yaw)

```

ThatSnail, 2016-06-01:
With regards to the first problem, I've had some success by running `PWM.start(pin)` on all the PWM pins first, then using `PWM.set_frequency(pin, freq)` and `PWM.set_duty_cycle(pin, duty)` explicitly afterwards.

Example:

    PWM.start("P9_21")
    PWM.start("P9_22")
    PWM.set_frequency("P9_21", 1000)
    PWM.set_frequency("P9_22", 1000)
    PWM.set_duty_cycle("P9_21", 100)
    PWM.set_duty_cycle("P9_22", 100)

pdp7, 2016-08-20:
Please reset now that PWM fixes have been merged in #108

closed 2016-10-19 by pdp7