adafruit/adafruit-beaglebone-io-python #103: Add PWM support for Linux 4.1+

pdp7, 2016-06-05:
This library only support PWM for Linux kernel 3.8.  The way to utilize PWM on the TI AM3358 in newer kernels has changed.

I'm creating this issue to represent the effort add support for newer kernels, notably [4.1](http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#4.1.x-ti) and [4.4](http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#4.4.x-ti) as those are long-term kernels supported by TI.  @RobertCNelson currently using 4.4 for the [latest Debian Jessie images](http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_lxqt).

I'll be working on adding support in [my fork](https://github.com/adafruit/adafruit-beaglebone-io-python) and will create a pull request once I have something working.

pdp7, 2016-06-05:
BeagleBone Bacon Cape PWM testing with Linux 4.1.15-ti-rt-r43 on Debian Jessie 8.3 from the 2016-01-24 SD card image:

**pwm-test-jessie-4-1.sh**
```
echo '7 7 7 7' > /proc/sys/kernel/printk
config-pin P9.42 pwm
config-pin -q P9.42
config-pin P9.16 pwm
config-pin -q P9.16
config-pin P9.42 pwm
config-pin -q P9.42

for i in `find /sys |grep pwmchip|grep /export$`
do 
   echo $1 > $i
   ls $i
done

for i in `find /sys |grep pwmchip|grep /pwm$1 |grep enable$`
do 
   echo 1 > $i
   echo $i: `cat $i`
done

for i in `find /sys |grep pwmchip|grep /pwm$1 |grep period$`
do 
   echo $2 > $i
   echo $i: `cat $i`
done

for i in `find /sys |grep pwmchip|grep /pwm$1 |grep duty_cycle$`
do 
   echo $3 > $i
   echo $i: `cat $i`
done
```

**Output:**
```
root@beaglebone:~# bash ./pwm.sh 0 2000 1000
P9_42 Mode: pwm
P9_16 Mode: pwm
P9_42 Mode: pwm
/sys/devices/platform/ocp/48302000.epwmss/48302200.ehrpwm/pwm/pwmchip2/export
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip7/export
/sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm/pwmchip4/export
/sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm/pwmchip0/export
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip6/export
/sys/devices/platform/ocp/48302000.epwmss/48302200.ehrpwm/pwm/pwmchip2/pwm0/enable: 1
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip7/pwm0/enable: 1
/sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm/pwmchip4/pwm0/enable: 1
/sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm/pwmchip0/pwm0/enable: 1
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip6/pwm0/enable: 1
/sys/devices/platform/ocp/48302000.epwmss/48302200.ehrpwm/pwm/pwmchip2/pwm0/period: 2000
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip7/pwm0/period: 2000
/sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm/pwmchip4/pwm0/period: 2000
/sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm/pwmchip0/pwm0/period: 2000
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip6/pwm0/period: 2000
/sys/devices/platform/ocp/48302000.epwmss/48302200.ehrpwm/pwm/pwmchip2/pwm0/duty_cycle: 1000
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip7/pwm0/duty_cycle: 1000
/sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm/pwmchip4/pwm0/duty_cycle: 1000
/sys/devices/platform/ocp/48300000.epwmss/48300200.ehrpwm/pwm/pwmchip0/pwm0/duty_cycle: 1000
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip6/pwm0/duty_cycle: 1000
```

pdp7, 2016-06-05:
This library currently expects to load a device tree overlay for each PWM pin:
[c_pwm.c#L160](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/c_pwm.c#L160)
```
    snprintf(fragment, sizeof(fragment), "bone_pwm_%s", key);
```

The device tree overlay causes a directory in /sys to be created for that PWM pin:
[c_pwm.c#L169](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/c_pwm.c#L169)
```
    snprintf(pwm_test_fragment, sizeof(pwm_test_fragment), "pwm_test_%s", key);
```

The PWM pin is then controlled via 3 files (period, duty, polarity) inside the /sys sub-directory for the pin:
[c_pwm.c#L175](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/c_pwm.c#L175)
```
    //create the path for the period and duty
    snprintf(period_path, sizeof(period_path), "%s/period", pwm_test_path);
    snprintf(duty_path, sizeof(duty_path), "%s/duty", pwm_test_path);
    snprintf(polarity_path, sizeof(polarity_path), "%s/polarity", pwm_test_path);
```

For example, this device tree overlay will configure P9_16 for PWM: 
[bone_pwm_P9_16-00A0.dts](https://github.com/beagleboard/devicetree-source/blob/master/arch/arm/boot/dts/bone_pwm_P9_16-00A0.dts)
```
			pwm_test_P9_16 {
				compatible	= "pwm_test";
				pwms 		= <&ehrpwm1 1 500000 1>;
				pwm-names 	= "PWM_P9_16";
			    pinctrl-names	= "default";
			    pinctrl-0	= <&pwm_P9_16>;
				enabled		= <1>;
				duty		= <0>;
				status 		= "okay";
			};
```

pdp7, 2016-06-06:
I'm reviewing **Bonescript** by @jkridner to see how it handles PWM on 4.1+ kernels.

The current development branch is: **[bonescript/0.5.x](https://github.com/jadonk/bonescript/tree/0.5.x)**

@MarkAYoder has been several commits this year related to PWM:

- [Starting to fix pwm](https://github.com/jadonk/bonescript/commit/d8a18b3fabd0af72b1295852c809e2f966699e23) 
- [Fixed export value. Added comments about pwmPath](https://github.com/jadonk/bonescript/commit/d37144d22ec0599ceb9401fecd0c0f371620dd1c)
- [Added pwm addresses. Still need to do eCap (P9_28 and P9_42)](https://github.com/jadonk/bonescript/commit/2557cc41df5c8f8ec3c96812178be02462f2da5a)

pdp7, 2016-06-08:
PWM related code from [PyBBIO](https://github.com/graycatlabs/PyBBIO/blob/master/bbio/platform/beaglebone/config.py) by @graycatlabs 
```
# Predefined resolutions for analogWrite():
RES_16BIT = 2**16
RES_8BIT  = 2**8
PERCENT   = 100

# Default frequency in Hz of PWM modules (must be >0):
PWM_DEFAULT_FREQ = 100000

# PWM config dict in form:
#  ['overlay_file', 'path/to/ocp_helper_dir', ['required', 'overlays']]

PWM_PINS = {
  'PWM1A' : ['bone_pwm_P9_14', '%s/pwm_test_P9_14.*' % OCP_PATH, 
             ['PyBBIO-epwmss1', 'PyBBIO-ehrpwm1']],
  'PWM1B' : ['bone_pwm_P9_16', '%s/pwm_test_P9_16.*' % OCP_PATH, 
             ['PyBBIO-epwmss1', 'PyBBIO-ehrpwm1']],

  'PWM2A' : ['bone_pwm_P8_19', '%s/pwm_test_P8_19.*' % OCP_PATH, 
             ['PyBBIO-epwmss2', 'PyBBIO-ehrpwm2']],
  'PWM2B' : ['bone_pwm_P8_13', '%s/pwm_test_P8_13.*' % OCP_PATH, 
             ['PyBBIO-epwmss2', 'PyBBIO-ehrpwm2']],

  'ECAP0' : ['bone_pwm_P9_42', '%s/pwm_test_P9_42.*' % OCP_PATH, 
             ['PyBBIO-epwmss0', 'PyBBIO-ecap0']],
  'ECAP1' : ['bone_pwm_P9_28', '%s/pwm_test_P9_28.*' % OCP_PATH, 
             ['PyBBIO-epwmss1', 'PyBBIO-ecap1']],

}
# Using the built-in pin overlays for now, I see no need for custom ones 

PWM1A = 'PWM1A'
PWM1B = 'PWM1B'
PWM2A = 'PWM2A'
PWM2B = 'PWM2B'
ECAP0 = 'ECAP0'
ECAP1 = 'ECAP1'

# ocp helper filenames:
PWM_RUN      = 'run'
PWM_DUTY     = 'duty'
PWM_PERIOD   = 'period'
PWM_POLARITY = 'polarity'

PWM_DEFAULT_PERIOD = int(1e9/PWM_DEFAULT_FREQ)
```

MatthewWest, 2016-07-18:
@pdp7 Any update on progress? I am also available to work on this issue, since my company would like to update to Debian 8, but we need PWM in this library to work.

Also I'm not sure where to ask, but do you have any idea who is responsible for releasing a new package version released with the 4.1+ support?

pdp7, 2016-07-20:
Hi, I've been travelling recently and just getting back to this now.  If we can develop a solution that doesn't cause a regression for 3.8 then it will be merged.  As for PyPi package, I'll have to check with @jwcooper or @tdicola

MatthewWest, 2016-07-20:
I thought I'd do a similar summary of how PWM / IO generally works in 4.1+.

# /sys changes
The locations of a number of key pieces of hardware have changed. This is probably the easiest fix to make (and has been mostly made for other pieces of the library).

```
/sys/devices/bone_capemgr.*/slots => /sys/devices/platform/bone_capemgr/slots
/sys/devices/ocp.* => /sys/devices/platform/ocp/
```

# Capes loaded on boot

There are two main changes. First, HDMI and EMMC are no longer virtual capes loaded on boot, but are handled natively. I haven't looked in detail into how this works.

Second, and more significantly, the idea of loading a separate overlay for each pin has been basically abandoned, in favor of using a [Universal IO cape](https://github.com/cdsteinkuehler/beaglebone-universal-io). The universal cape is enabled on boot by default, with an option to disable it in `boot/uEnv.txt` by removing the `cape_universal=enable` argument from the `cmdline=` line (more detail [here](https://groups.google.com/forum/#!topic/beagleboard/2D5Pz3r7ZZ8)).

Depending on the `dtb=` line of `uEnv.txt`, different univeral capes are loaded. The `dtb`s which previously disabled the HDMI or EMMC behave the same way (see [the boot script](https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L419) for details).

Given that a universal cape is loaded at boot, attempting to load a device tree overlay which conflicts with any of its pins will fail. The universal cape uses almost all the pins on the board: all but the ADC and I2C pins, and depending on the `dtb=` line of `uEnv.txt`, the HDMI, HDMI audio, or EMMC pins.

So 4.1+ replaces the concept of loading in new device tree overlays for each pin / group of pins with a combination of a universal cape and pinmuxing. There is no need to reinvent the wheel with the pinmuxing---the `config-pin` executable provided by [beaglebone-universal-io](https://github.com/cdsteinkuehler/beaglebone-universal-io) works fine. However, since it's not included with the console images, we'll need to figure out whether we want to introduce a dependency or whether we should just pull it in ourselves.

# BoneScript approach to PWM

Now I'll describe the approach taken by BoneScript.

[src/hw_mainline.js](https://github.com/jadonk/bonescript/blob/master/src/hw_mainline.js)

```js
    } else if(template == 'bspwm') {
        fs.writeFileSync(pinmux+"/state", 'pwm');
        // Buld a path like: /sys/devices/platform/ocp/48304000.epwmss/48304200.ehrpwm/pwm/pwmchip5
        // pin.pwm.chip looks up the first address and pin.pwm.addr looks up the second
        // file_find figures which pwmchip to use
        // pin.pwm.index tells with half of the PWM to use (0 or 1)
        var pwmPath = my.file_find('/sys/devices/platform/ocp/'+pin.pwm.chip
                + '.epwmss/'+pin.pwm.addr+'.pwm/pwm', 'pwmchip', 1);
        pwmPrefix[pin.pwm.name] = pwmPath + '/pwm' + pin.pwm.index;
        if(debug) winston.debug("pwmPrefix[pin.pwm.name] = " + pwmPrefix[pin.pwm.name]);
        if(debug) winston.debug("pin.pwm.sysfs = " + pin.pwm.sysfs);
        if(!my.file_existsSync(pwmPrefix[pin.pwm.name])) {
            fs.appendFileSync(pwmPath+'/export', pin.pwm.index);
        }
        fs.appendFileSync(pwmPrefix[pin.pwm.name]+'/enable', 1);
```

Given a pin ("P9_21" or similar), this section of code sets the pinmux state to PWM. It finds the path to the PWM chip. I'm not sure why it uses a path with `'.pwm/pwm'` instead of `'.ehrpwm/pwm'`, as in my system running `4.4.15-bone11`, the correct path uses `'.ehrpwm'`. It saves the path to the pwmPrefix dictionary, which makes future look-ups easy.

There was another step in the changes made by BoneScript at kernel version 4.1. They depended on exporting PWMs from the /sys/class/pwm directory. That strategy can still be found in [src/hw_universal.js](https://github.com/jadonk/bonescript/blob/master/src/hw_universal.js). As of kernel 4.4, that strategy no longer works, so they use the strategy shown above in [src/hw_mainline.js](https://github.com/jadonk/bonescript/blob/master/src/hw_mainline.js).

# Implementation thoughts

We'll need to keep track of some additional information about pins which have pwm as an output choice. Specifically, we'll need the chip address, and the device address. When initializing a pin for PWM, there are two steps: first, mux the pin to the correct value. This can be done by calling `config-pin` or by writing our own function and maintaining some sort of data structure with all the pins and mux settings, then writing to `/sys/devices/platform/ocp/ocp:P#_##_pinmux/state`.

One possible implementation would be to define an interface in a header file, which is implemented by two separate source files, one for <=3.8 and the other for >=4.1.

Calling @RobertCNelson - would you be willing to read through and make sure I don't have any major misunderstandings of what's going on? From what I can tell you appear to understand well how all the pieces fit together.

cocasema, 2016-07-21:
> I'm not sure why it uses a path with '.pwm/pwm' instead of '.ehrpwm/pwm', as in my system running 4.4.15-bone11, the correct path uses '.ehrpwm'.

@MatthewWest 
Just a note, on my bbb with 4.4.12-ti-r30 some pwm path contain '.ecap/pwm'.

```
# uname -a
Linux arm 4.4.12-ti-r30 #1 SMP Thu Jun 9 07:50:17 UTC 2016 armv7l armv7l armv7l GNU/Linux

# ll /sys/class/pwm/
 pwmchip0 -> ../../devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/
 pwmchip1 -> ../../devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip1/
 pwmchip2 -> ../../devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip2/
 pwmchip4 -> ../../devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip4/
 pwmchip6 -> ../../devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/
```

pdp7, 2016-07-21:
@MatthewWest Thank you for the summary and your thoughts.  That is very helpful.  I think you are right that the necessary information for a pin capable of PWM can be borrowed from **[bonescript/src/bone.js](https://github.com/jadonk/bonescript/blob/master/src/bone.js)**

For example, pin `P8_13` has name `EHRPWM2B` at [Line 282](https://github.com/jadonk/bonescript/blob/master/src/bone.js#L282):
```
    {
        "name": "EHRPWM2B",
        "gpio": 23,
        "mux": "gpmc_ad9",
        "eeprom": 15,
        "pwm": {
            "module": "ehrpwm2",
            "sysfs": 6,
            "index": 1,
            "muxmode": 4,
            "path": "ehrpwm.2:1",
            "name": "EHRPWM2B",
            "chip": "48304000",
            "addr": "48304200"
        },
        "key": "P8_13",
        "muxRegOffset": "0x024",
        "options": [
            "gpmc_ad9",
            "lcd_data22",
            "mmc1_dat1",
            "mmc2_dat5",
            "ehrpwm2B",
            "pr1_mii0_col",
            "NA",
            "gpio0_23"
        ]
    },
```

MatthewWest, 2016-08-08:
I've implemented a patch which I have confirmed works for Linux 4.1+, though I haven't had a chance to test it with Linux 3.8. The pull request is at #108.

pdp7, 2016-08-20:
Resolved by @MatthewWest's excellent work in #108

closed 2016-08-20 by pdp7

tomek242, 2017-09-13:
Hello folks.
Maybe You help me with my problem.
I try to build simply mobile robot based on this project:
http://myroboticadventure.blogspot.com/2014/05/my-first-working-robot-its-alive.html
I use Debian 7.11 with standard kernel and standard Adafruit I/O library taken from here:
https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/installation-on-ubuntu
When I try to wrote simple program in Python like this:

import Adafruit_BBIO.PWM as PWM
import time
pin1 = "P9_14"
pin2 = "P9_16"
pin3 = "P8_13"
pin4 = "P8_16"
PWM.start(pin1,0)
PWM.start(pin2,0)
PWM.start(pin3,0)
PWM.start(pin4,0)
for i in range(0, 100):
              PWM.set_duty_cycle(pin,80)
              PWM.set_duty_cycle(pin,80)
              PWM.set_duty_cycle(pin,80)
              PWM.set_duty_cycle(pin,80)
              time.sleep(.1)

PWM.stop(pin1)
PWM.stop(pin2)
PWM.stop(pin3)
PWM.stop(pin4)
PWM.cleanup()

my BBB show me a errors: Duty is larger than periodand, only rotates with three wheels, output P9_14 as if it did not work at all. After restart my BBB does not boot the wheels at all, and display message: Invalid sysfs file or something like that.
I was try other debian images but no result. Could someone help me in here?
Thanks.