adafruit/adafruit-beaglebone-io-python #105: Can't toggle PWM on and off, won't seem to come back on reliably

jstoner, 2016-06-22:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

Beaglebone black
```
0s 👍  #> cat /etc/dogtag
BeagleBoard.org Debian Image 2015-11-03
0s 👍  #> uname -a
Linux rcdev 3.8.13-bone79 #1 SMP Tue Oct 13 20:44:55 UTC 2015 armv7l GNU/Linux

```

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**
```

0s 👍  #> . env/bin/activate
0s 👍  #> python --version
Python 2.7.11

```

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

none 

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

I am trying to do sort of PWM within PWM. I have a thread running, calling PWM.start() and PWM.stop() on a pin. Here's the simplest version of my code:
```

import Adafruit_BBIO.PWM as PWM
from time import sleep

while True:
    PWM.start('P9_14', 50, 3)
    sleep(2)
    PWM.stop('P9_14')
    sleep(2)

```
I have it set up with a test LED. 

Expected result:
 modulate and show six flashes, and then a two second pause, and repeat ad infinitum. 

Actual result:
six flashes, and then a two second pause, and then a single flash, and then a few seconds, and then another single flash, and the single flashes repeat.

I have seen other unpredicted behavior with more sophisticated code--long periods on instead of flashing, that sort of thing, but when I boil it down to something basic, this is what I see. Ultimately I am planning to go to 100,000Hz, modulated over a period of an hour, with this application.

Here's my post to the adafruit forums:

https://forums.adafruit.com/viewtopic.php?f=49&t=98191

jstoner, 2016-06-22:
This does work:

```
import Adafruit_BBIO.PWM as PWM
from time import sleep

while True:
    PWM.start('P9_14', 50, 3)
    sleep(2)
    PWM.set_duty_cycle('P9_14', 0)
    sleep(2)
```
Though I still think PWM.stop() should work.

pdp7, 2016-08-20:
Please try again now that #108 has been merged.

closed 2016-10-19 by pdp7