adafruit/adafruit-beaglebone-io-python #107: New PyPI version needed

MatthewWest, 2016-07-26:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Arrow BeagleBone Black Industrial, Debian 8.5, Linux kernel 4.4.15-bone11

- Python version (run `python -version` or `python3 -version`): Python 3.4.2

- Error message you are receiving, including any Python exception traces:  None

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): None

Just opening an issue to request a new release tag and PyPI release. I'd rather not be tying my use of the library to a specific git commit, since I need 4.1+ support (I'm working around the lack of PWM support in 4.1+).

pdp7, 2016-08-20:
@jwcooper @tdicola: Would either of you be able to handle updating Adafruit_BBIO on PyPi?

https://pypi.python.org/pypi/Adafruit_BBIO/

PeteLawler, 2016-09-04:
I'd actually nominate this to be called a bug because the README.rst says that the easiest way to install is via 'pip install', but that's not at all the case now as there's been significant changes since the last update to the PyPi.

pdp7, 2016-09-05:
@tdicola @jwcooper: could we please update PyPi with the latest version of `Adafruit_BBIO`?

PeteLawler, 2016-10-02:
It'd appear this has been achieved since bc373997aaecfffd4449a789e2bf09a38a70eba3 - with https://pypi.python.org/pypi/Adafruit_BBIO/1.0.0 becoming available.

pdp7, 2016-10-05:
Yeah, I published v1.0.0 last week.  Thanks for the reminder,.

closed 2016-10-05 by pdp7