adafruit/adafruit-beaglebone-io-python #115: SPI: IOError: [Errno 22] Invalid argument in xfer and xfer2 functions

kozga, 2016-09-28:
Beagle Bone Black with kernel 4.4.8. 

Simple script:
```
from Adafruit_BBIO.SPI import SPI        
                                         
spi = SPI(1, 0)                          
spi.mode = 0                             
spi.msh = 500000                         
                                         
msg = spi.xfer2([0xd7, 0x00])            
print ' '.join('0x%02x' % b for b in msg)
```

Result:
```
Traceback (most recent call last):
  File "./spi_flash.py", line 14, in <module>
    msg = spi.xfer2([0xff])
IOError: [Errno 22] Invalid argument
```

The cause of this error is uninitialised spi_ioc_transfer structure. After initializing struct in xfer2 and changing malloc to calloc in xfer, functions work as expected.

pdp7, 2016-09-29:
@kozga  Thanks for reporting this.

> After initializing struct in xfer2 and changing malloc to calloc in xfer, functions work as expected.
Does that mean that you've been able to create a fix?

If so, could you please create a pull request so that I can review?

thanks

pdp7, 2016-10-15:
@kozga  I've merged the pull request.  Could you please retest and close this issue if resolved?  Thanks!

pdp7, 2016-10-15:
@kozga  FYI - test after new build from master:
```
root@beaglebone:~# cat xfer.py 
from Adafruit_BBIO.SPI import SPI        

spi = SPI(1, 0)                          
spi.mode = 0                             
spi.msh = 500000                         

msg = spi.xfer2([0xd7, 0x00])            
print ' '.join('0x%02x' % b for b in msg)

root@beaglebone:~# python xfer.py
0x00 0x00
```

kozga, 2016-10-18:
It works, but before that I had to patch another error in get_spi_bus_path_number() in common.c (see my comment in #94). Anyway, I'm closing this bug.

closed 2016-10-18 by kozga

pdp7, 2016-10-19:
@kozga Could you please create a pull request with that patch for `path` in `common.c`?

pdp7, 2016-10-19:
@kozga thanks for your insights.  I've included a fix in commit 452f2e7e