adafruit/adafruit-beaglebone-io-python #117: SPI Problems in Python on Adafruit_BBIO-1.0.0

pdp7, 2016-10-05:
From Adafruit forum post:
[SPI Problems in Python on Adafruit_BBIO-1.0.0](https://forums.adafruit.com/viewtopic.php?f=49&t=104291#p522124)
Written by mrburns42:
> I have disabled the universal cape in the /boot/uEnv.txt. If I cat the slots there are none active. I installed the latest overlays from the URL you provided. With the old Adafruit library0.0.3, the SPI access fails with the following. Notice that I cat the slots after the command. Back in the 7.X images, running this same test script would automatically load the appropriate cape.
```
root@beaglebone:/home/steve# ./SPI_Test.py
Traceback (most recent call last):
  File "./SPI_Test.py", line 5, in <module>
    spi = SPI(0,0)      #/dev/spidev1.0
IOError: [Errno 2] No such file or directory
root@beaglebone:/home/steve# cd /sys/devices/platform/bone_capemgr/
root@beaglebone:/sys/devices/platform/bone_capemgr# cat slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
```
I swapped to the SDCard where I attempted to patch the SPI python library for the 1.0.0. This is an 8.4 image without the upgrade installed. I applied the latest overlay from the Github. The universal cape is disabled. Below is the installed capes before and after my simple SPI script.
```
root@beaglebone:/sys/devices/platform/bone_capemgr# cat slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
root@beaglebone:/sys/devices/platform/bone_capemgr# /home/steve/SPI_Test.py
[  161.937952] bone_capemgr bone_capemgr: part_number 'BB-SPIDEV0', version 'N/A'
[  161.945655] bone_capemgr bone_capemgr: slot #4: override
[  161.951093] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[  161.958131] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-SPIDEV0'
[  161.984557] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-SPIDEV0-00A0.dtbo' loaded; overlay id #0
Read Byte[1] was 0x0.
Read Byte[2] was 0x0.
Read Byte[3] was 0x0.
Read Byte[4] was 0x0.
Read Byte[5] was 0x0.
Read Byte[6] was 0x0.
root@beaglebone:/sys/devices/platform/bone_capemgr# cat slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-SPIDEV0
root@beaglebone:/sys/devices/platform/bone_capemgr#
```
Notice that all the SPI read bytes return as 0x0. The SPI clock does not toggle and there is nothing attached to BBB headers but the logic analyzer. On the old version the SPI clock toggles and the read data is 0xFF.
```

pdp7, 2016-10-05:
@MatthewWest Any thoughts on this issue?  Thanks!

RobertCNelson, 2016-10-05:
@pdp7 i've noticed spidev is broken on v4.4.x, remove the edma fix that made things work in v4.1.x:

https://github.com/RobertCNelson/dtb-rebuilder/blob/4.4-ti/src/arm/am335x-boneblack.dts#L12

Regards,

MatthewWest, 2016-10-05:
@pdp7 I basically didn't make any changes to the SPIDEV code and don't have a working understanding of how it works at the moment. Also I don't have the bandwidth during an academic term to deal with it, sorry.

mrburns42, 2016-10-05:
Here is my simple SPI test code.  With nothing attached to the BBB, it should run without error and it should printed read values of  0xFF.  At least that is the behavior on 3.8.   On 4.4,  it prints out an error message.   You have to disable the universal cape before running this. 
`#!/usr/bin/python
from Adafruit_BBIO.SPI import SPI
import time

spi = SPI(0,0)	#/dev/spidev1.0

#Set the maximum clock frequency to 500kHz so it is easy to view on the CRO
spi.msh=500000

# Transfer 0x9F,  and a bunch of 0x00.  The 0x9F commands the SPI EEPROM to read
# its  ID Registers the 0x00 are dummy bytes that pads out the length to the
# number of read bytes requested.
result = spi.xfer2([0x9F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00])
# The result of the transfer will be an array of 7 bytes where the first byte is what
# was read while the command byte was written and the rest are the devices
# contents of the ID registers.

for i in range(1, 7):
  val = result[i]
  print "Read Byte[%d] was 0x%x." %(i, val)
`

pdp7, 2016-10-15:
@mrburns42 Here are the results running the test.  I do not get any error.  However, "Read Byte" always displays 0x0.  I believe that would still be an issue?

### Execute test program:
```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
root@beaglebone:~# python ./mrburns42-test.py 
Read Byte[1] was 0x0.
Read Byte[2] was 0x0.
Read Byte[3] was 0x0.
Read Byte[4] was 0x0.
Read Byte[5] was 0x0.
Read Byte[6] was 0x0.
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-SPIDEV0
root@beaglebone:~# ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Oct 15 19:13 /dev/spidev1.0
crw-rw---- 1 root spi 153, 1 Oct 15 19:13 /dev/spidev1.1
root@beaglebone:~# 
```

### dmesg:
```
root@beaglebone:~# dmesg
[ 1566.978926] bone_capemgr bone_capemgr: part_number 'BB-SPIDEV0', version 'N/A'
[ 1566.978957] bone_capemgr bone_capemgr: slot #4: override
[ 1566.978974] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[ 1566.978993] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-SPIDEV0'
[ 1567.001603] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-SPIDEV0-00A0.dtbo' loaded; overlay id #0
```

### Test program:
```
root@beaglebone:~# cat mrburns42-test.py 
#!/usr/bin/python
from Adafruit_BBIO.SPI import SPI
import time

spi = SPI(0,0)  #/dev/spidev1.0

#Set the maximum clock frequency to 500kHz so it is easy to view on the CRO
spi.msh=500000

# Transfer 0x9F,  and a bunch of 0x00.  The 0x9F commands the SPI EEPROM to read
# its  ID Registers the 0x00 are dummy bytes that pads out the length to the
# number of read bytes requested.
result = spi.xfer2([0x9F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00])
# The result of the transfer will be an array of 7 bytes where the first byte is what
# was read while the command byte was written and the rest are the devices
# contents of the ID registers.

for i in range(1, 7):
  val = result[i]
  print "Read Byte[%d] was 0x%x." %(i, val)
```

### System Info:
```
root@beaglebone:~# uname -a
Linux beaglebone 4.4.24-bone14 #1 Sun Oct 9 23:53:07 UTC 2016 armv7l GNU/Linux
root@beaglebone:~# cat /etc/debian_version 
8.6
root@beaglebone:~# cat /etc/dogtag 
BeagleBoard.org Debian Image 2016-09-18
root@beaglebone:~# 
```

pdp7, 2016-10-19:
@mrburns42 You wrote:
>  On 4.4, it prints out an error message. You have to disable the universal cape before running this. 

I am able to run `mrburns42-test.py` OK with `cape-universaln` enabled.

**Could you paste the error message that you received?**

```
root@beaglebone:~# uname -a
Linux beaglebone 4.4.21-bone13 #1 Fri Sep 16 07:34:28 UTC 2016 armv7l GNU/Linux
root@beaglebone:~# cat /etc/debian_version 
8.6
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
root@beaglebone:~# python mrburns42-test.py
Read Byte[1] was 0x0.
Read Byte[2] was 0x0.
Read Byte[3] was 0x0.
Read Byte[4] was 0x0.
Read Byte[5] was 0x0.
Read Byte[6] was 0x0.
```

mrburns42, 2016-10-21:
Currently I am running version 7.9 to have a working SPI interface.   I saved an SD Card with 
  1. Base Install 8.4 Expanded File System 
  2. Universal Cape removed from /boot/uEnv.txt 
  3. Lastest overlay stuff added from https://github.com/beagleboard/bb.org-overlays
                                                                                    
root@beaglebone:/home/steve# uname -a
Linux beaglebone 4.4.9-ti-r25 #1 SMP Thu May 5 23:08:13 UTC 2016 armv7l GNU/Linux

When I run the test, I get 
  File "./SPI_Test.py", line 5, in <module>
    spi = SPI(0,0)      #/dev/spidev1.0
IOError: [Errno 2] No such file or directory

After running the test, there are no slots loaded as shown below

root@beaglebone:/sys/devices/platform/bone_capemgr# cat slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1

If I list the installed Adafruit python library, it is still at 0.0.30

root@beaglebone:/sys/devices/platform/bone_capemgr# pip list
Adafruit-BBIO (0.0.30)

I did a apt-get update and a pip install Adafruit_BBIO
However, the Adafruit python library remained at 0.0.30, which can in the base image.  The only way I could update it was to remove it and then re-install it.  After the reinstall, I did get to 

root@beaglebone:/home/steve# pip list
Adafruit-BBIO (1.0.0)

Now running the test, I get

root@beaglebone:/home/steve# ./SPI_Test.py
Traceback (most recent call last):
  File "./SPI_Test.py", line 14, in <module>
    result = spi.xfer2([0x9F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00])
IOError: [Errno 22] Invalid argument

root@beaglebone:/sys/devices/platform/bone_capemgr# cat slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-SPIDEV0

So, my conclusions are: 
1.  The 8.X image files contain the old 0.0.30 version of the Adafruit python libraries.  These old libraries are not compatible with 8.X  and then will not update themselves.  They must be removed and then reinstalled. 
2. The xfer2 in SPI has a problem that has already been documented elsewhere.    

Best regards.

---- Steve

RobertCNelson, 2016-10-21:
```
root@beaglebone:/home/steve# uname -a
Linux beaglebone 4.4.9-ti-r25 #1 SMP Thu May 5 23:08:13 UTC 2016 armv7l GNU/Linux
```

Please upgrade, your kernel:

```
cd /opt/scripts/tools/
git pull
sudo ./update_kernel.sh --ti-channel --lts-4_4
```
in 4.4.21-ti-r47, i had fixed the spidev interface.. (on of our fixes from the 4.1.x line had broken v4.4.x)..

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/04b9bf61266c38ae94f4a3b22af16177e4a6276b

Regards,

pdp7, 2016-10-21:
@mrburns42 nice to have github.com back up!  Good point about creating a new PyPI release.  I merged several pull requests recently, so it is an appropriate time for me to do so.

Could you run your test program with `strace`?  I think the error is related to an `ioctl()`.  @kozga had raised this in #115 as you mentioned.

After that, could you install the latest `master` from this repo and run the test program with `strace` again?

mrburns42, 2016-10-24:
Sorry it has taken a couple of days.  One of the BBB was loaned out to a colleague and the other one was at work. 
 
The output from strace is almost 1000 lines long.   Below is the output for the xfer2 command that fails. 

open("./SPI_Test.py", O_RDONLY|O_LARGEFILE) = 4
fstat64(4, {st_mode=S_IFREG|0755, st_size=757, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f8b000
read(4, "#!/usr/bin/python\nfrom Adafruit_"..., 4096) = 757
write(2, "    ", 4    )                     = 4
write(2, "result = spi.xfer2([0x9F, 0x00, "..., 63result = spi.xfer2([0x9F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00])
) = 63
close(4)                                = 0
munmap(0xb6f8b000, 4096)                = 0
write(2, "IOError", 7IOError)                  = 7
write(2, ": ", 2: )                       = 2
write(2, "[Errno 22] Invalid argument", 27[Errno 22] Invalid argument) = 27
write(2, "\n", 1
)                       = 1
rt_sigaction(SIGINT, {SIG_DFL, [], SA_RESTORER, 0xb6da9ae1}, {0xd7625, [], SA_RESTORER, 0xb6da9ae1}, 8) = 0
close(3)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++

Not sure what you wanted me to do with the latest master.  Therefore,  I followed the manual install procedure from github and rebuilt local on the BBB.  

Now after running the script,  I do not get the invalid argument on xfer2.  Instead, I get the 0x00 values that you had seen.  However, hooking up a logic analyzer and running a slightly different script that is compatible with the HW currently attached to my BBB,  I see the proper signals on all the SPI lines,  the SPI device recognizes the commands, and I do get read data back.  

Therefore,  I think SPI is working in what I have installed right now.  The difference in this version and the previous version seems to be only the pullup resistor on the MISO line.  The previous version must have had this pin specified as pullup.  That is why, when I run the old script version with no HW, I get 0xFF.  The newer code must have either no termination resistor specified, or it has this specified as pulldown.    Personally, I think pullup is the correct setting, since MISO can be paralleled between several SPI devices.

pdp7, 2016-10-28:
@mrburns42 Unfortunately, I don't have a SPI EEPROM.  I'm trying to figure out what tests I can do to replicate the problem you are having.

Is there any other SPI slave devices that you want to get working?  I'm hoping we might have something in common.

I do have a couple TFT LCDs that can be controlled via SPI.  I was able to get the [Adafruit 2.2" LCD](https://www.adafruit.com/product/1480) to work with the [Adafruit_Python_ILI9341](https://github.com/adafruit/Adafruit_Python_ILI9341) library by @tdicola.

I'm using **Linux 4.1.30-ti-r69**.   I have **cape-universal** enabled:
```
# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,univ-all
```

I used `config-pin` setup the SPI0 and GPIO pins for LCD:
```
config-pin P9_12 gpio
config-pin P9_15 gpio
config-pin P9_17 spi
config-pin P9_18 spi
config-pin P9_21 spi
config-pin P9_22 spi
```

**Next I'll test the MCP3008 ADC chip via SPI0 on BeagleBone Black**

mrburns42, 2016-10-31:
Hi Drew,
I am sorry if you got the impression from my last post that there are still problem.  There is no longer any SPI problems that I am aware of.   The steps to get SPI working were.
1.  Uninstall the Adafruit BBIO libraries and reinstall. (This moved from 0.0.3 to 1.0)
2.  Update the Linux Kernel per Robert Nelson above.
3.  Perform the manual install of the Adafruit BBIO .

After that the SPI accesses are working.  The only item I noted was that MISO is not pulled up on the BBB.  The previous overlay version had this pin defined as an input with pullup.  

The above steps would be combined into a single patch, but all the pieces to produce that patch seem to be known now. 

---- Steve

pdp7, 2016-11-09:
Thanks for the detailed follow-up.  Should I close this issue?

mrburns42, 2016-11-09:
I am fine with closing the issue.

pdp7, 2016-11-11:
thanks for your input

closed 2016-11-11 by pdp7

caxefaizan, 2020-11-13:
Hi. i am using BBB to get values in SPI mode but am not able to find a proper documentation to make it work. I have connected the pins for SPI0 with the Ic from which i want to get the values.
so my code includes
`from Adafruit_BBIO.SPI import SPI`

then i initialize the spi as `spi = SPI(1,0)` 
followed by
`spi.open(1,0)`
`spi.writebytes([address])`

since it was not working i tried the tutorial on github.
but using `spi.xfer2([32, 11, 110, 22, 220])` also doesnt seem to work.