adafruit/adafruit-beaglebone-io-python #118: Problem when PWM start in different frequencies

dengber, 2016-10-10:
Hello,
I encountered a problem when start PWM in different frequencies on channels, would you help to resolve it? thanks!

When starting PWM on P9_14 and P9_16 in different frequencies, RuntimeError occurred.
>import Adafruit_BBIO.PWM as PWM
>PWM.start('P9_14', 0, 100)
>PWM.start('P9_16', 0, 200)
>>Traceback (most recent call last):
>>  File "<stdin>", line 1, in <module>
>>RuntimeError: Problem with a sysfs file

If start in same frequency, no Error occurs.
> PWM.start('P9_16', 0,100)

I also tried to start in same frequency, then change one of them to different frequency, the same RuntimeError occurred when changing frequency.

The following is my hardware/software versions:
- Beaglebone Black with Debian 8.4
- Python 2.7.9
- Adafruit-BBIO (1.0.0) (installed by pip)

pdp7, 2016-10-12:
I wasn't able to reproduce this issue.  Could you please try it with [the latest Debian image](https://beagleboard.org/latest-images)?

dengber, 2016-10-13:
My Debian is the latest image with file name bone-debian-8.4-lxqt-4gb-armhf-2016-05-13-4gb.img.xz. (Though it is labeled "Debian 8.5 2016-05-13 4GB SD LXQT" in the image download page, it link to a Debian 8.4 image.)

I retried the python code again with a new flashed SD card, which with Adafruit-BBIO 0.0.30 preinstalled, the error changed to 
> IOError: [Errno 2] No such file or directory: '/slots'

Then I uninstalled Adafruit-BBIO 0.0.30 and installed the lastest Adafruit-BBIO 1.0.0 with pip, and tried the python code, the error was still:
> RuntimeError: Problem with a sysfs file

pdp7, 2016-10-13:
@dengber Ok, thanks for clarifying.  It would be helpful if you could run `strace` so that I can better understand what is happening.

Install `strace` (if it doesn't already exist):
```
sudo apt-get install strace
```

Then run `strace`, replace `test.py` with the name of your test program:
```
strace -f -o strace.log python test.py
```

Paste the contents of `strace.log` into a [gist](https://gist.github.com/).

dengber, 2016-10-14:
@pdp7 thanks for response, the following link is content of strace.log.

https://gist.github.com/dengber/4fca06df1db3d09ee6105c1598776e97

pdp7, 2016-10-15:
@dengber Thanks for the log.  I can also reproduce this now.  I'm further investigating the cause.

### execute test program:
```
root@beaglebone:~# python test.py 
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    PWM.start('P9_16', 0, 200)
RuntimeError: Problem with a sysfs file
```

### error in dmesg:
```
[   91.363062] ehrpwm 48302200.ehrpwm: Period value conflicts with channel 0
```

### slots file:
```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
```

### system info:
```
root@beaglebone:~# cat /etc/dogtag 
BeagleBoard.org Debian Image 2016-09-18
root@beaglebone:~# cat /etc/debian_version 
8.6
root@beaglebone:~# uname -a
Linux beaglebone 4.4.24-bone14 #1 Sun Oct 9 23:53:07 UTC 2016 armv7l GNU/Linux
```

### test program:
```
root@beaglebone:~# cat test.py 
import Adafruit_BBIO.PWM as PWM
PWM.start('P9_14', 0, 100)
PWM.start('P9_16', 0, 200)
```

pdp7, 2016-10-15:
@dengber  I found this BoneScript issue which seems similar:
https://github.com/jadonk/bonescript/issues/49

@NikolaMorena suggested:
> Workaround: use pins that are not connected with same EHRPWM module, e.g. P9_14 and P8_13

I tested an this does work OK:
```
root@beaglebone:~# cat test.py 
import Adafruit_BBIO.PWM as PWM
PWM.start('P9_14', 0, 100)
PWM.start('P8_13', 0, 200)
root@beaglebone:~# python ./test.py 
root@beaglebone:~# 
```

I suggest using those pins if you are able to.

dengber, 2016-10-17:
@pdp7 Use pwm pins of different EHRPWM modules resolve the problem, thank you for your kindly supporting on this issue!

closed 2016-10-17 by dengber

sebastianimsys, 2018-10-23:
adafruit_bbio.pwm problem with the cape manager debian 9.5

sebastianimsys, 2018-10-23:
adafruit_bbio.pwm problem with the cape manager debian 9.5 ....can you help me please?﻿