adafruit/adafruit-beaglebone-io-python #120: v4.1.x + cape_universaln + Adafruit_BBIO.GPIO

jcdammeyer, 2016-10-18:
I've been asked to start a new issue so here it is.
Problem: Debian Jesse v4.1.x now comes with cape_universaln installed and I can't find out where that's done.  Because the cape_universal_ grabs all the GPIO a .dts that uses one of the pins fails to load as does the Adafruit_BBIO.GPIO library.  Would like to prevent it from being loaded in the first place.

Materials: BBB RevB booting from MicroSD card.  Headless operation with just Ethernet and power supply. Sparkfun Prototype CAPE with connector wired in to P8-11. CAPE LED connected to P9.12.

Method: Edit /boot/uEnv.txt to remove cape_universaln from slots file.  Add a One-WIre dtbo file that talks to DS1822 temperature sensor and Edit uEnv.txt to install to slots file.
Eventually add LEDs and other GPIO pins to the same .dts (dtbo) file and program information into EEROM so One-Wire and assorted I/O are a physical cape in location 3 rather than a virtual cape.

Observation:  Although removing the cape_universaln works as a cat $SLOTS shows only the 4 empty physical capes and dmesg shows that CAPE #3 has an uninitialized EEROM, the virtual cape defined by BB-W1-P8.11.dts does not show up.  It must be copied to the slots file before my Python program can access the w1_slave file to get the temperature.

Running a small test program rr_blinking_lights.py from "The BeagleBone Black Primer" by Brian McLaughlin  doesn't affect the GPIO pins.  But running the Blink.py from the /examples folder under Cloud9 does.    So I may have typed it in wrong.

Conclusion:   The 5 books I have on the BBB are all targeted at the older device handling methods including Derek Molloy's excellent book.  What's needed is a something similar to the Adafruit page 
https://learn.adafruit.com/introduction-to-the-beaglebone-black-device-tree/device-tree-background
but with a bit more depth.   
What to do when it doesn't work.  
What not to do anymore like is now 
cape_enable=bone_capemgr.enable_partno=BB-W1-P8.11
incorrect?

For now.  This is what I do but I'm not happy with the process.  It's like I'm missing something.
/* Inside uEnv.txt */
##Enable the One Wire interface on P8-11
cape_disable=bone_capemgr.disable_partno=cape-universaln
cape_enable=bone_capemgr.enable_partno=BB-W1-P8.11

/* Install One Wire Virtual Cape and start up Python Server as a background task */
 sudo sh -c "echo BB-W1-P8.11 >/sys/devices/platform/bone_capemgr/slots"
 sudo python tcpServer.py >/dev/null 2>/dev/null &

/* See which DS1822 one wires have been discovered, DS18B20 would start with 28- */
 ls /sys/devices/w1*/22*/w1_slave
/sys/devices/w1_bus_master1/22-000000219f58/w1_slave

/* Get the information from the DS1822.
 cat /sys/devices/w1_bus_master1/22-000000219f58/w1_slave
5b 01 4b 46 7f ff 05 10 b5 : crc=b5 YES
5b 01 4b 46 7f ff 05 10 b5 t=21687

[BB-W1-P8.11.dts.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/534913/BB-W1-P8.11.dts.txt)
[uEnv.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/534916/uEnv.txt)

RobertCNelson, 2016-10-18:
@jcdammeyer 

```
uname_r=4.1.4-ti-r8
cmdline=coherent_pool=1M quiet
```

You have an early rootfs, on Oct 14, 2015, i made this a runtime check:

https://github.com/RobertCNelson/boot-scripts/commit/7730db559cf4b03b0d845b2bdc3e04542a7dcff6

so just:

cd /opt/scripts/
git pull

Regards,

1: repo at /opt/scripts/ : https://github.com/RobertCNelson/boot-scripts

pdp7, 2016-10-18:
Also, I believe that if you want to completely disable `cape-universal` you'll need remove `cape_universal=enable` from `cmdline` in `/boot/unEnv.txt`

From:
```
cmdline=coherent_pool=1M quiet cape_universal=enable
```

to:
```
cmdline=coherent_pool=1M quiet 
```

jcdammeyer, 2016-10-18:
Thanks Robert.   The 'pull' did change the /opt/scripts/boot/am335x_evm.sh to match the one in the link you posted.

I believe I did start this MicroSD card with an 8.1 Debian.  Specifically
bone-debian-8.1-lxqt-4gb-armhf-2015-08-09-4gb.img

With apt-get update and apt-get upgrade I'm into 
root@beaglebone:/opt/scripts/boot# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 8.6 (jessie)
Release:        8.6
Codename:       jessie
root@beaglebone:/opt/scripts/boot# uname -a
Linux beaglebone 4.1.4-ti-r8 #1 SMP PREEMPT Thu Aug 6 21:01:16 UTC 2015 armv7l GNU/Linux

Because I've done the pull, and the files have changed, and since my cmdline didn't have cape_universal=enable, I don't have to remove it.  What I did do is comment out the cape_disable and now cat $SLOTS shows the virtual cape is still gone.  YAHOO!
#cape_disable=bone_capemgr.disable_partno=cape-universaln

And now if I add in cape_universal=enable onto the cmd it's back and behaving as it should.

So why does this next statement not work?
cape_enable=bone_capemgr.enable_partno=BB-W1-P8.11
but this does.
 sudo sh -c "echo BB-W1-P8.11 >/sys/devices/platform/bone_capemgr/slots"

john@beaglebone:/lib/firmware$ ls BB-W1* -al
-rw-r--r-- 1 root root 982 Oct 16 14:29 BB-W1-P8.11-00A0.dtbo

Doesn't work if I add the -00A0 onto the end of it either.

pdp7, 2016-10-18:
Are there any errors in dmesg?  

Usually, there should be additional information in the kernel log when an overlay fails to load.

jcdammeyer, 2016-10-18:
How I wish.  The slot3 invalid signature is from the cape EEROM being uninitialized.  It appears to be OK.  Yet it doesn't show up in the slots file.  When cape_universaln was loaded I'd get the error that the pin was already reserved.  
Also perhaps another clue.  If I do echo the dts file to slots and then look at the slots file it's in virtual cape 5.  Yet the dmesg log shows it in slot 4.  And the dmesg log after sending it to slots shows extra information.  See below.  
So there's something different going on with the echo verses the cape_enable.  But what?

[    4.186480] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,000B,1114BBBK2834'
[    4.186535] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    4.239680] bone_capemgr bone_capemgr: slot #0: No cape found
[    4.299681] bone_capemgr bone_capemgr: slot #1: No cape found
[    4.359663] bone_capemgr bone_capemgr: slot #2: No cape found
[    4.389680] bone_capemgr bone_capemgr: Invalid signature '00000000' at slot 3
[    4.396885] bone_capemgr bone_capemgr: slot #3: No cape found
[    4.402772] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-W1-P8.11-00A0' VER 'N/A' PR '0'
[    4.402805] bone_capemgr bone_capemgr: slot #4: override
[    4.402830] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[    4.402857] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-W1-P8.11-00A0'
[    4.403409] bone_capemgr bone_capemgr: initialized OK.

[ 3535.476975] bone_capemgr bone_capemgr: part_number 'BB-W1-P8.11', version 'N/A'
[ 3535.477020] bone_capemgr bone_capemgr: slot #5: override
[ 3535.477038] bone_capemgr bone_capemgr: Using override eeprom data at slot 5
[ 3535.477057] bone_capemgr bone_capemgr: slot #5: 'Override Board Name,00A0,Override Manuf,BB-W1-P8.11'
[ 3535.488674] bone_capemgr bone_capemgr: slot #5: dtbo 'BB-W1-P8.11-00A0.dtbo' loaded; overlay id #0
[ 3535.534011] Driver for 1-wire Dallas network protocol.
jo

RobertCNelson, 2016-10-18:
@jcdammeyer 

```
So why does this next statement not work?
cape_enable=bone_capemgr.enable_partno=BB-W1-P8.11
but this does.
sudo sh -c "echo BB-W1-P8.11 >/sys/devices/platform/bone_capemgr/slots"
```

make sure "BB-W1-P8.11.dtbo" is under /lib/firmware/ then run:

```
sudo update-initramfs -uk `uname -r`
```

The *.dtbo should now be copied to the initramfs, so it'll work pre-rootfs mount /..

Regards,

jcdammeyer, 2016-10-18:
Thanks Robert.  
On a new reboot, my dts virtual driver is in place.  
john@beaglebone:~$ cat $SLOTS
---
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-W1-P8.11

Now the dmesg shows
[    4.403554] bone_capemgr bone_capemgr: initialized OK.
...
[    4.425670] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-W1-P8.11-00A0.dtbo' loaded; overlay id #0
...
[   13.527681] Driver for 1-wire Dallas network protocol.

I suppose since this is an image of the rootfs,  that if I make a change to my driver,  the system will continue to use the one on the ramfs when starting up until I rebuild ramfs?  So this operation is something one would do at the end of the development cycle to create a headless stand alone embedded system?

Clearly looking back at all this:  
apt-get update
apt-get upgrade 
isn't enough though since the 'git pull' was fairly critical in getting things to work correctly with the cape_universaln removal.  
Is there something else that will do the equivalent of the ' git pull'

Until I have problems how do I know that I need to do a git pull or any other types of commands?

Most of the books discussing the Device Tree use
/sys/devices/bone_capemgr.x/slots  where .x is a specific version I guess.  And they don't have 'platform' in the path.

You've been very helpful.    Thank you again.
John

pdp7, 2016-10-19:
@jcdammeyer you wrote:
> On a new reboot, my dts virtual driver is in place. 
Does that mean that this issue is resolved?

jcdammeyer, 2016-10-19:
Not quite.  I need to still test that the Adafruit_BBIO_GPIO Python library still works.  

Also as part of this, was the question as to why the pull was needed if the image was updated/upgraded in September 2016 but the change referred to by Robert was done a year earlier.

Hopefully no one else will run into this problem because the pull isn't needed.  But if it is I'd like this issue to show the problem and the entire solution.  So far it feels 99% done and we all know how long that last 1% can take.

pdp7, 2016-10-19:
@jcdammeyer wrote:
> Also as part of this, was the question as to why the pull was needed if the image was updated/upgraded in September 2016 but the change referred to by Robert was done a year earlier.

@RobertCNelson: could you please comment about this?

RobertCNelson, 2016-10-20:
@jcdammeyer 
this should be easily explained by you running:
```
cat /etc/dogtag
```
Regards,

jcdammeyer, 2016-10-20:
Thanks Robert.  The results of that are:
BeagleBoard.org Debian Image 2015-08-09

I'm hanging in on this thread and this subject a little while longer because I really dislike reading a thread that stops when the original poster states. Ah. Fixed it!   
...
Yes.  That's it.  That's all.  
...
Then when someone asks, the Linux guru's chastise the poster for not reading the thousands of web forums and wasting their time.  I'd like to be able to summarize this issue with a list of points that addresses the initial issue and shows in the final closing posting what was done and why.  So often the 'why' is missing.
So Robert's point addresses the 'why' really quite well.  
Which version am I running and why is one more important than the other?

1. There is uname -a
Linux beaglebone 4.1.4-ti-r8 #1 SMP PREEMPT Thu Aug 6 21:01:16 UTC 2015 armv7l GNU/Linux

2. lsb_release -a
Distributor ID: Debian
Description:    Debian GNU/Linux 8.6 (jessie)
Release:        8.6
Codename:       jessie

3. cat /etc/dogtag
BeagleBoard.org Debian Image 2015-08-09

All the literature that I've seen states that the Jesse release was done summer 2016.  Since /etc/dogtag is just a text file why should that even be important.  

And to stay with the original issue, how does anyone know that they have to do a 'pull' to get the cape_universaln removed from being automatically installed. 

Anyone is welcome to post a link here to a document that describes this all in one place.  Clearly apt-get upgrade changed my Debian Image to 8.6.  The Beagleboard .org site lists Debian 8.5 2016-05-13 4GB SD LXQT  as the latest image.   If I had started all over with that image would that not have required the pull?

RobertCNelson, 2016-10-20:
@jcdammeyer for us, "/etc/dogtag" is the only real way to determine the initial state of the rootfs.  lsb_release, reads /etc/debian_version, which is controlled via the base-files package: https://packages.qa.debian.org/b/base-files.html 

Correct, this shouldn't have been an issue with the "latest" version listed on beagleboard.org

PS: we are currently testing a replacement https://rcn-ee.net/rootfs/bb.org/testing/2016-10-16/

Regards,

pdp7, 2016-10-28:
@jcdammeyer are there any outstanding questions for this issue?

jcdammeyer, 2016-10-28:
I've been distracted all week with a new Delta 3D printer (Eventually Beagle with Replicape driven) that there hasn't been time to do much on this issue.  However, the one test I did appeared to have the GPIO behaving as expected so I think I'll close this one.  Thanks for everyone's help.

closed 2016-10-28 by jcdammeyer

pdp7, 2016-10-28:
Thanks for the @jcdammeyer.  The delta + replicape sounds exciting!   Is there a URL I can checkout with more details about your project?

jcdammeyer, 2016-10-29:
This is what I bought.  
https://www.3dprintersonlinestore.com/geeetech-metal-delta-3d-printer-kit
It's a piece of crap  with the rod end tolerances so bad that there's more than 1mm slop on the hot ends due to slop in the rods.  The linear bearings were originally metal shrouded and are now cheaper units in plastic carriers.  Wiggle one of those and again the heater nozzle moves almost 2mm.   
To adjust it requires editing the configuration.h file and recompiling and uploading the arduino firmware.  I don't think they've ever heard of EEROM.
The extruders are really cheap and keep blocking up.  I've had two successful prints and since then haven't had any luck.  My Cartesian based 3D printer, (also really cheap) had been outstanding and reliable.  It's a tool, turn it on, download the file to the Raspberrry Pi running Octopi and through the Octoprint interface and walk away.

This is the Replicape.
http://www.thing-printer.com/product/replicape/
Nice board.  I think support will be good.  But until my mechanics are stable no amount of software will fix the problems.  It too uses OctoPrint.