adafruit/adafruit-beaglebone-io-python #122: Add eQEP support for 4.x kernel

theonteg, 2016-11-13:
Hello,
Is is possible to add support to read position and change period of eQEP in kernel 4.x ?

pdp7, 2016-11-14:
@theonteg I believe this should be a matter of writing to files in /sys.  Do you know that paths for position and periond?

theonteg, 2016-11-14:
@pdp7 yes,
for eQEP0: /sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/
for eQEP1: /sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/
for eQEP2: /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/
The files enabled, period and position.

This is available in kernel 4.4.31-ti-r67. Some previous versions doesn't have these files.

pdp7, 2016-11-16:
@theonteg Thanks for the details.  I see that @rseetham added [RotaryEncoder](https://github.com/graycatlabs/PyBBIO/tree/master/bbio/libraries/RotaryEncoder) to PyBBIO.

It looks like it should be possible to add something similar to Adafruit_BBIO.  I can't give an estimate as to when I will have time to implement this feature, but I'd welcome any pull requests.

pdp7, 2016-12-01:
FYI - I came across this code today by @mvduin and wanted to make a note in case it useful for implementation here:
[py-uio/ti/qep.py](https://github.com/mvduin/py-uio/blob/master/ti/qep.py)
> add preliminary QEP code

mvduin, 2016-12-01:
Note that it doesn't use the (very limited, non-mainline) kernel driver but directly accesses the peripheral. This means that apart from DT differences it should work on any kernel (the uio_pdrv_genirq driver has existed since ancient times, and is enabled on rcn's kernels all the way down to 3.8-series if you use a not-too-ancient build).

pdp7, 2017-02-02:
@theonteg Here's an update:
I found a bug in the tieqep driver in the 4.9 kernel (which is currently in testing for the beaglebone debian image).   I've tracked down the solution and Robert C. Nelson, who builds the beaglebone kernels and debian images, has now merged the fix:beaglebone
*tieqep: add patch to fix unhandled fault on 4.9 kernel*
https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/f89418b8f79dea8a843dab279b7c47c5a4b79ef9

RobertCNelson, 2017-02-02:
4.4.45-ti-r86/4.4.45-ti-rt-r86 with @pdp7 's fix should hit the apt repo later this afternoon.

Regards,

pdp7, 2017-02-02:
@robertcnelson Will there also be a package for the new 4.9.7-ti-r19 kernel?

RobertCNelson, 2017-02-02:
@pdp7 crap, sorry meant: 4.9.7-ti-r19/4.9.7-ti-rt-r19, complete brainfart..

Regards,

pdp7, 2017-04-20:
I'm using P8.11 and P8.12 for testing a rotary encoder.

Using cape-universal:
```
debian@beaglebone:~$ cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,univ-all
 5: P-O-L-   1 Override Board Name,00A0,Override Manuf,BB-ADC
```

Configure the pin mode for eQEP:
```
debian@beaglebone:~$ config-pin p8.11 qep && config-pin p8.12 qep

debian@beaglebone:~$ config-pin -q p8.11 && config-pin -q p8.12 
P8_11 Mode: qep
P8_12 Mode: qep
```

System info:
```
debian@beaglebone:~$ uname -a
Linux beaglebone 4.4.47-ti-r87 #1 SMP Mon Feb 6 22:21:49 UTC 2017 armv7l GNU/Linux
debian@beaglebone:~$ cat /etc/debian_version 
8.7
debian@beaglebone:~$ sudo cat /etc/dogtag 
BeagleBoard.org Debian Image 2015-12-07
```

pdp7, 2017-05-10:
_NOTE: my BBB is running Debian 8.8 with Linux 4.4.62-ti-r99 kernel_

I've now been able to use eqep0, eqep1 and eqep2 OK with `config-pin`.  I had to modify `cape-universala-00A0.dts` to enable pins `P8_33` and `P8_35` for `eqep1`: https://github.com/pdp7/bb.org-overlays/commit/fd65f9ec580cd210cc3c401ce77ebd631f25e262

Enable eqep0 for pins `P9_27` and `P9_92` _(an alias for `P9_42.1`)_:
```
config-pin P9.27 qep
config-pin P9.92 qep 
```
Enable eqep1 for pins `P8_33` and `P8_35`:
```
config-pin P8.33 qep 
config-pin P8.35 qep
```
Enable eqep2 for pins `P8_11` and `P8_11`:
```
config-pin P8.11 qep 
config-pin P8.12 qep 
```
eqep0 position:
```
cat /sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/position
```
eqep1 position:
```
cat /sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/position
```
eqep2 position:
```
cat /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/position
```

Next step is to add eqep0 and eqep1 to the Rotary Encoder class

RobertCNelson, 2017-05-10:
and pushed to the apt repo:

```
sudo apt update ; sudo apt upgrade
```

and it should show an upgrade for bb-cape-overlays
bb-cape-overlays_4.4.20170510.0

reboot and it should work, let us know..

pdp7, 2017-05-10:
Thanks @RobertCNelson!

pdp7, 2017-05-12:
I have  documented how-to to enable all the eqep pins:
https://github.com/adafruit/adafruit-beaglebone-io-python/commit/c418cdae9a2a2c0d52412561c0125b0d227af4eb

BeagleBone must boot with cape-universal enabled and load the cape-universala overlay in order to
use all the eQEP pins:

### Install the latest Device Tree overlays:
```
sudo apt-get upgrade bb-cape-overlays
```

### File: /boot/uEnv.txt
```
uname_r=4.4.62-ti-r99
cmdline=coherent_pool=1M quiet cape_universal=enable
cape_enable=bone_capemgr.enable_partno=cape-universala
```

### File: /sys/devices/platform/bone_capemgr/slots
```
0: PF----  -1 
1: PF----  -1 
2: PF----  -1 
3: PF----  -1 
4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universala
```

### eqep0: P9_27, P9_92
```
config-pin P9_27 qep
config-pin P9_92 qep # alias for P9_42.1
cat /sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/position
```

### eqep1: P8.33, P8.35
```
config-pin P8.33 qep 
config-pin P8.35 qep
cat /sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/position
```

### eqep2: P8.11, P8.12
```
config-pin P8.11 qep 
config-pin P8.12 qep 
cat /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/position
```

### eqep2b: P8.41, P8.42
_alternate pins for eqep2 (mutually exclusive)_
```
config-pin P8.41 qep 
config-pin P8.42 qep 
cat /sys/devices/platform/ocp/48304000.epwmss/48304180.eqep/position
```

### TODO: implement in corresponding methods in `Encoder.py`

pdp7, 2017-08-23:
I merged my work-in-progress (WIP) for rotary encoder support with eQEP in Pull Request #149.

@theonteg any feedback on what I've created so far?

theonteg, 2017-08-24:
I'm sorry, but I have burned my board and until I replace it I won't be able to check... 

Thanks for the implementation though, although I workaround it by reading direct from the paths provided I can't wait to check BBIO!

AKAMEDIASYSTEM, 2017-11-03:
Hi, I have followed the instructions here (thanks @pdp7!!), but I believe I'm encountering a problem with the encoder on P8.33 and P8.35. When I try to use `config-pin` to set those pins to qep, I see this error:
```
P8_33 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P8_33_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P8_33_pinmux/state
```

...despite this, I can get the other two encoders working. Here's my setup:
```
sudo ./version.sh
git:/opt/scripts/:[a5d5f437baeaa7f8e14d5ae037268abf7a67e88e]
eeprom:[A335BNLT00C0414BBBK1848�]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.4.91-ti-r133]
nodejs:[v6.11.5]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171101.0-0rcnee1~stretch+20171101]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

...and my apologies if this is posted in the wrong area - I looked around for awhile and this seemed like the most germane place to post this. Happy to delete or move it if needed!

Thanks,

AKA

RobertCNelson, 2017-11-03:
@AKAMEDIASYSTEM 
P8.33 is used for the hdmi output.. to use this pin for something else like the eqep, you need to disable it:
add 
```
disable_uboot_overlay_video=1
```
to /boot/uEnv.txt

Regards,

AKAMEDIASYSTEM, 2017-11-03:
Thank you so much for your fast help - everything works great now.

...so I shouldn't bother trying to downgrade my kernel from 4.4.etc to 4.14.foo?

Thanks again!

RobertCNelson, 2017-11-03:
4.4 -> 4.14 is an upgrade..

Right now 4.4.x/v4.9.x/4.14.x are pretty similar, i have no plans to EOL 4.4.x this year (2017)..

Regards,

AKAMEDIASYSTEM, 2017-11-03:
Oh, my mistake! Thanks for the clarifications and again, for the super-quick help!
Have a good weekend!
AKA

dplanella, 2017-11-14:
In terms of giving feedback, the implementation works on my board (thanks!). The only big issue I found so far is the verbose debug output turned on by default. I submitted issue #191 for this, which also documents my test setup.

A smaller issue I found was when I mistakenly set the `eqep_num` to EQEP2B upon initialization. As I don't have EQEP2B enabled, the `Encoder` module quit with an index out of range exception. Perhaps this could be handled more graciously? Happy to submit another issue for that.

In any case, it seems that given that the support in the kernel is there and that the Encoder module is available to write code for rotary encoders is also there, should this issue be closed?

dplanella, 2017-11-16:
The comments in the Encoder module state that the universal cape needs to be loaded in `/boot/uEnv.txt` to be able to use the QEP inputs. Like so:

```
cmdline=cape_universal=enable # plus all the other optione
cape_enable=bone_capemgr.enable_partno=cape-universala
```

There is also another cape_universal option in `/boot/uEnv.txt`: 

```
enable_uboot_cape_universal=1
```

This makes it three different places where the universal cape is enabled. It sounds unnecessarily redundant, so I must be doing something wrong.

In addition to that, I just learnt that the universal cape is enabled by default, so I'm not sure if there are some legacy options being thrown in here, or if there is anything to be enabled at all for QEP usage.

In other words what is the proper way to enable the use of QEP0, QEP1 and QEP2 hardware in the latest Debian images? Or are they all already enabled by default?

RobertCNelson, 2017-11-16:
The proper way forward is via

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

However for some users they have an old version of UBoot so make sure to verify.

debian@beaglebone:~$ cd /opt/scripts/tools/
debian@beaglebone:/opt/scripts/tools$ git pull
debian@beaglebone:/opt/scripts/tools$ sudo ./version.sh | grep bootloader
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.07-00002-gd316f5106c]

Regards,

dplanella, 2017-11-16:
Thanks a lot for the reply and the link. On my previous comment I should have said I had already read that, but I still could not figure out why the universal cape needs to be enabled in the three different places.

I'm still not sure if I'm specifying legacy options, but I'd be happy to update the QEP instructions once I've figured out, hence the question.

As far as I know, I'm running the latest version of uboot:

```
$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[a5d5f437baeaa7f8e14d5ae037268abf7a67e88e]
eeprom:[A335BNLT00C03614BBBK0298]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.4.91-ti-r133]
nodejs:[v6.12.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171107.0-0rcnee1~stretch+20171107]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

RobertCNelson, 2017-11-16:
@dplanella so with u-boot overlays enabled, only this option is needed to enable cape_universal

```
enable_uboot_cape_universal=1
```

The other two:

```
cmdline=cape_universal=enable # plus all the other optione
cape_enable=bone_capemgr.enable_partno=cape-universala
```
are legacy for kernel overlays.

Regards,

dplanella, 2017-11-16:
Perfect, that worked nicely, thanks!

So it seems that on a recent Debian image QEP code should work out of the box, given that:

- The universal cape is already enabled by default
- The pins are configured as `qep` by the `Encoder.RotaryEncoder` class. I.e. no need to do it on the command line

The caveat is that the universal cape seems to make eqep0 (pins 9.27 and 9.92) and eqep2 (pins P8.11, P8.12) available. So if you're using them, you're all set.

However, eqep1 (P8.33, P8.35) and eqep2b (P8.41, P8.42, alternative, mutually exclusive pins for eqep2) don't seem to be available. Here's the debug output from the Encoder.RotaryEncoder class initialization:

http://pastebin.ubuntu.com/25975075

dplanella, 2017-11-23:
@RobertCNelson, if I want to enable all eqep modules, it seems I need to use `cape-universala` instead of `cape-universal`. What is the proper way to load this other cape with uboot overlays?

I did read https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays, but I can't quite figure it out from there. I.e. do I need to explicitly load `cape-universala` (and if so, how?) and comment out `enable_uboot_cape_universal=1`? Or alternatively, use a particular set of `disable_*` options to effectively get `cape-universala` loaded?

RobertCNelson, 2017-11-23:
So with u-boot overlays, just disable the virtual capes you don't want to use, then all the pins used for that function will be available to you..

No worrying about which cape-xyz to load. ;)

Regards,

dplanella, 2017-11-24:
@RobertCNelson that did it, thanks! I simply had to ensure `cape-universal` was enabled (as per the default), and then uncomment `disable_uboot_overlay_video=1` to make the QEP channels that share pins with HDMI available. Now all channels are available.

The issue that got me confused is that upon making QEP2b available with the settings above, QEP2 stopped working on my code. This was due to issue #200: if both the eQEP2 and eQEP2b channels are available, they are also both configured in sequence in the Encoder Python API. Given that they are mutually exclusive, if eQEP2 is configured first and eQEP2b is configured afterwards, eQEP2 will be inadvertently disabled.

pdp7, 2017-11-30:
@dplanella thanks for all your contributions.  do you think I should close this issue now?

dplanella, 2017-11-30:
I'd say so, thanks!

closed 2017-12-01 by pdp7