adafruit/adafruit-beaglebone-io-python #125: RuntimeError: Problem with the cape manager

acseckin, 2016-12-04:
Beaglebone Black, Debian 8.6, Python 2.7.9

python code:

```
import Adafruit_BBIO.PWM as PWM
PWM.start("P8_13", 50)
```

error:

```
PWM.start("P8_13", 50)
RuntimeError: Problem with the cape manager
```

pdp7, 2016-12-05:
@acseckin Please paste the contents of **/boot/uEnv.txt**

Adafruit_BBIO expects cape-universal to be enabled when running Linux kernel 4.1+.

Also please make sure you are running the latest version from PyPI:
**pip install Adafruit_BBIO**

acseckin, 2016-12-05:
uEnv.txt

```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.30-ti-r64
#uuid=
#dtb=

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
#dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
#dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

cmdline=coherent_pool=1M quiet cape_universal=enable

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M quiet cape_universal=enable video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

uuid=563dc38e-4bd3-427f-aa36-2f0c3ef1d9dc

cape_enable=bone_capemgr.enable_partno=BB-UART1,BB-UART4,BB-UART5
cape_disable=bone_capemgr.disable_partno=BB-UART2
```
slots

```
0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,BB-UART1
 5: P-O-L-   1 Override Board Name,00A0,Override Manuf,BB-UART4
 6: P-O-L-   2 Override Board Name,00A0,Override Manuf,BB-UART5
```

```
I root@beaglebone:/home/debian#  pip install Adafruit_BBIO
Requirement already satisfied: Adafruit_BBIO in /usr/local/lib/python2.7/dist-packages/Adafruit_BBIO-1.0.0-py2.7-linux-armv7l.egg
```

pdp7, 2016-12-05:
It would be helpful if you could run `strace` so that I can better understand what is happening.

Install `strace` (if it doesn't already exist):
```
sudo apt-get install strace
```

Then run `strace`, replace `test.py` with the name of your test program:
```
strace -f -o strace.log python test.py
```

Paste the contents of `strace.log` into a [gist](https://gist.github.com/).

acseckin, 2016-12-05:
How can i paste the whole file? is there any simple way?

acseckin, 2016-12-05:
[strace.log ](https://gist.github.com/acseckin/13f7cc4ff6f87377587184f3324f5e42)

acseckin, 2016-12-06:
I have a project deadline (9 Dec) . Can you suggest a qucik solution or a stable Beaglebone Black image to working UARTs and PWMs at the same time?

pdp7, 2016-12-06:
@acseckin  It seems that loading the UART overlays at boot causes cape-universal overlay not to load.  Adafruit_BBIO depends on cape-universal for Linux 4.1+ kernel (as used in Debian 8.x Jessie). 

I'd suggest commenting out the enable_partno and disable_partno lines:
```
#cape_enable=bone_capemgr.enable_partno=BB-UART1,BB-UART4,BB-UART5
#cape_disable=bone_capemgr.disable_partno=BB-UART2
```
Ensure that cape-universal is still enabled:
```
cmdline=coherent_pool=1M quiet cape_universal=enable
```
Reboot and slots should have:
```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
```

You should then be able to use `config-pin` to configure the pins instead of loading specific overlays.

Based on http://elinux.org/Beagleboard:Cape_Expansion_Headers, I'd suggested:
```
root@beaglebone:~# config-pin P9.11 uart
root@beaglebone:~# config-pin P9.13 uart
root@beaglebone:~# config-pin P9.24 uart
root@beaglebone:~# config-pin P9.26 uart
root@beaglebone:~# config-pin P9.42 uart
```

Additional details in of [beaglebone-universal-io README](https://github.com/cdsteinkuehler/beaglebone-universal-io) by @cdsteinkulher

acseckin, 2016-12-06:
P9.42 not a uart port according to https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/uart

`config-pin P9.42 pwm`

also uart 5 pins gives this

```
root@beaglebone:~# config-pin P8.38 uart
P8_38 pinmux file not found!
cape-universala overlay not found
run "config-pin overlay cape-universala" to load the cape
root@beaglebone:~# config-pin P8.38 uart

```

acseckin, 2016-12-06:
disabling HDMI fixes UART5 with 
```
config-pin P8.38 uart
config-pin P8.37 uart
```

and slots:

```
0: PF---- -1
1: PF---- -1
2: PF---- -1
3: PF---- -1
4: P-O-L- 0 Override Board Name,00A0,Override Manuf,univ-emmc
```
uEnv.txt : https://gist.github.com/acseckin/b37ae6502da3ffb5a6ee4d5ac9981ce0

**but p9_42 pwm still not working**

pdp7, 2016-12-07:
*Can you use a different pin for PWM?*

It appears that there is an issue with [P9_42 as it is an ECAP pin](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L192).  

Bonescript addressed this issue back in September:
[Fixed PWM for ECAP (P9_28 and P9_42)](https://github.com/jadonk/bonescript/commit/7fee6177def213b5e98e4992dfec01a03c200bbd)

I should be able to fix Adafruit_BBIO based that on, but it may quicker for you to just use another pin.

acseckin, 2016-12-07:
I'm ok with that pins. P9_42 is supernumerary for now.

pdp7, 2016-12-07:
I've created issue #126 to allow P9_42 to be used as PWM

acseckin, 2016-12-07:
I think we can close this issue.

closed 2016-12-07 by acseckin

DarioushK1, 2017-07-11:
Hi @pdp7 ,

I'm new to beaglebone black. I followed up your steps (&Robert Nelson) in building kernel 4.4.ti and now my encoder is working. Also I tried to use Adafruit Ultimate GPS and it is working right now. However every time I have to config-pin P8.11 qep, config-pin P8.12 qep, config-pin P9.24 uart and config-pin P9.26 uart before run my codes. Do you have any solution so that eqep2 and uart enable on boot automatically ?
Sorry if I ask stupid question. I'll appreciate if you can help me.

Best,
Darioush

pdp7, 2017-07-11:
@DarioushK1 please see this documentation for running config-pin at boot:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/doc/run_config-pin_during_startup.md

DarioushK1, 2017-07-11:
@pdp7 Thank you. it worked.

Darioush

suriab, 2018-03-04:
can we use config-pin to enable 1 wire or any suggestions on how to enable that.