adafruit/adafruit-beaglebone-io-python #127: unable to setup ADC

ManMan88, 2016-12-08:
- Platform/operating system:  BeagleBoneBlack with Ubuntu 16.04.1 LTS (GNU/Linux 4.4.34-ti-r69 armv7l)

- Python version : Python 2.7.12

- Error message: 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper
```

- List the steps to reproduce the problem: 
installing:
```
sudo ntpdate pool.ntp.org
sudo apt-get update
sudo apt-get install build-essential python-dev python-pip -y
git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
sudo python setup.py install
cd ..
sudo rm -rf adafruit-beaglebone-io-python
```

in python:

```
import Adafruit_BBIO.ADC as ADC
ADC.setup()
```

LudovicGuerra, 2016-12-08:
I agree. I add that for me, if I lauch it once connected by SSH, it is working. But if i lauch it by a service, it doesn't work. But when I launched it with a sleep 40 on the serivce, it worked. Need to find out the cause (I add also that once it worked, the UART started to bug saying typeError in the serialposix)

pdp7, 2016-12-09:
@ManMan88 Please paste the contents of `/boot/uEnv.txt` and `/sys/devices/platform/bone_capemgr/slots`.

Adafruit_BBIO expects cape-universal to be in use for Linux 4.1+ kernels.

Also, I noticed that you are using Ubuntu.  Where did the image you are using come from?

I only test Adafruit_BBIO with the official BeagleBoard.org Debian images created by @robertcnelson:
* [BeagleBoard.org Latest Firmware Images](http://beagleboard.org/latest-images)
* [Debian Image Testing Snapshots](http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Image_Testing_Snapshots)

The kernels are built using this repo:
https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-4.4.y

pdp7, 2016-12-09:
@LudovicGuerra It sounds like you are describing a different problem.  Please open up a new issue for it.

ManMan88, 2016-12-09:
@pdp7 I used the image from this link: https://rcn-ee.com/rootfs/2016-11-10/flasher/BBB-eMMC-flasher-ubuntu-16.04.1-console-armhf-2016-11-10-2gb.img.xz
It was given by: http://elinux.org/BeagleBoardUbuntu

The contents of  `/boot/uEnv.txt` is:
```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.34-ti-r69
##uuid=
#dtb=

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
#dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
#dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

cmdline=coherent_pool=1M quiet cape_universal=enable

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M quiet cape_universal=enable video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

uuid=f62d5c8b-d754-477f-8341-d1cccea1d3e5
```

And the contents of `/sys/devices/platform/bone_capemgr/slots` is: 
```
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
```

RobertCNelson, 2016-12-09:
@ManMan88 I've tried to keep the ubuntu target as close to the debian target for overlays, so this should still work on ubuntu.. (Sadly, other then booting, i really don't test ubuntu)

What is shown by:

```
dmesg | grep bone
```
after this occured:
```
raceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper
```

Regards,

pdp7, 2016-12-09:
@ManMan88 I believe the issue is that cape-universal overlay isn't loaded.

You can try to load with:
```
echo cape-universaln > /sys/devices/platform/bone_capemgr/slots
```

You should then see:
```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
```

If that fails, then please install the latest Device Tree overlays from:
https://github.com/beagleboard/bb.org-overlays

ManMan88, 2016-12-09:
@RobertCNelson, this is the output to what you asked me to do: 
```
[    2.400035] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,000C,1816BBBK2050'
[    2.400068] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    2.438234] bone_capemgr bone_capemgr: slot #0: No cape found
[    2.482229] bone_capemgr bone_capemgr: slot #1: No cape found
[    2.526225] bone_capemgr bone_capemgr: slot #2: No cape found
[    2.570225] bone_capemgr bone_capemgr: slot #3: No cape found
[    2.576126] bone_capemgr bone_capemgr: initialized OK.
```
@pdp7 , your first suggestion worked :) thanks! I very appreciate the fast help!

RobertCNelson, 2016-12-09:
@pdp7 just ran into this today, both the elinux.org images ubuntu/debian, lack "cape_universal=enable" option in /boot/uEnv.txt

/boot/uEnv.txt: (current)
```
cmdline="coherent_pool=1M quiet"
```

/boot/uEnv.txt: (needed)
```
cmdline="coherent_pool=1M quiet cape_universal=enable"
```

Regards,

pdp7, 2016-12-15:
@ManMan88 Is this issue resolved?

ManMan88, 2016-12-15:
Yes, it is resolved. Thanks.

I have another issue with PWM that might be similar to this problem. However, I'll post a new issue on it.

acseckin, 2017-06-28:
I have same problem when calling from crontab. [This code](https://github.com/acseckin/hmrid/blob/master/runhmrid.py) works from terminal but gives these results when calling from crontab.

```
Traceback (most recent call last):
  File "/home/debian/start.py", line 129, in <module>
    nhid=hmrid()
  File "/home/debian/start.py", line 22, in __init__
    self.setupPins()
  File "/home/debian/start.py", line 50, in setupPins
    ADC.setup()
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper
```

pdp7, 2017-06-29:
@acseckin I've been able to replicate this problem and am investigating.

pdp7, 2017-06-29:
I'm still investigating but I noticed that issue seems to be `/sys/bus/iio/devices/iio:device0/in_voltage1_raw` does not exist run the script is run at boot.  From strace log:
```
open("/sys/devices/platform/bone_capemgr/slots", O_RDWR|O_LARGEFILE) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=4096, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6
fef000
read(3, " 0: PF----  -1 \n 1: PF----  -1 \n"..., 4096) = 64
read(3, "", 4096)                       = 0
write(3, "BB-ADC", 6)                   = 6
close(3)                                = 0
munmap(0xb6fef000, 4096)                = 0
nanosleep({0, 200000000}, NULL)         = 0
open("/sys/bus/iio/devices/iio:device0/in_voltage1_raw", O_RDONLY) = -1 ENOENT
```

acseckin, 2017-06-29:
I think the uEnv.txt configurations begin little bit after from crontab. How can we wait for enough time?

pdp7, 2017-06-30:
@acseckin I added a delay of 1 second after loading the ADC overlay and that seems to have resolved the issue.  The script does now run OK at bootup.

Here is what I changed:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ git diff
diff --git a/source/c_adc.c b/source/c_adc.c
index b7f8f93..21ad284 100644
--- a/source/c_adc.c
+++ b/source/c_adc.c
@@ -57,6 +57,7 @@ BBIO_err initialize_adc(void)
     if (err == BBIO_OK) {
         strncat(adc_prefix_dir, "/sys/bus/iio/devices/iio:device0/in_voltage", sizeof(adc_prefix_dir));
         snprintf(test_path, sizeof(test_path), "%s%d_raw", adc_prefix_dir, 1);
+        sleep(1);
         fh = fopen(test_path, "r");
 
         if (!fh) {
```

For now, I've created a branch `adc-fix` which contains that change as commit 1997b53a26e1ca19eddad56c64c1dfb767838ce4.

I'm looking at if this delay should be moved to the `load_device_tree()`.

acseckin, 2017-06-30:
Here is my new message after` python /home/debian/init.py >/home/debian/cronlog.out 2>&1 &`

```
[sudo] password for debian: *** buffer overflow detected ***: python terminated
Configuration
Setup pins
Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper
Setup pins
```

pdp7, 2017-07-01:
@acseckin Could you paste the output of `uname -a`?  @RobertCNelson recently told me there was an ADC bug that could could cause the buffer overlow error and it was fixed in newer kernel.

Please run `/opt/scripts/tools/update_kernel.sh`

pdp7, 2017-07-01:
@acseckin Please run this to upgrade to the latest 4.9 kernel build by @RobertCNelson:
`/opt/scripts/tools/update_kernel.sh --lts-4_9`

This is the ADC fix that I referenced above:
https://github.com/RobertCNelson/ti-linux-kernel/commit/dca02651cee78260ad9b6f1dce8f40b28367b7e2

pdp7, 2017-07-01:
I've just tried `4.9.35-ti-r44` but unfortunately that does not resolve the issue:

```
debian@beaglebone:~$ cat cronlog.out 
*** buffer overflow detected ***: python terminated
Configuration
Setup pins
Unable to setup ADC system. Possible causes are: 
  - A cape with a conflicting pin mapping is loaded 
  - A device tree object is loaded that uses the same name for a fragment: helper
Setup pins
```

```
debian@beaglebone:~$ uname -a
Linux beaglebone 4.9.35-ti-r44 #1 SMP PREEMPT Fri Jun 30 22:17:16 UTC 2017 armv7l GNU/Linux
```

acseckin, 2017-07-01:
Same results for my settings.

pdp7, 2017-07-01:
@acseckin I've found that delay of 30 seconds resolves the issue.  Add  `time.sleep(30)` before `ADC.setup()` in `init.py`:
```diff
--- init.py.orig	2017-07-01 05:37:39.696002694 +0000
+++ init.py	2017-07-01 10:15:37.641031199 +0000
@@ -62,7 +62,15 @@
         #print self.rxoff,self.ryoff
     def setupPins(self):
         self.log2file("ADC")
+
+        print("setupPins: sleep")
+        time.sleep(30)
+        print("setupPins: call ADC.setup()")
         ADC.setup()
+        print("setupPins: ADC.setup() complete")
+        os.system("cat /sys/devices/platform/bone_capemgr/slots")
+        os.system("cat /sys/bus/iio/devices/iio:device0/in_voltage1_raw")
+
         self.log2file("digital pins")
         GPIO.setup("P9_11", GPIO.IN)
         GPIO.setup("P8_17", GPIO.IN)
@@ -93,7 +101,7 @@
                         for i in [5,4,3,2,1,0]:
                             self.lcddisp("Reboot:"+str(i))
                             time.sleep(1)
-                        os.system("sudo reboot")
+                        #os.system("sudo reboot")
                     elif GPIO.input("P8_17")|GPIO.input("P8_15")|GPIO.input("P8_11")|GPIO.input("P8_9"):
                         break
         elif GPIO.input("P8_17"):

```

acseckin, 2017-07-01:
Thank you a lot. Now it is working. Here are steps as a summary:
1- **run  update**
`/opt/scripts/tools/update_kernel.sh --lts-4_9`
2- **write code** (attached) with `time.sleep()` and 
```
os.system("cat /sys/devices/platform/bone_capemgr/slots")
os.system("cat /sys/bus/iio/devices/iio:device0/in_voltage1_raw")
```
3- **Edit crontab** of root with `sudo crontab -e -u root` and append this line 
`@reboot python /home/debian/init.py >/home/debian/cronlog.out 2>&1 &`
4- **reboot** and working fine.

pdp7, 2017-07-01:
@acseckin great to hear!  OK to close this issue?

acseckin, 2017-07-01:
Yes you can close.

closed 2017-07-03 by pdp7

acseckin, 2017-07-13:
Hi add this line to uEnv.txt also works

cape_enable=bone_capemgr.enable_partno=BB-UART2,BB-ADC