adafruit/adafruit-beaglebone-io-python #129: Doesn't support Ubuntu core beaglebone leds

elopio, 2016-12-27:
I have a beagle bone with ubuntu core installed from http://people.canonical.com/~ogra/snappy/all-snaps/daily/current/ubuntu-core-16-beaglebone.img.xz

The problem here is that the leds 0 and 1 are not called usr1 and usr3. They are called heartbeat and mmc0:

```
$ ls /sys/class/leds
beaglebone:green:heartbeat  beaglebone:green:mmc0  beaglebone:green:usr2  beaglebone:green:usr3
```

So, with the adafruit library I can only use the tow last leds.

I'm not sure what would be the best way to support this. Maybe in here:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/cf9771a101c540b126f50268e9ade202dc3c8e4e/source/event_gpio.c#L291
if the gpio 53 or 54, and the file file doesn't exist, you could try to set mmc0 and heartbeat instead.

RobertCNelson, 2016-12-27:
@elopio this project uses a lot of features in the beagleboard.org kernel fork, many of which are heading mainline (just slowly)..  With this issue, in 3.8.13, we had these named usr0 -> usr3, when the black went mailine, this detail was missed.. Here's the patch we use:

https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-4.9.y/patches/soc/ti/bone_common/0001-ARM-dts-am335x-bone-common-update-leds-to-match-3.8..patch

Regards,

elopio, 2016-12-27:
Hey @RobertCNelson! So, let me see if I understood correctly. The right name for all the leds is usrX, and at some point in the future that will be changed in the mainline kernel and consequently in Ubuntu when we sync. Right?

RobertCNelson, 2016-12-27:
@elopio 

It's not really a matter between right and wrong..

Case 1: old 3.8.13 software expects usr0 -> usr3 (using mainline breaks old 3.8.13 software)
Case 2: mainline has had it since 2013-09-17 (thus changing it breaks mainline users)

So I've elected to just patch it locally for the beagleboard.org community and leave the mainline version untouched, as it's been that default for too long to change.

Regards,

elopio, 2016-12-28:
I see. Thanks for the info.
So I'm still wondering, should this library support both cases by checking which files exist?

RobertCNelson, 2016-12-28:
It can be added to the library, but everything else still requires things start are not in mainline..
Regards,

elopio, 2016-12-28:
Got it. So even if we get the leds working, many other things will not work. 
I will talk to our bbb maintainer to see if we can apply your patches to make a bb kernel snap.

pdp7, 2017-01-04:
This issue seems to me like something that might be useful to support.  I've created a branch to add support for alternate USR LED naming: `issue129-usr-leds`

To test this, I've built a kernel from `ti-linux-4.4.y` branch of `ti-kernel-kernel-dev` with a modified `am335x-bone-common.dtsi`:
```
commit 8008f97b6c724b9f5429416a647219d1b3932889
Author: Drew Fustini <drew@pdp7.com>
Date:   Fri Dec 30 04:57:09 2016 -0600

    test

diff --git a/arch/arm/boot/dts/am335x-bone-common.dtsi b/arch/arm/boot/dts/am335x-bone-common.dtsi
index 1f496145..2c5e1f2 100644
--- a/arch/arm/boot/dts/am335x-bone-common.dtsi
+++ b/arch/arm/boot/dts/am335x-bone-common.dtsi
@@ -26,14 +26,14 @@
                compatible = "gpio-leds";
 
                led@2 {
-                       label = "beaglebone:green:usr0";
+                       label = "beaglebone:green:heartbeat";
                        gpios = <&gpio1 21 GPIO_ACTIVE_HIGH>;
                        linux,default-trigger = "heartbeat";
                        default-state = "off";
                };
 
                led@3 {
-                       label = "beaglebone:green:usr1";
+                       label = "beaglebone:green:mmc0";
                        gpios = <&gpio1 22 GPIO_ACTIVE_HIGH>;
                        linux,default-trigger = "mmc0";
                        default-state = "off";
```

pdp7, 2017-01-09:
@elopio I've just merged #130 which adds support for the alternate USR LED labels.  **Please test again in Ubuntu Core and let me know if the issue is resolved.**

The following `/sys/class/leds/` paths should all work now:

Normal USR LED labels:
* beaglebone:green:usr0
* beaglebone:green:usr1
* beaglebone:green:usr2
* beaglebone:green:usr3

Alternate labels:
* beaglebone:green:heartbeat
* beaglebone:green:mmc0
* beaglebone:green:cpu0
* beaglebone:green:mmc1

pdp7, 2017-01-12:
@elopio is it OK to close this issue now?

elopio, 2017-01-12:
Sorry, I haven't had time to give it a try. I think I can play with my bbb today.
The code looks good, thanks a lot for working on it.

pdp7, 2017-01-25:
@elopio Hi, just wanted to check if you had been able to test yet.  Thanks.

elopio, 2017-01-28:
Yes, it works :) Thanks a lot @pdp7, and sorry for the late reply.

pdp7, 2017-01-29:
@elopio awesome, thanks for letting us know

closed 2017-01-29 by pdp7