adafruit/adafruit-beaglebone-io-python #133: Compilation issues due to strict aliasing

archey, 2017-02-02:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  ArchLinux 64-bit

- Python version (run `python -version` or `python3 -version`):  python2

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

When compiling the adafruit beaglebone-io-python libraries on gcc 6.3.1, it fails due to strict aliasing rules. I was curious if these 3 lines could be corrected to fix the issue with compilation.

[Compiliation errors](https://gist.github.com/archey/484ba457c68448a3f001c8a52c629686)

pdp7, 2017-02-03:
@archey thanks for letting us know.  I'll take a look.

archey, 2017-02-07:
Hows this coming along? Just curious no rush

pdp7, 2017-02-07:
Thanks for the ping.  I'll take a look tonight and update.

pdp7, 2017-02-08:
@archey Are you running Arch on the BeagleBone itself?  If so, is there an image that I can download to test?

Currently, I'm downloading archlinux-2017.02.01-dual.iso to run in VirtualBox on my x86_64 host.  This seemed like the quickest way to get GCC 6.3. I'm hoping I'll hit the error you mentioned when I try to build adafruit-beaglebone-io-python on it.

pdp7, 2017-02-08:
I'm leaning towards modifying the build to ignore the gcc strict-aliasing warning.  Py_RETURN_TRUE
 and Py_RETURN_FALSE are part of the Python C API and I think are OK to use:
https://docs.python.org/2/c-api/bool.html

pdp7, 2017-02-08:
@archey I've pushed commit baf41fc which adds `-Wno-strict-aliasing` to `CFLAGS` to ignore gcc warning.  Please test and let me know how it works.

archey, 2017-02-08:
@pdp7 I am testing it on all arches x86_64, i686, armv6, armv7, aarch64. I was hoping you wouldn't add the ignore flag, but Im sure it will fix the issues. You can use ArchLinux ARM to test it on the beaglebone with this image: https://archlinuxarm.org/platforms/armv7/ti/beaglebone-black or https://archlinuxarm.org/platforms/armv7/ti/beaglebone

pdp7, 2017-02-08:
@archey thanks.  yeah, I don't like having to permanently ignore warnings.   `Py_RETURN_TRUE`
and `Py_RETURN_FALSE` seem to be necessary for the Python C API.  I'm not sure how to rework the code to avoid that strict-aliasing warning for those lines.

archey, 2017-02-08:
@pdp7 I understand, its not an issue in 1.0.0 the current version that we have compiled. I will test the cflag change in a bit.

pdp7, 2017-02-08:
@archey Ah, I think the reason is that I enabled extra gcc warnings in this commit: c8fe070

I think I didn't notice the strict aliasing warning as I'm on an older version of gcc.

archey, 2017-02-10:
@pdp7 it works fine with the patch, you can commit it and maybe bump the current tag

pdp7, 2017-02-10:
Thanks, I've created a new release.

closed 2017-02-10 by pdp7