adafruit/adafruit-beaglebone-io-python #136: PWM Initialization Produces RuntimeError: Problem with cape manager

elfgoh, 2017-03-09:
## Steps to reproduce the error (vauge details included due to expediency)

- Use a debian [console image](https://rcn-ee.com/rootfs/bb.org/testing/2017-02-12/console/) as a base
- Install python 3.4
- Create a venv
- Install pip using the [get-pip.py[(https://pip.pypa.io/en/stable/installing/)
- Activate venv
- Install adafruit BBIO library (1.0.3) using pip
- run the following script `# python3.4 bbio-test.py`
>#!/usr/bin/python3.4
import Adafruit_BBIO.PWM as PWM
PWM.start("P9_14",50)

## System details
- Platform: BeagleBone Black Industrial, Rev. C
- Kernel: 4.48-ti-r88
- Distro: Debian 8.7 [console image](https://rcn-ee.com/rootfs/bb.org/testing/2017-02-12/console/) 
Python Version: 3.4.2
Adafruit_BBIO version: 1.0.3
- Output from `# cat /sys/devices/platform/bone_capemgr/slots`
> 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1

pdp7, 2017-03-09:
@elfgoh Please create a gist with the output from dmesg after booting up.

I'll download that image and attempt to reproduce.

In the meantime, I recall that console images in the past didn't have overlays installed.  Please install [bb.org-overlays](https://github.com/beagleboard/bb.org-overlays):
```
git clone https://github.com/beagleboard/bb.org-overlays
cd ./bb.org-overlays
sudo ./install.sh
```

elfgoh, 2017-03-09:
Seems like the bb cape overlays are installed. Can I double check that they are the same one as bb.org overlays?

`# dpkg-query -l |grep overlay`
> ii  bb-cape-overlays                     4.1.20170304-0rcnee0~bpo80+20170305   armhf        Device tree overlays for Beaglebone.`

elfgoh, 2017-03-09:
This is the [gist](https://gist.github.com/elfgoh/5dc11cb9d3ba1ef985fa560e9d9a6ade) from `# dmesg` after a reboot

pdp7, 2017-03-09:
It looks like the bb-cape-overlays package is from 2017-03-04 so that should be find.

Thanks for the dmesg gist.

Please do the same for the contents of **/boot/uEnv.txt**.

elfgoh, 2017-03-09:
[Gist](https://gist.github.com/elfgoh/92c9c2f64dd3eaa7e515e0136d26f9ca) for `/boot/uEnv.txt`

elfgoh, 2017-03-09:
@pdp7 fwiw python3.4 was installed via apt-get and there isn't any other python versions afaik on the system

pdp7, 2017-03-09:
I'm not running the 2017-02-12 console image and I am observing the same behavior as you.

I think the issue here is that [cape-universal](https://github.com/cdsteinkuehler/beaglebone-universal-io) is not being loaded as expected.  Adafruit_BBIO requires cape-universal be loaded when the Bone is running a 4.1+ kernel (like on the Debian Jessie images).

I'm going to ask Robert Nelson who creates the images if this is a known issue with the console image.

**Have you tried the [Debian Jessie IoT image](https://rcn-ee.com/rootfs/bb.org/testing/2017-02-12/iot/)?**  It is smaller than the full image but much more capable than the console image.

pdp7, 2017-03-09:
I've just posted on the BeagleBoard mailing list to see what Robert and possibly others think of using cape-universal and the console image:
https://groups.google.com/d/msg/beagleboard/9aoq_ilz1J8/Sfbk0CdzAwAJ

elfgoh, 2017-03-09:
Yes I have tried the IoT image and the issue doesn't happen. I am trying to use an image that is as minimal. Afaik, the IoT image had some bits that I had to remove as I was aiming for a system with the smallest footprint. 

It seemed more productive to use the leaner console image as a starting image. But perhaps this approach isn't ideal. Happy to listen to differing perspectives!

RobertCNelson, 2017-03-09:
I've added config-pin to the overlay project:

https://github.com/beagleboard/bb.org-overlays/commit/79221773d290657807318a31c06e5cd7da87c81d

Regards,

pdp7, 2017-03-09:
Thanks @RobertCNelson.  Should the console image then also load a cape-universal overlay when it starts up?

RobertCNelson, 2017-03-09:
it should, the cape universal overlay loading happens in the boot script.

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L812

Regards,

RobertCNelson, 2017-03-09:
ps, i just "finally" added config-pin to the bb-cape-overlays package

https://github.com/rcn-ee/repos/commit/46941bb9e7e0e0593ea0425a16fe0d0733ea95dd

So end users who just want to use the overlays as is, can just install

sudo apt install bb-cape-overlays

(no more git/make/etc requried)..

Regards,

pdp7, 2017-03-09:
Thanks @RobertCNelson.

@elfgoh Could you try upgrading `bb-cape-overlays`?

elfgoh, 2017-03-10:
Thanks for the very quick response @RobertCNelson ad @pdp7 . I have upgraded and there doesnt seem to be any improvement unfortunately

`# pip list|grep BBIO`
> DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Adafruit-BBIO (1.0.3)

`# dpkg-query -l |grep overlay`
> ii  bb-cape-overlays                     4.1.20170309-0rcnee0~bpo80+20170309   armhf        Device tree overlays for Beaglebone.

`# python3.4 bbio-test.py`
> Traceback (most recent call last):
  File "bbio-test.py", line 4, in <module>
    PWM.start("P9_14",50)
RuntimeError: Problem with the cape manager

`# cat bbio-test.py`
> #!/usr/bin/python3.4
import Adafruit_BBIO.PWM as PWM

>PWM.start("P9_14",50)

elfgoh, 2017-03-10:
[Output](https://gist.github.com/elfgoh/1f71b30e48d59711584c885390d43e2a) from `# sudo strace -f -o trace.txt python3.4`

> Python 3.4.2 (default, Oct  8 2014, 14:38:51)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from Adafruit_BBIO import PWM
>>> PWM.start('P8_13', 50, 1000, 0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Problem with the cape manager
>>>

elfgoh, 2017-03-10:
`dmesg` [output](https://gist.github.com/elfgoh/b1c062b04b70f3e2793ebd2bf2229cb0) after a fresh reboot

elfgoh, 2017-03-10:
Seems like still no cape-universal overlay appears in slots:

`# cat /sys/devices/platform/bone_capemgr/slots `
 > 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1

elfgoh, 2017-03-10:
`config-pin` has also been installed and runs without issues

RobertCNelson, 2017-03-10:
@elfgoh 

open /opt/scripts/boot/am335x_evm.sh

make sure it matches:

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L811-L817

or just run "git pull" in /opt/scripts and get the full latest..

Regards,

elfgoh, 2017-03-10:
Looks like it is fixed @RobertCNelson 

The current `/opt/scripts/boot/am335x_evm.sh` needs to be updated. Is there any timeline for the updates to go into the console image? If not, I am happy to just git pull for the time being

Thanks a lot @RobertCNelson and @pdp7

RobertCNelson, 2017-03-10:
@elfgoh so the /opt/scripts/ directory has never been packaged.  When we generated a weekly snapshot the top of the master branch is always cloned in.

Regards,

elfgoh, 2017-03-10:
I see. So I guess I should wait for the next [snapshot image](https://rcn-ee.com/rootfs/bb.org/testing/) if I wished to have the fix included by default?

elfgoh, 2017-03-10:
Slightly off topic: but the [latest snapshot on 7 Mar](https://rcn-ee.com/rootfs/bb.org/testing/2017-03-07/) does not have passwordless root login enabled

ie `ssh root@192.169.7.2` results in a password prompt

pdp7, 2017-03-10:
@elfgoh This is the recent change that @RobertCNelson made to improve security:
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#i_take_full_responsibility_for_knowing_my_beagle_is_now_insecure

pdp7, 2017-03-10:
@elfgoh  I was able to use Adafruit_BBIO successfully with the `bone-debian-8.7-console-armhf-2017-02-12-2gb.img` image with these steps:
```
sudo apt-get update
sudo apt-get install build-essential git-core device-tree-compiler
git clone https://github.com/beagleboard/bb.org-overlays/
cd ./bb.org-overlays/
./install.sh
cd /opt/scripts/
git pull 
reboot
```
The `cape-universaln` overlay is now loaded:
```
root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln
```
Then run:
```
apt-get install python
apt-get install python-setuptools
sudo apt-get install python-pip
sudo apt-get install python-dev python3-dev
pip install Adafruit_BBIO
python bbio-test.py
```

elfgoh, 2017-03-10:
Thanks. Using the same image and slightly different configuration, `cape-universal` has been loaded by running `git pull` in `/opt/scripts` and get the full latest of https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh

I confirm that this issue can be closed at the Adafruit_BBIO end

closed 2017-03-10 by pdp7

wgaonar, 2017-03-27:
Hello @pdp7 and @RobertCNelson, 
I have been having this issue with the BeagleBone Green Wireless (BBGW). The curious aspect is that I am been able to run the PWM only when I am connected through the WiFi to the BBGW, in the other cases I get the runtime error. 
I found a way to solve this in the mean time. It consist of editing the /opt/scripts/boot/am335x_evm.sh file in order to force to load the cape-universal overlay in this way:				
TI_AM335x_BeagleBone_Green_Wireless)
       if [ -f /usr/local/lib/node_modules/node-red-node-beaglebone/.bbgw-dont-load ] ; then
		overlay="univ-bbgw"
        else
		overlay="univ-bbgw"
	fi

My question here is: What would be the consequences of doing that?
Thanks for your time.