adafruit/adafruit-beaglebone-io-python #137: Access to GPIO and PWM as non-root user not functioning

nicnicknicky, 2017-03-14:
Hello, I encountered several issues when trying to use the GPIO and PWM as a non-root user. The solutions I have seen in other similar issues have not used any concrete solutions and most were for kernels 3.xx.

Right before posting this issue, I managed to make a scrappy solution enable the GPIOs with PWM to work when my ownership is changed to non-root user. This solution was based off reading various threads and I pieced together my own which happened to work.  
This was my [main reference](https://github.com/cnobile2012/RobotControl/tree/master/contrib).
Essentially my contribution was that I went to look at how gpio and pwm were called in the different libraries and implemented permission changes on those files.

My kernel version:
```sh
#uname -a
Linux BeagleBoneBlack 4.4.36-ti-r72 #1 SMP Wed Dec 7 22:29:53 UTC 2016 armv7l GNU/Linux
```
I am still new to quite new to linux so can I verify if I am actually doing the right thing to get it to work?

I have a user *nick* which is part of the group *digital*

Create custom udev rule in `/etc/udev/rules.d/80-gpio-permissions.rules`
 ```
KERNEL=="gpio*", SUBSYSTEM=="gpio", ACTION=="add", PROGRAM="/usr/local/bin/udev-gpio-permissions.sh"
```

Create script in `/usr/local/bin/udev-gpio-permissions.sh`

```sh
  #!/bin/bash
  
  # Result: chown nick:digital /sys/class/gpio
  
  chown -R nick:digital /sys/devices/gpio
  chown -R nick:digital /sys/class/gpio
  chown -R nick:digital /sys/devices/platform/ocp
  chown -R nick:digital /sys/class/pwm
  
  #GPIO ( active_low device direction edge power subsystem uevent value )
  find /sys/class/gpio -type d -exec chmod 2770 {} \;
  find /sys/class/gpio -name "direction" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "edge" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "value" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "active_low" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "device" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "power" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "subsystem" -exec chmod 0660 {} \;
  
  #PWM ( duty_cycle enable period polarity power uevent )
  find /sys/devices/platform/ocp -name "duty_cycle" -exec chmod 0660 {} \;
  find /sys/devices/platform/ocp -name "enable" -exec chmod 0660 {} \;
  find /sys/devices/platform/ocp -name "period" -exec chmod 0660 {} \;
  find /sys/devices/platform/ocp -name "polarity" -exec chmod 0660 {} \;
  find /sys/devices/platform/ocp -name "power" -exec chmod 0660 {} \;
  find /sys/class/pwm -name "duty_cycle" -exec chmod 0660 {} \;
  find /sys/class/pwm -name "enable" -exec chmod 0660 {} \;
  find /sys/class/pwm -name "period" -exec chmod 0660 {} \;
  find /sys/class/pwm -name "polarity" -exec chmod 0660 {} \;
  find /sys/class/pwm -name "power" -exec chmod 0660 {} \;
  
  chmod 0220 /sys/class/gpio/export
  chmod 0220 /sys/class/gpio/unexport
  chmod 0220 /sys/class/pwm/export
  chmod 0220 /sys/class/pwm/unexport
  
  find /sys/class/gpio -name "uevent" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "autosuspend_delay_ms" -exec chmod 0660 {} \;
  find /sys/class/gpio -name "control" -exec chmod 0660 {} \;
  find /sys/class/pwm -name "uevent" -exec chmod 0660 {} \;
  find /sys/devices/platform/ocp -name "uevent" -exec chmod 0660 {} \;
  
  # Additional code for getting AIN pins set up
  ain_activator=/sys/devices/platform/bone_capemgr/slots
  chown -R nick:digital $ain_activator/
  chmod -R 2770 $ain_activator/
  ```

Ensure the above file is executable
  ```
  ls -al /usr/local/bin/udev-gpio-permissions.sh
  -rwxr--r-- 1 root staff 896 Jan 31 11:23 /usr/local/bin/udev-gpio-permissions.sh
  ```
Run `udevadm control --reload-rules`

Reboot and test!

pdp7, 2017-03-18:
@nicnicknicky thanks for posting.  I've just returned from a conference this past week and will check this out

pdp7, 2017-04-04:
@nicnicknicky thank you, I've tested your udev rule and script and added to this repo in a new `udev` directory

closed 2017-04-05 by pdp7

troywweber7, 2017-09-28:
I think the real solution is actually a part of this issue: https://github.com/beagleboard/linux/issues/124

dplanella, 2017-11-17:
This is a closed issue, but for those landing here and thinking the workaround works, I found out that at least for a 4.4.x kernel it doesn't for PWM. It seems that the PWM device tree attributes (`duty_cycle` & co.) are only created as files on demand. Thus the commands along these lines at boot time:

```
find /sys/devices/platform/ocp -name "duty_cycle"
```
return no results, as the `duty_cycle` file is not there. I don't actually know what creates it, but I noticed that it only gets created after an app using the PWM module is run.

RobertCNelson, 2017-11-17:
it's created after you export 0 (or 1) in the pwm system node.  only v4.11.x+ has the udev patch for permissions. (didn't want to break older userspace, we still need to fix bonescript before i backport that patch to v4.4.x/v4.9.x)

Regards,

dplanella, 2017-11-17:
Aha, thanks @RobertCNelson! So I guess in the meantime the bit to export 0 should be added to the workaround script for it to work.

dplanella, 2017-11-17:
@RobertCNelson would a similar patch be required for the quadrature encoder inputs? I wrote a udev rule based on the existing `81-gpio-noroot.rules`, but seems to have had no effect:

```
# /etc/udev/rules.d/83-eqep-noroot.rules
#
# Corrects sys eQEP permissions on the Beaglebone so non-root users in the eqep group can set up 
# the quadrature encoder inputs
#
# Change group to eqep
SUBSYSTEM=="eqep", PROGRAM="/bin/sh -c '/bin/chown -R root:eqep /sys/devices/platform/ocp/*.epwmss/*.eqep'"
# Change user permissions to ensure user and group have read/write permissions
SUBSYSTEM=="eqep", PROGRAM="/bin/sh -c '/bin/chmod -R ug+rw /sys/devices/platform/ocp/*.epwmss/*.eqep'"
```
I created a new group, `eqep`, then reloaded the udev rules:

```
$ sudo udevadm control --reload
```

And I finally rebooted, but the ownership/permissions didn't get changed after reboot.

That's on a 4.4.x kernel:

```
$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[a5d5f437baeaa7f8e14d5ae037268abf7a67e88e]
eeprom:[A335BNLT00C03614BBBK0298]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.4.91-ti-r133]
nodejs:[v6.12.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171107.0-0rcnee1~stretch+20171107]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```