adafruit/adafruit-beaglebone-io-python #141: Inconsistency in fd_lookup

ltjax, 2017-03-21:
fd_lookup in event_gpio.c uses 0 as the return value for "not found", but that's a valid return value for open(), albeit usually used for stdout.
Probably just an oddity, since opeing the FD with index 0 is very unlikely.

pdp7, 2017-03-24:
@ltjax thanks for pointing this out.  I'll take a look.

pdp7, 2017-04-12:
I've been tracing the code to see how it behaves.

Here is test python program:
```
#debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat ~/test.py 
import Adafruit_BBIO.GPIO as GPIO
from time import sleep

GPIO.setup("P8_14", GPIO.IN)
sleep(1);
if GPIO.input("P8_14"):
  print("HIGH")
else:
  print("LOW")
sleep(1);

print "####### NEXT UP: 17 #######"

GPIO.setup("P8_17", GPIO.IN)
sleep(2);
if GPIO.input("P8_17"):
  print("HIGH")
else:
  print("LOW")
sleep(1)

print "####### REPEAT: 17 #######"
if GPIO.input("P8_17"):
  print("HIGH")
else:
  print("LOW")
sleep(1)

print "####### NEXT #######"
GPIO.setup("P8_20", GPIO.IN)
sleep(1)
if GPIO.input("P8_20"):
  print("HIGH")
else:
  print("LOW")
sleep(1)

print "####### NEXT #######"
GPIO.setup("P8_24", GPIO.IN)
sleep(2);
if GPIO.input("P8_24"):
  print("HIGH")
else:
  print("LOW")

sleep(2)
print "####### REPEAT: 17 #######"

sleep(2);
if GPIO.input("P8_17"):
  print("HIGH")
else:
  print("LOW")

sleep(2);

sleep(2);
print "####### NEXT #######"

GPIO.setup("P8_28", GPIO.IN)
sleep(2);
if GPIO.input("P8_28"):
  print("HIGH")
else:
  print("LOW")
[... 165 lines cut ...]
fd_lookup(): ---------------------
fd_lookup(): f=0xe424d0
fd_lookup(): f->fd=5
fd_lookup(): f->gpio=27
fd_lookup(): f->initial=1
fd_lookup(): f->is_evented=0
fd_lookup(): f->next=0xed82d8
fd_lookup(): ---------------------
fd_lookup(): f=0xed82d8
fd_lookup(): f->fd=4
fd_lookup(): f->gpio=26
fd_lookup(): f->initial=1
fd_lookup(): f->is_evented=0
fd_lookup(): f->next=(nil)
fd_lookup(): NO MATCH
fd_lookup(): return 0
FUNCTION ENTER: open_value_file()
open_value_file(): gpio=88
open_value_file(): filename=/sys/class/gpio/gpio88/value
open_value_file(): add_fd_list(gpio=88, fd=8)
open_value_file(): return fd=8
LOW
####### NEXT #######
LOW
```

pdp7, 2017-04-12:
It would appear that in cases where `fd_lookup()` returns `0` that `open_value_file()` will then be called.  It will attempt to open the gpio value file and will call `add_fd_list()`.  The return value of `0` is not used as file descriptor (which would be stdout).

closed 2017-04-12 by pdp7