adafruit/adafruit-beaglebone-io-python #142: gpio_set_value leaks file descriptors for LEDs

ltjax, 2017-03-21:
In event_gpio.c: if the first open() in gpio_set_value succeeds (the one in the LED branch), the fd will be overwritten by the second open, hence leak.

pdp7, 2017-03-24:
@ltjax was this addressed by either of the Pull Requests?  thanks.

ltjax, 2017-03-24:
No, I just found it while looking for my problems. I never hit this case

pdp7, 2017-04-05:
fyi - I've used this [python program](https://gist.github.com/pdp7/e7753faff9a9bd27389a0173c71f9312) to test:
```
import Adafruit_BBIO.GPIO as GPIO
import time

for i in range(4):
    GPIO.setup("USR%d" % i, GPIO.OUT)
    GPIO.output("USR%d" % i, GPIO.HIGH)
    GPIO.output("USR%d" % i, GPIO.LOW)

for i in [3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46]:
    GPIO.setup("P8_%d" % i, GPIO.OUT)
    GPIO.output("P8_%d" % i, GPIO.HIGH)
    GPIO.output("P8_%d" % i, GPIO.LOW)

for i in [11,12,13,14,15,16,17,18,21,22,23,24,25,26,27,28,29,30,31,41,42]:
    GPIO.setup("P9_%d" % i, GPIO.OUT)
    GPIO.output("P9_%d" % i, GPIO.HIGH)
    GPIO.output("P9_%d" % i, GPIO.LOW)
```

pdp7, 2017-04-05:
@ltjax thanks, I've fixed the leak

closed 2017-04-05 by pdp7