adafruit/adafruit-beaglebone-io-python #144: stretch: manual build conflicting types for 'ctrl_dir/ocp_dir'

RobertCNelson, 2017-04-12:
Just caught this is on the builders this morning, stretch-iot image:

```
Cloning into '/opt/source/adafruit-beaglebone-io-python'...
remote: Counting objects: 90, done.
remote: Compressing objects: 100% (75/75), done.
remote: Total 90 (delta 18), reused 48 (delta 9), pack-reused 0
Unpacking objects: 100% (90/90), done.
overlays/ADAFRUIT-SPI0-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
overlays/ADAFRUIT-SPI0-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
overlays/ADAFRUIT-SPI1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
overlays/ADAFRUIT-SPI1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
overlays/ADAFRUIT-UART1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
overlays/ADAFRUIT-UART1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
overlays/ADAFRUIT-UART2-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
overlays/ADAFRUIT-UART2-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
overlays/ADAFRUIT-UART4-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
overlays/ADAFRUIT-UART4-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
overlays/ADAFRUIT-UART5-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
overlays/ADAFRUIT-UART5-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
running install
running bdist_egg
running egg_info
creating Adafruit_BBIO.egg-info
writing Adafruit_BBIO.egg-info/PKG-INFO
writing top-level names to Adafruit_BBIO.egg-info/top_level.txt
writing dependency_links to Adafruit_BBIO.egg-info/dependency_links.txt
writing manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
reading manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
copying Adafruit_I2C.py -> build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/Adafruit_BBIO
copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
creating build/lib.linux-armv7l-2.7/overlays
copying overlays/__init__.py -> build/lib.linux-armv7l-2.7/overlays
copying overlays/builder.py -> build/lib.linux-armv7l-2.7/overlays
running build_ext
building 'Adafruit_BBIO.GPIO' extension
creating build/temp.linux-armv7l-2.7
creating build/temp.linux-armv7l-2.7/source
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-sw1gMG/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Isource/include/ -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv7l-2.7/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-sw1gMG/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Isource/include/ -I/usr/include/python2.7 -c source/event_gpio.c -o build/temp.linux-armv7l-2.7/source/event_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-sw1gMG/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Isource/include/ -I/usr/include/python2.7 -c source/c_pinmux.c -o build/temp.linux-armv7l-2.7/source/c_pinmux.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-sw1gMG/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Isource/include/ -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-sw1gMG/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Isource/include/ -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
source/common.c:59:6: error: conflicting types for 'ctrl_dir'
 char ctrl_dir[43];
      ^~~~~~~~
In file included from source/common.c:46:0:
source/common.h:71:15: note: previous declaration of 'ctrl_dir' was here
   extern char ctrl_dir[35];
               ^~~~~~~~
source/common.c:60:6: error: conflicting types for 'ocp_dir'
 char ocp_dir[33];
      ^~~~~~~
In file included from source/common.c:46:0:
source/common.h:72:15: note: previous declaration of 'ocp_dir' was here
   extern char ocp_dir[25];
               ^~~~~~~
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
Log: umount: [/var/www/html/ssd/git/omap-image-builder/ignore/tmp.w3cUR5GFBw/dev/pts]
Log: umount: [/var/www/html/ssd/git/omap-image-builder/ignore/tmp.w3cUR5GFBw/proc]
Log: umount: [/var/www/html/ssd/git/omap-image-builder/ignore/tmp.w3cUR5GFBw/sys]
Log: umount: [/var/www/html/ssd/git/omap-image-builder/ignore/tmp.w3cUR5GFBw/run]
```

pdp7, 2017-04-12:
@RobertCNelson thanks, I'll check it out

RobertCNelson, 2017-04-12:
@pdp7 ahh i see the issue:

```
voodoo@qcom:~/git/omap-image-builder$ uname -r
3.4.0-14.10-cm-qs600-3.1-g075a761
```
That's my local quad core armhf build box..

it's tripping on:

https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L48-L64

i need to overide that with a flag. ;)

Regards,

RobertCNelson, 2017-04-12:
Working around it locally:

https://github.com/RobertCNelson/omap-image-builder/commit/eea039963af2c37536193860a2a24e8407a7b00f

Regards,

pdp7, 2017-04-13:
@RobertCNelson I'm confused why it necessary for your script to change '4.1.0' to '3.4.0' in this part of setup.py:
```
if kernel >= '4.1.0':
    kernel41 = [('BBBVERSION41', None)]
else:
    kernel41 = None
```
I would have thought the `#ifdef BBBVERSION41` in `source/common.c` and `source/common.h` would have prevented this from being an issue.

pdp7, 2017-04-13:
@cocasema It seems that `common.h` changed in ceb4681 _(Add libadafruit-bbio with C++ wrappers for PWM/GPIO)_ to define these variables as `extern`: 
```
-int gpio_mode;
-int gpio_direction[120];
+extern int gpio_mode;
+extern int gpio_direction[120];
 
 #ifdef BBBVERSION41
-    char ctrl_dir[43];
-    char ocp_dir[33];
+  extern char ctrl_dir[43];
+  extern char ocp_dir[33];
 #else
-    char ctrl_dir[35];
-    char ocp_dir[25];
+  extern char ctrl_dir[35];
+  extern char ocp_dir[25];
 #endif
```

Do you think this could be the reason for the gcc error that @RobertCNelson encountered on his build system running Linux kernel 3.4.0?
```
source/common.c:59:6: error: conflicting types for 'ctrl_dir'
 char ctrl_dir[43];
      ^~~~~~~~
In file included from source/common.c:46:0:
source/common.h:71:15: note: previous declaration of 'ctrl_dir' was here
   extern char ctrl_dir[35];
               ^~~~~~~~
source/common.c:60:6: error: conflicting types for 'ocp_dir'
 char ocp_dir[33];
      ^~~~~~~
In file included from source/common.c:46:0:
source/common.h:72:15: note: previous declaration of 'ocp_dir' was here
   extern char ocp_dir[25];
               ^~~~~~~
```

RobertCNelson, 2017-04-13:
@pdp7 & @cocasema so setup.py does a pre-build time check to decide weather to build for a 3.8.x or 4.1.x+ kernel.. 

```
if kernel >= '4.1.0':
    kernel41 = [('BBBVERSION41', None)]
else:
    kernel41 = None
```

My issue, one of my build boxes is running it's manufacture's kernel based on 3.4.0. So it went down the old "3.8.x" build path and discovered:
```
char ctrl_dir[43]; = extern char ctrl_dir[35];
```
and
```
char ocp_dir[33]; = extern char ocp_dir[25];
```
So the old 3.8.x build path is now actually broken, we just need to add this to the mis configured extern's...
``
#ifdef BBBVERSION41
extern char ctrl_dir[43];
extern char ocp_dir[33];
#else
extern char ctrl_dir[35];
extern char ocp_dir[25];
```
Regards,

cocasema, 2017-04-14:
The change to declare arrays as external in header was to fix "multiple
definitions of symbol" error. That should not affect logic.

cocasema, 2017-04-14:
Just got to a laptop.

Yeah, I see the issue. In common.c we have this code and we might end up with define of  BBBVERSION41 in .c file only.
```
#include <linux/version.h>
#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0)
#  ifndef BBBVERSION41
#    define BBBVERSION41
#  endif
#endif
```

pdp7, 2017-04-14:
@cocasema should that be add to the header file as well?

cocasema, 2017-04-15:
I not sure how many other this might break - common.h is included everywhere.
I'd first try moving arrays above the `#define` common.c to sync definition and declaration.
I'm out till Monday and won't be able to make/test this change sooner.

```
#include "common.h"

#ifdef BBBVERSION41
char ctrl_dir[43];
char ocp_dir[33];
#else
char ctrl_dir[35];
char ocp_dir[25];
#endif

#include <linux/version.h>
#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0)
#  ifndef BBBVERSION41
#    define BBBVERSION41
#  endif
#endif

int gpio_mode;
int gpio_direction[120];

/* move form here
#ifdef BBBVERSION41
char ctrl_dir[43];
char ocp_dir[33];
#else
char ctrl_dir[35];
char ocp_dir[25];
#endif
*/
```

pdp7, 2017-04-15:
@cocasema ok thanks.  @RobertCNelson has worked around this issue on his build server, so I don't believe this is a critical issue.  However, it will be good to get this resolved in case anyone else tries to build on an older system.

I did get the idea looking at the code above that maybe the conditional logic is not needed.  The arrays should just always be defined for the larger BBBVERSION41 size:
```
char ctrl_dir[43];
char ocp_dir[33];
```

@cocasema do you think there would be an issue with that approach?

cocasema, 2017-04-18:
> I did get the idea looking at the code above that maybe the conditional logic is not needed. The arrays should just always be defined for the larger BBBVERSION41 size:
```
char ctrl_dir[43];
char ocp_dir[33];
```

@pdp7 If extra size is not a problem - then this is the easiest fix.

pdp7, 2017-04-19:
@cocasema thanks for your insight.  I'm going remove the conditional logic for kernel version and use the larger size of array allocation.

closed 2017-04-19 by pdp7

cocasema, 2017-04-19:
We can also remove ctrl_dir: https://github.com/cocasema/adafruit-beaglebone-io-python/commit/2d7866cfc9bbe64579bec6d7005ac2d457d9a599

pdp7, 2017-04-19:
@cocasema could you please raise a pull request for https://github.com/cocasema/adafruit-beaglebone-io-python/commit/2d7866cfc9bbe64579bec6d7005ac2d457d9a599 ?

commit 01f73c57 (2017-04-19): remove conditional logic for ctrl_dir and ocp_dir size

Remove conditional logic for kernel version to determine the
size of ctrl_dir and ocp_dir.  Instead, the maximum possible
size is used for these arrays regardless of the kernel version

Resolves #144