adafruit/adafruit-beaglebone-io-python #150: Terminating one python/Adafruit_BBIO process, kills all other python/Adafuit_BBIO processes

fragrans, 2017-07-09:
I want to use different python/Adafruit_BBIO scripts to control different board IOs.
I have two scripts, and I can start them well, one by one.
But when I terminate one of the scripts with CTRL+C; the other will die silently too. 
 - os-release
PRETTY_NAME="Debian GNU/Linux 7 (wheezy)"
NAME="Debian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support/"
BUG_REPORT_URL="http://bugs.debian.org/"

- python version: 2.7.3

Then how can I use python/Adafruit_BBIO with multiple process independently?

pdp7, 2017-07-10:
Could you please post the source code for the two Python program that you are running?

Also, I'd like to know more about what is installed on your BeagleBone.  Please paste the output of:
`uname -a`
`cat /etc/debian_version`
`cat /etc/dogtag`
`cat /boot/uEnv.txt`
`cat /proc/cmdline`
`cat /sys/devices/platform/bone_capemgr/slots`

fragrans, 2017-07-14:
Sorry for this late reply; I flashed the Black board with debian8; The problem disappeared. It all just works like a charm. 
Thanks anyway.

closed 2017-07-15 by pdp7