adafruit/adafruit-beaglebone-io-python #151: BeagleBone Blue naming convensions

MarkAYoder, 2017-07-28:
Is anyone updating the Adafruit Python library for the Blue? 

[2] doesn't appear to have any definitions for the Blue's connectors. 

For example, GPIO1_17 appears on the Black at P9_23.  On the Blue it appears on pin 4 for the GP0 header[3].

Should this be referred to as GP0_4 since it's pin 4, or should it be GP0_2 since it's the second IO pin on the header, or maybe GP0_1 is we should start numbering with 0?

Another example, The analog in's appear on the ADC header.

Should AIN0 be referred to as AIN0, or ADC_3, or ADC_1, or ADC_0?  I can justify every choice.

My suggestion is to make it as easy as possible for the beginner and stick to the schematic and silk screen.

For example:
GP0_0    GPIO1_25
GP0_1    GPIO1_17
GP1_0    GPIO3_2
UT0_RX   UART0_RX
S1_MOSI  SPI1_MOSI

ADC0     AIN0
AIN0     AIN0  (both refer to AIN0)

What do you think?  These should also apply to BoneScript.

--Mark

[2] https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c
[3] https://github.com/beagleboard/beaglebone-blue/raw/master/BeagleBone_Blue_sch.pdf  (page 1, bottom middle)

pdp7, 2017-08-06:
@MarkAYoder Thanks for creating this issue.  Unfortunately, I'm traveling right now, but I will review and respond when I'm back home on Thursday (Aug 10).

pdp7, 2017-08-12:
@MarkAYoder I'm working on resolving a regression in PR #152 and will review this once I have resolve that issue.

pdp7, 2017-08-23:
@MarkAYoder is there a reason that you switch the default build to python3 in commit https://github.com/adafruit/adafruit-beaglebone-io-python/pull/153/commits/b01fb01196ddcd5ec71d923ad83d1fab594f1f2d ?

pdp7, 2017-08-23:
@MarkAYoder ok, no problem.  I'll just revert the change to the build default in 
 https://github.com/adafruit/adafruit-beaglebone-io-python/commit/b01fb01196ddcd5ec71d923ad83d1fab594f1f2d

pdp7, 2017-08-23:
I'm actually going to leave the Makefile as is per b01fb01.  I don't think the change of "make all" to python3 should cause any problems.

closed 2017-08-23 by pdp7

DarioushK1, 2018-02-28:
Hi, I'm kinda new to beaglebones. I previously used Adafriut_BBIO.pwm for controlling servo motors in beaglebone black, now I'm looking for something similar so I can control servo motors in beaglebone blue. Does Adafriut_BBIO work for the blue?

Thank you.
Darioush

DarioushK1, 2018-02-28:
I use subprocess.call to run rc_test_servos in my python code to control servo motors. but the problem is that I can not exit the code until I press ctrl+c. I want rc_test_servos to run multiple times in my code but nothing can exit the rc_test_servos in python (even sys.exit(), subprocess.kill(), subprocess.terminate() and so on). I will appreciate if anyone can help me.