adafruit/adafruit-beaglebone-io-python #154: Installation of Adafruit_BBIO fails

elfgoh, 2017-08-16:
If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **Linux beaglebone 4.4.59-ti-r96 on Debian Jessie, based off console image `bone-debian-8.7-console-armhf-2017-03-19-2gb.img`**

- Python version (run `python -version` or `python3 -version`):  **3.4.2**

- Error message you are receiving, including any Python exception traces: 
```
 Running setup.py (path:/tmp/pip-build-630pjdxi/Adafruit-BBIO/setup.py) egg_info for package Adafruit-BBIO
    overlays/ADAFRUIT-SPI0-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
    overlays/ADAFRUIT-SPI0-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
    overlays/ADAFRUIT-SPI1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
    overlays/ADAFRUIT-SPI1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
    overlays/ADAFRUIT-UART1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
    overlays/ADAFRUIT-UART1-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
    overlays/ADAFRUIT-UART2-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
    overlays/ADAFRUIT-UART2-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
    overlays/ADAFRUIT-UART4-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
    overlays/ADAFRUIT-UART4-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
    overlays/ADAFRUIT-UART5-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
    overlays/ADAFRUIT-UART5-00A0.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
    
Downloading/unpacking 1
  Could not find any downloads that satisfy the requirement 1
Cleaning up...
No distributions at all found for 1
Storing debug log for failure in /root/.pip/pip.log
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

```
pip3 install Adafruit_BBIO
```

pdp7, 2017-08-16:
@elfgoh thanks for reporting.  I'll attempt to reproduce this and determine the cause of the issue.

elfgoh, 2017-08-17:
Sorry I am closing this issue as this was an error on my part

closed 2017-08-17 by elfgoh

pdp7, 2017-08-17:
thanks for following up