adafruit/adafruit-beaglebone-io-python #155: Enhancement: SHT1x support

PeteLawler, 2017-08-21:
Doug Rohm has a [Raspberry Pi SHT1x library](https://github.com/drohm/pi-sht1x) and I'm looking at porting it. If someone beats me to it, I'm not going to complain :)

So far, I've got a fork [over on Gitlab](https://gitlab.com/petelawler/plaw-bbb-sht1x) but it just struck me now that instead of being a separate library it should probably be done inside the adafruit beaglebone library itself.
On the other hand, this is a unique protocol it's talking, so is that what you really want with your IO library? Anyway... if someone wants to hack it in and get it happening faster than me ;)

Thoughts, comments?

pdp7, 2017-08-21:
@PeteLawler  Thanks for bringing this up.

I'm not opposed to adding support directly in this repo, but I think it might be as it's own repo.  I think =[Adafruit_Python_DHT](https://github.com/adafruit/Adafruit_Python_DHT) would be a good model for it:
> Python library to read the DHT series of humidity and temperature sensors on a Raspberry Pi or Beaglebone Black.

@tdicola what do you think?

PeteLawler, 2017-08-21:
@pdp7 ah hmm that certainly makes more sense, at least modelling after that library if not having a combined DHT/SHT.

closed 2017-08-21 by PeteLawler