adafruit/adafruit-beaglebone-io-python #157: Support new GPIO character device interfce

pdp7, 2017-08-23:
Linux kernel gpio subsystem maintainer Linus Walleij presented GPIO descriptors in this presentation:
* Video: [GPIO for Engineers and Makers](https://www.youtube.com/watch?v=lQRCDl0tFiQ)
* Slides: [GPIO for Engineers and Makers](http://elinux.org/images/9/9b/GPIO_for_Engineers_and_Makers.pdf)

Linus W. details kernel-external API changes:
- `/sys/class/gpio` sysfs interface is marked as deprecated
-  new character device `/sys/bus/gpiochipN` or `/dev/gpiochipN`

Linus W. emphasized these points:
#### The Rules of Linux Userspace GPIO
1. You do not access GPIOs from userspace
2. YOU DO NOT ACCESS GPIOS FROM USERSPACE
3. Read Documentation/gpio/drivers-on-gpio.txt
4. Use the character device

#### Features the Character Device Has
* Discovery mechanism (not magic numbers)
* Cleanup of resources on closing or crashing
* Open Drain / Open Source
* Get and set multiple lines at once
* Good examples: tools/gpio/*Features the Character Device Has
* Discovery mechanism (not magic numbers)
* Cleanup of resources on closing or crashing
* Open Drain / Open Source
* Get and set multiple lines at once
* Good examples: tools/gpio/*

pdp7, 2017-09-11:
Information on when these gpiolib features were merged into the kernel:

### gpio: add a userspace chardev ABI for GPIOs
* commit `3c702e9987e26
* `[GIT PULL] GPIO bulk changes for kernel v4.6` (Mar 17, 2016)
* ioctl
  * GPIO_GET_CHIPINFO_IOCTL

### gpio: add userspace ABI for GPIO line information
* commit `521a2ad6f862a`
* `[GIT PULL] GPIO bulk changes for kernel v4.6` (Mar 17, 2016)
* ioctl
  * GPIO_GET_LINEINFO_IOCTL

### gpio: userspace ABI for reading/writing GPIO lines
* commit: `d7c51b47ac11e`
* `[GIT PULL] bulk GPIO changes for v4.8` (Jul 25, 2016)
* ioctl
  * GPIOHANDLE_GET_LINE_VALUES_IOCTL
  * GPIOHANDLE_SET_LINE_VALUES_IOCTL
  * GPIO_GET_LINEHANDLE_IOCTL

### gpio: userspace ABI for reading GPIO line events
* commit `61f922db72216`
* `[GIT PULL] bulk GPIO changes for v4.8` (Jul 25, 2016)
* ioctl
  * GPIOHANDLE_GET_LINE_VALUES_IOCTL
  * GPIO_GET_LINEEVENT_IOCTL

pdp7, 2017-12-02:
lsgpio utlitiy uses these new ioctl()'s to list all lines for each gpiochip:
```
debian@beaglebone:~/gpio$ sudo ./lsgpio 
Linus W GPIO chip: gpiochip3, "gpio", 32 GPIO lines
	line  0: unnamed unused
	line  1: unnamed unused
	line  2: unnamed unused
	line  3: unnamed unused
	line  4: unnamed unused
	line  5: unnamed unused
	line  6: unnamed unused
	line  7: unnamed unused
	line  8: unnamed unused
	line  9: unnamed unused
	line 10: unnamed unused
	line 11: unnamed unused
	line 12: unnamed unused
	line 13: unnamed unused
	line 14: unnamed "P9_31" [kernel output]
	line 15: unnamed "P9_29" [kernel output]
	line 16: unnamed "P9_30" [kernel]
	line 17: unnamed "P9_28" [kernel output]
	line 18: unnamed "P9_92" [kernel]
	line 19: unnamed "P9_27" [kernel]
	line 20: unnamed "P9_91" [kernel]
	line 21: unnamed "P9_25" [kernel]
	line 22: unnamed unused
	line 23: unnamed unused
	line 24: unnamed unused
	line 25: unnamed unused
	line 26: unnamed unused
	line 27: unnamed unused
	line 28: unnamed unused
	line 29: unnamed unused
	line 30: unnamed unused
	line 31: unnamed unused
Linus W GPIO chip: gpiochip2, "gpio", 32 GPIO lines
	line  0: unnamed unused
	line  1: unnamed "P8_18" [kernel]
	line  2: unnamed "P8_07" [kernel]
	line  3: unnamed "P8_08" [kernel]
	line  4: unnamed "P8_10" [kernel output]
	line  5: unnamed "P8_09" [kernel]
	line  6: unnamed "P8_45" [kernel]
	line  7: unnamed "P8_46" [kernel]
	line  8: unnamed "P8_43" [kernel]
	line  9: unnamed "P8_44" [kernel]
	line 10: unnamed "P8_41" [kernel]
	line 11: unnamed "P8_42" [kernel]
	line 12: unnamed "P8_39" [kernel]
	line 13: unnamed "P8_40" [kernel]
	line 14: unnamed "P8_37" [kernel]
	line 15: unnamed "P8_38" [kernel]
	line 16: unnamed "P8_36" [kernel]
	line 17: unnamed "P8_34" [kernel]
	line 18: unnamed unused
	line 19: unnamed unused
	line 20: unnamed unused
	line 21: unnamed unused
	line 22: unnamed "P8_27" [kernel]
	line 23: unnamed "P8_29" [kernel]
	line 24: unnamed "P8_28" [kernel]
	line 25: unnamed "P8_30" [kernel]
	line 26: unnamed unused
	line 27: unnamed unused
	line 28: unnamed unused
	line 29: unnamed unused
	line 30: unnamed unused
	line 31: unnamed unused
Linus W GPIO chip: gpiochip1, "gpio", 32 GPIO lines
[... 191 lines cut ...]
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
ioctl(4, GPIO_GET_LINEINFO_IOCTL, 0xbeff65cc) = 0
+++ exited with 0 +++
```

pdp7, 2018-02-08:
**[Update on Python bindings for libgpiod](https://github.com/brgl/libgpiod/issues/5#issuecomment-364054538):**
> Yes, I was recently contacted by @sgjava who showed me his automatically generated ctypes bindings.
>
> You can find them here.
>
> However I plan to write object oriented python 3 bindings (using C extensions) right after I finish the C++ bindings.
>
> There were some issues raised on the linux-gpio mailing list about the safety of ctypes for public structures with different memory layout on 32 and 64 bit architectures.
>
> Please note that I just released v1.0 of libgpiod with an improved API which shall remain stable from now on.

**[sgjava/libgpiod-extra](https://github.com/sgjava/libgpiod-extra)**
> Automated build, Python and Java bindings for libgpiod
>
> Python bindings
> ctypesgen is used to create a Python bindings from libgpiod's gpiod.h C header file. I've added an insert file to add the missing ctypes Structures.

pdp7, 2018-02-21:
From https://github.com/brgl/libgpiod/issues/5#issuecomment-367170393
@sgjava has created **[userspaceio](https://github.com/sgjava/userspaceio)** with Python 3 bindings using CFFI:
> Python and Java bindings for GPIO, SPI, I2C and Serial interfaces in Linux user space

pdp7, 2018-03-20:
**[bookwar/python-gpiodev](https://github.com/bookwar/python-gpiodev)**
> Library to operate GPIO pins via character device
> #### GPIOHandle for character device
> New GPIO interface has been [introduced](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=1a46712aa99594eabe1e9aeedf115dfff0db1dfd) recently.
>
> It exposes GPIO interface as `/dev/gpiochip0` character device and provides several [ioctl
syscalls](https://github.com/torvalds/linux/blob/master/include/uapi/linux/gpio.h) for bulk operations on sets of GPIO pins.
>
> In [gpiodev/src/gpioctl.c](gpiodev/src/gpioctl.c) we wrap the ioctl calls into C-functions suitable for later use.
>
> In [gpiodev/gpio.py](gpiodev/gpio.py) the ctypes bindings created and then used to define the main GPIOHandle class.
>
> Check [examples](examples) for usage.

silver2row, 2018-04-06:
Hello Sir,

I am a novice w/ all this info. but if you need support, I will take time out to look over things and test stuff.

Seth

P.S. If you want to direct me as to what is important right now versus what can wait, I can make some notes and get back to you.

silver2row, 2018-09-03:
@pdp7 ,

Hello Sir...I have been reading and trying to catch up to what you may already know about these instances listed above.

...

https://github.com/torvalds/linux/blob/master/Documentation/driver-api/gpio/drivers-on-gpio.rst.

This is what I found recently in Linux. I have been watching the YouTube.com video on gpio and how we are not supposed to call it from UserSpace. I am almost done w/ the video.

Seth

P.S. I will continue my search on these subjects. Are you still wanting assistance for any particular subject matter?

silver2row, 2018-09-03:
Hello,

I still have the /sys/class/gpio (legacy) system.

Seth

P.S. I will work on it. Be back later in the year or hopefully, sooner.

silver2row, 2018-09-06:
Okay and Hello,

I found the userspaceio GitHub.com page. This is good motion in action. I found the fellow that did a presentation on Linux Embedded Conference, too. He has a GitHub.com page for libgpiod. 

Seth

P.S. I was able to get my LED lit and turned off by the character device system.

silver2row, 2018-09-07:
@pdp7 

Hello Again Sir...Look here: https://www.hackster.io/silver2row/the-new-character-device-idea-0968d6. I made a "funny" set up of making a LED blink on and after the return button has been pressed, the LED turns off when the pin is high w/ `gpioset --mode=wait gpiochip1 28=0` for P9_12.

Seth

P.S. I thought this was fascinating. Now, I need to read more as usual before I can get more into things. So, let me know if you need support. I am actually building a CNC for fun, just for posting, but I will have time to read more. Just show me the way and I can start on ideas.

pdp7, 2018-09-10:
@silver2row thanks, that is very interesting

silver2row, 2018-09-11:
@pdp7 ,

No issue, sir. I thought you would have liked that idea. 

Seth

P.S. I know you may not know everything on this subject but neither do I. If I can ever catch up to what you have read already, I may be able to help. I will keep searching online for ideas and issues pertaining to character device support. So, just let me know if you have some nice docs. on this subject.