adafruit/adafruit-beaglebone-io-python #159: Porting to am437x

alpintrekker, 2017-08-26:
Hello,

I'm interested porting GPIO, and only GPIO module to Dabian Jessie, running on am437x.

Since ther is no overlay support here at all, I don't need overlays, and any IO, other GPIO going to be handled by native python modules.

Do I consider correctly, that the task of porting would be limited to editing **pin_t table** in [common.c](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c) and editing some compile options?

Any advice would be appreciated,

Regards,
András

pdp7, 2017-08-26:
@alpintrekker I've not used AM437x SoC before.  Which board do you have?

alpintrekker, 2017-08-26:
@pdp7 this is [CM-T43 from CompuLab](http://www.compulab.com/products/computer-on-modules/cm-t43/#devres) on their development board, SB-SOM, but it is the same Sitara family like AM335x on BeagleBone (where I used to use Adafruit_BBIO.GPIO), except it has a reasonably better set of peripherals and a bit higher price tag in low volumes.

pdp7, 2017-08-28:
@alpintrekker  I imagine you will need to change `pwm_table[]`  and ` table[]` in `common.c`.   I suspect the the memory map addresses for peripherals may be different and therefore exact paths in /sys may differ.  The device tree source files for your board should provide insights on what values to use.

pdp7, 2017-10-18:
@alpintrekker This work by @RobertCNelson  may be of interest if you are still working on porting to different: #173 

I am going to close this issue for now.  Please create a pull request if you have made changes to support AM437x.

closed 2017-10-18 by pdp7