adafruit/adafruit-beaglebone-io-python #161: Adafruit_BBIO pip3 install fails in Beaglebone, but passes for version 1.0.3

sayanee, 2017-08-31:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Beaglebone

- Python version (run `python -version` or `python3 -version`):  Python 3.4.2

- Error message you are receiving, including any Python exception traces:  

```sh
pip3 install Adafruit_BBIO
...
File "/usr/local/lib/python3.4/dist-packages/Adafruit_BBIO/Encoder.py", line 110
        print "config_pin> pin={0} result={1}".format(pin, result)
                                             ^
    SyntaxError: invalid syntax

Successfully installed Adafruit-BBIO
Cleaning up...
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

1. Take a Beaglebone with debian image
1. Install `pip3` and `Adafruit_BBIO`

There was error in installing the latest version, but no error when the previous version was installed.

```sh
$ apt-get update
$ apt-get install pip3
$ pip3 install Adafruit_BBIO
......
File "/usr/local/lib/python3.4/dist-packages/Adafruit_BBIO/Encoder.py", line 110
        print "config_pin> pin={0} result={1}".format(pin, result)
                                             ^
    SyntaxError: invalid syntax

Successfully installed Adafruit-BBIO
Cleaning up...

$  pip3 uninstall Adafruit_BBIO
    Successfully uninstalled Adafruit-BBIO
$ pip install Adafruit_BBIO==1.0.3
Successfully installed Adafruit-BBIO
Cleaning up...
```

pdp7, 2017-08-31:
Thanks for reporting @sayanee.  I'll get that fixed.

pdp7, 2017-09-01:
This print syntax error was fixed in https://github.com/adafruit/adafruit-beaglebone-io-python/commit/538e564ad9ca1b56d7d934484753bf303df23a04

Was Adafruit_BBIO 1.0.4 being used?

pdp7, 2017-09-01:
I've created release https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.0.5

sayanee, 2017-09-01:
thank you @pdp7! I will test it out with the new release by next week.