adafruit/adafruit-beaglebone-io-python #165: Set gpio mode failed, missing file or invalid permissions on GPIO.setup

craigw76, 2017-09-22:
Running a Beaglebone Black. I have just  installed the latest  IoT release from the BeagleBoard page 
_bone-debian-9.1-iot-armhf-2017-08-31-4gb_ after corrupting my previous install. I had a python app that was watching a GPIO pin for input. The app failed with the same error, so I pared it down to as simple as I could. Just the import and setup will cause the failure. It has the same failure when running with sudo, so I don't think it's a permission issue. What am I missing. It must be something pretty simple since there are so few hits on Google with this error.

After I wrote the image to an SD card, I then followed the install procedures for Adafruit_BBIO. 
I used the input test from the readme. Created test.py and ran it with sudo.
sudo python test.py
```
import Adafruit_BBIO.GPIO as GPIO
GPIO.setup("P8_14", GPIO.IN)
if GPIO.input("P8_14"):
      print("HIGH")
else:
      print("LOW")
```
I get this output:
```
Traceback (most recent call last):
  File "test.py", line 2, in <module>
    GPIO.setup("P8_14", GPIO.IN) 
ValueError: Set gpio mode failed, missing file or invalid permissions.

```

RobertCNelson, 2017-09-22:
@craigw76 please run:

```
sudo /opt/scripts/tools/version.sh
```
and 
```
dmesg | grep bone
```

Regards,

craigw76, 2017-09-22:
```
$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[2ce750d881941c5189db9e189af90517e11c079f]
eeprom:[A335BNLT0A5B2213BBBK1184]
dogtag:[BeagleBoard.org Debian Image 2017-08-31]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-rc2-00002-g7c9353]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.10-rc3-00001-gfb375b]
kernel:[4.4.84-ti-r120]
nodejs:[v6.11.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

```
```
$ dmesg | grep bone
[    2.367662] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,0A5B,2213BBBK1184'
[    2.367692] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    2.406109] bone_capemgr bone_capemgr: slot #0: No cape found
[    2.446102] bone_capemgr bone_capemgr: slot #1: No cape found
[    2.486100] bone_capemgr bone_capemgr: slot #2: No cape found
[    2.510140] bone_capemgr bone_capemgr: Invalid signature 'ffffffff' at slot 3
[    2.517347] bone_capemgr bone_capemgr: slot #3: No cape found
[    2.517462] bone_capemgr bone_capemgr: initialized OK.
[    5.164852] systemd[1]: Set hostname to <beaglebone>.

```

RobertCNelson, 2017-09-22:
Hi @craigw76 so you have a very old version of u-boot on your eMMC.

Do you need to keep that data?

If not:

```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=100
```
If you 'need' it, GND P8_43 (same pin that's connected to the BOOT push button)

Then reboot, and try again.

Regards,

craigw76, 2017-09-22:
Hi @RobertCNelson, Thanks! I had my app and data backed up, just not the entire OS. That did it. It seems to work now. I figured it was something simple I'd missed. It seems to be working now.
```
$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[2ce750d881941c5189db9e189af90517e11c079f]
eeprom:[A335BNLT0A5B2213BBBK1184]
dogtag:[BeagleBoard.org Debian Image 2017-08-31]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-rc2-00002-g7c9353]
kernel:[4.4.84-ti-r120]
nodejs:[v6.11.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

```
```
$ dmesg | grep bone
[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
[    2.429091] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,0A5B,2213BBBK1184'
[    2.429125] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    2.429185] bone_capemgr bone_capemgr: slot #0: auto loading handled by U-Boot
[    2.429210] bone_capemgr bone_capemgr: slot #1: auto loading handled by U-Boot
[    2.429233] bone_capemgr bone_capemgr: slot #2: auto loading handled by U-Boot
[    2.429255] bone_capemgr bone_capemgr: slot #3: auto loading handled by U-Boot
[    2.429807] bone_capemgr bone_capemgr: initialized OK.
[   10.341775] systemd[1]: Set hostname to <beaglebone>.
```

craigw76, 2017-09-23:
@RobertCNelson All seems good now. I'll close this case.

Thanks again

closed 2017-09-23 by craigw76

MajdBy, 2017-11-14:
When I execute: 
`sudo /opt/scripts/tools/version.sh`
I get the following : 
`sudo: /opt/scripts/tools/version.sh: command not found`

RobertCNelson, 2017-11-14:
@MajdBy So you just have a really old image..
you can run:
```
cat /etc/dogtag
```
But what's your real issue?

Regards,

MajdBy, 2017-11-21:
The issue is that I can't set my GPIO as Input becomming always this error: 
`ValueError: Set gpio mode failed, missing file or invalid permissions`

Running `cat /etc/dogtag` gave:
`rcn-ee.net console Debian Image 2016-03-11`

The image is not that old, is it ?

pdp7, 2017-11-21:
@MajdBy Please try the current "Debian 9.2 2017-10-10 4GB SD IoT image" on http://beagleboard.org/latest-images

candronikos, 2017-11-23:
I'm also getting this error and can confirm that I am using the latest image (BeagleBoard.org Debian Image 2017-10-10) but, on a very strange note, the issue is disappears  as Node-RED is started. What would explain this?

RobertCNelson, 2017-11-23:
One of the first things node red does is load cape universal.. so that'll explain why the gpio want available.

candronikos, 2017-11-23:
Yeah the term cape definitely comes up in the error logs when trying to start the PWMs. Is this a bug in the library? Any idea of a fix or workaround?

magaete, 2019-09-16:
Hi,

I resolve the problem "Set gpio mode failed, missing file or invalid permissions on GPIO.setup" editing the file /etc/group. I add my usser to the group GPIO