adafruit/adafruit-beaglebone-io-python #166: SPI not working on beaglebone black debian stretch

hisham-rumaney, 2017-10-01:
Hi when I run the following test code:

**spi = SPI(0,0)          
print(spi.xfer([0x01,0x00]))**

I get these results:
**[[0,0]]**

**My MISO and MOSI pins are connected together using a wire and I am using SPI0 so I am using the corresponding pins**

here are the details of the linux: 
**Linux beaglebone 4.9.44-ti-r56**
**PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian**

RobertCNelson, 2017-10-01:
@hisham-rumaney please run:

```
sudo /opt/scripts/tools/version.sh
```
So we know your kernel/uboot combination..

hisham-rumaney, 2017-10-01:
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[8e0b1e6f135ff3c30468f49a7a952c98be97219a]
eeprom:[A335BNLT0A5C2713BBBK7464]
dogtag:[BeagleBoard.org Debian Image 2017-08-24]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-rc2-00002-g84a7f2]
kernel:[4.9.44-ti-r56]
nodejs:[v6.11.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170821-0rcnee1~stretch+20170821]
pkg:[firmware-ti-connectivity]:[20161130-3]

hisham-rumaney, 2017-10-01:
[image]

RobertCNelson, 2017-10-01:
@hisham-rumaney okay u-boot/kernel combination is fine for cape_universal.. @pdp7 adafruit-beaglebone-io just calls' config-pin right? ;)

Regards,

hisham-rumaney, 2017-10-01:
I urgently need the SPI to work. Do you have any advice for getting this to work?

hisham-rumaney, 2017-10-01:
All if you look at the top part of the image you will see the output of slots

hisham-rumaney, 2017-10-01:
if anyone would like to skype to solve this problem that would be amazing

hisham-rumaney, 2017-10-01:
I got it to work by using 

[image]

pdp7, 2017-10-02:
@hisham-rumaney thanks for following up and good to see it works.

@RobertCNelson no it does not use config-pin but I believe it should. I had started a [config-pin branch](https://github.com/adafruit/adafruit-beaglebone-io-python/tree/config-pin) and need to continue working on it to fix issues like this. Most recent commit for reference: https://github.com/adafruit/adafruit-beaglebone-io-python/commit/b92608458e2122f40901b6d5af74dcf11e9c2c02

pdp7, 2017-10-02:
I'll update https://github.com/adafruit/adafruit-beaglebone-io-python/issues/158 as progress continues on that branch

closed 2017-10-02 by pdp7

RobertCNelson, 2017-10-02:
thanks @pdp7 i really should add a daily "test" to my weekly uptime machines, to verify adafruit-bb-io works on all releases..

if i jumper the spi's, cross a uart rx/tx, and something for pwm, i figure (ecap?). ;)

Dmytro-Bez, 2020-03-26:
Hello everyone. Is the question how to properly configure and describe the code for organizing communication between Arduino and BB through this interface?