adafruit/adafruit-beaglebone-io-python #168: RE: Hello and PWM on BBBW w/ Adafruit_BBIO

silver2row, 2017-10-02:
Hello,

**uname -a**: Linux beaglebone 4.9.50-ti-r61 #1 SMP PREEMPT Fri Sep 15 20:10:18 UTC 2017 armv7l GNU/Linux

**Python**: Python 2.7.13 (default, Jan 19 2017, 14:48:08)
                   [GCC 6.3.0 20170118] on linux2

I am using the BBBW for software testing and hardware testing. I am testing a piece of software from the book, "Getting Started with BeagleBone" (Richardson 2014). I am on page 78, i.e example 6 - 9.

Software:

    from flask import Flask, render_template
    app = Flask(__name__)
    import Adafruit_BBIO.GPIO as GPIO
    import Adafruit_BBIO.PWM as PWM

    PWM.start("P8_13", 0.0)

    @app.route("/")
    def hello():
        if GPIO.input(P8_11):
            doorStatus = "open"
        else:
            doorStatus = "closed"
        templateData = {
            'doorStatus': doorStatus,
        }
        return render_template('main-door.html', **templateData)
    
    @app.route('/ledLevel/<level>')
    def pin_state(level):
        PWM.set_duty_cycle("P8_13", float(level))
        return "LED level set to " + "."
    
    if __name__ == "__main__":
        app.run(host='192.168.7.2', port=5000, debug=True)

Issue or Error:

RuntimeError: Problem with a sysfs file

Seth

P.S. I hope this is enough info. If you have any questions, please do not hesitate to contact me.

pdp7, 2017-10-02:
@silver2row  My first guess would be permissions.

How do you invoke the python script?

Ideally, I'd like to have you start it with strace so that the system calls are recorded.  I suspect we'll know what the issue is by seeing which open() or stat() calls fail.

Something like this should do it:
```sudo strace -f -o /tmp/strace.log python ./name-of-your.py```

silver2row, 2017-10-03:
Okay...I am going to get to work soon with the listed commands. If I come across issues, I will get back to you.

Seth

silver2row, 2017-10-03:
Hello Sir,

Seth here. I do not have strace. Oh and I use the sudo python taco.py to run my .py files so far.

Seth

silver2row, 2017-10-03:
https://github.com/strace/strace/blob/master/INSTALL is what I found so far for strace. If you want me to download it, I can. 

Seth

P.S. If you can figure this out by ptrace or strace, that is okay with me. Just give me the go-ahead and I will get it and go with the flow.

silver2row, 2017-10-03:
@pdp7 

Someone just me of how the capemgr works and I am trying it now. I will get back to you soon.

Seth

pdp7, 2017-10-03:
@silver2row Please run this command to install strace:
```sudo apt-get install strace```

silver2row, 2017-10-03:
@pdp7 

I invoke python by: 
**sudo python taco.py** 

I tried strace and this happened: 
  **File "./LED-door.py", line 6, in <module>
    PWM.start("P8_13", 0.0)
RuntimeError: Problem with a sysfs file**

version.sh:

**git:/opt/scripts/:[ec0819e303842004ef1f1bd6c2339929c54eb6fd]
eeprom:[A335BNLTBWA51650BBWG2883]
dogtag:[BeagleBoard.org Debian Image 2017-09-17]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g5414234f35]
kernel:[4.9.50-ti-r61]
nodejs:[v6.11.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171002.0-0rcnee1~stretch+20171002]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]**

Dogtag:
**BeagleBoard.org Debian Image 2017-09-17**

Seth

P.S. I installed it and it is done. I hope this helps you.

pdp7, 2017-10-03:
@silver2row Please run this command to install strace:
```sudo apt-get install strace```

Then run this:
```sudo strace -f -o /tmp/strace-taco.txt python taco.py```

This should create the file` /tmp/strace-taco.txt`.  Please paste the contents of that file into a new [GitHub Gist](https://gist.github.com) and include the link here.

silver2row, 2017-10-03:
@pdp7 

Hello...how do I get all that info. copied to get it to the GitHub Gist?

Seth

P.S. If I need to find out, I will search while you take time out to reply.

silver2row, 2017-10-03:
Hello,

I am on line 300 and at 3%. Are you sure this is the step we need to take?

Seth

silver2row, 2017-10-03:
Hello Again,

I have over 7500 lines yanked in VIM and I cannot load any of them in the GitHub Gist.

Seth

pdp7, 2017-10-03:
@silver2row The lines of interest are those that relate to accessing the `/sys` filesystem.  Please run this command:
```grep /sys/ /tmp/strace-taco.txt ```
and paste the output into a gist.

silver2row, 2017-10-03:
Hello @pdp7,

This should be it. If you think I made an error, please let me know. https://gist.github.com/silver2row/0c5df8619b0f66606f019301ceda13d9

Seth

pdp7, 2017-10-03:
@silver2row Thanks that is what I was interested in.  However, I did not appear that there are any errors as I had been expecting to see.

I'd like to rule out issue with the pin mode.  Please run this command which should verify the mode that P8_13 is set to:
```config-pin -q p8.13```

This will set p8.13 mode to pwm:
```config-pin p8.13 pwm```

@robertcnelson any ideas?  I'm thinking might be a permissions issue with flask being able to write to `/sys`?  Maybe [disable u-boot overlays](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays)? thanks!

silver2row, 2017-10-03:
@pdp7 

Hello Again,

    config-pin -q p8.13 
...shows on my terminal this: P8_13 Mode: default Direction: in Value: 0

    config-pin p8.13 pwm
...I run this cmd. I get back this when I run my software as sudo python LED-door.py

**Traceback (most recent call last):
  File "LED-door.py", line 6, in <module>
    PWM.start("P8_13", 0.0)
RuntimeError: Problem with a sysfs file**

If you have any other ideas, please let me know. If not, I understand. I can suck it up and move on.

Seth

P.S. I will keep plugging at it until the halting on this issue has commenced. Peace for now!

silver2row, 2017-10-06:
Hello,

I changed a direction file in /sys/class/gpio/gpio23 and another file called uevent in the same directory. I was unable to change the value file in the same directory.

I came across the same issue. So, I have figured out that the issue, if not "value," is in another directory.

Seth

P.S. I want to figure out exactly what /sys/fs file is creating issues. I am going to look into this further. If you have any evidence of this situation, please do not distress. Show me! Thank you again, pdp7.

silver2row, 2017-10-06:
Hello Again,

I found some more files that will not allow me to write. They are not writeable for some reason. For example: cd /sys/fs/btrfs/features brings me to a set of files that I can change but the issue is that I cannnot. I want to change extended_iref to 0 instead of having it at 1. 

Seth

P.S. Even with sudo as my argument cmd on the terminal, permission is denied. Are there special permissions that would lead me to changing specific files?

silver2row, 2017-10-06:
Hello,

I found some info. from kernel.org about those files and directories.

I think this is better left to some very experienced programmers in C++ or whatever. I do not know enough, as is, to go further. 

Seth

P.S. If one day I learn some, I will return service.

pdp7, 2017-10-08:
@silver2row hello, sorry for my delayed response.  Is this still an issue?   Please let me know and I will continue to investigate and followup with you.  Thanks.

pdp7, 2017-10-08:
@silver2row I have not yet tried to fully reproduce your setup.  Did you type in the code from Matt's book?  Or were you able to download the source code from somewhere?

pdp7, 2017-10-08:
@silver2row my apologies for overlooking this earlier, but I see that you are running the Linux 4.9 kernel.  I believe there is a regression that is preventing PWM from working properly with Adafruit_BBIO.  I need to investigate and resolve that issue.

**Please try installing the Linux 4.4 kernel instead with these instructions:**
```
cd /opt/scripts/tools/
sudo git pull
sudo ./update_kernel.sh --lts-4_4
sudo reboot
```

silver2row, 2017-10-08:
Hello,

I was using kernel 4.9.x. I can try the 4.4.x kernel, I guess. Oh and yes sir, I got the book reference directly out of the book. 

Seth

P.S. I hope all goes well with your search to resolve this PWM issue with 4.9.x. If you need any support on reading documents, please let me know.

silver2row, 2017-10-12:
[image]

This is my error code outside of the Terminal on a Flask debugger screen online at 192.168.7.2:5000.

Here are the error codes on the terminal when using kernel 4.4.x:

     * Running on http://192.168.7.2:5000/ (Press CTRL+C to quit)
     * Restarting with stat
     * Debugger is active!
     * Debugger PIN: 972-185-751
    192.168.7.1 - - [12/Oct/2017 00:39:09] "GET / HTTP/1.1" 500 -
    Traceback (most recent call last):
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1997, in __call__
        return self.wsgi_app(environ, start_response)
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1985, in wsgi_app
        response = self.handle_exception(e)
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1540, in handle_exception
        reraise(exc_type, exc_value, tb)
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1982, in wsgi_app
        response = self.full_dispatch_request()
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1614, in full_dispatch_request
        rv = self.handle_user_exception(e)
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1517, in handle_user_exception
        reraise(exc_type, exc_value, tb)
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1612, in full_dispatch_request
        rv = self.dispatch_request()
      File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1598, in dispatch_request
        return self.view_functions[rule.endpoint](**req.view_args)
      File "/home/debian/LED-door.py", line 10, in hello
        if GPIO.input("P8_11"):
    RuntimeError: You must setup() the GPIO channel first
    192.168.7.1 - - [12/Oct/2017 00:39:09] "GET /?__debugger__=yes&cmd=resource&f=style.css 
    HTTP/1.1" 200 -
    192.168.7.1 - - [12/Oct/2017 00:39:10] "GET /?__debugger__=yes&cmd=resource&f=jquery.js 
    HTTP/1.1" 200 -
    192.168.7.1 - - [12/Oct/2017 00:39:10] "GET /?__debugger__=yes&cmd=resource&f=debugger.js 
    HTTP/1.1" 200 -
    192.168.7.1 - - [12/Oct/2017 00:39:10] "GET /?__debugger__=yes&cmd=resource&f=ubuntu.ttf 
    HTTP/1.1" 200 -
    192.168.7.1 - - [12/Oct/2017 00:39:10] "GET /?__debugger__=yes&cmd=resource&f=console.png 
    HTTP/1.1" 200 -
    ^CException in thread Thread-1 (most likely raised during interpreter shutdown):
    Traceback (most recent call last):
      File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
      File "/usr/lib/python2.7/threading.py", line 754, in run
      File "/usr/local/lib/python2.7/dist-packages/werkzeug/serving.py", line 702, in inner
      File "/usr/local/lib/python2.7/dist-packages/werkzeug/serving.py", line 539, in serve_forever
      File "/usr/lib/python2.7/SocketServer.py", line 236, in serve_forever
      File "/usr/lib/python2.7/threading.py", line 585, in set
      File "/usr/lib/python2.7/threading.py", line 407, in notifyAll
    <type 'exceptions.TypeError'>: 'NoneType' object is not callable

If you need any other info, please contact me.

Seth

P.S. I am sorry for opening up this issue again but I cannot get out of debugging mode on the Flask application. I will contact their forum on Stack Exchange under the header Flask, too.

silver2row, 2017-10-12:
Seth

I am using config-pin to address P8_19. 
    sudo config-pin p8.19 pwm returns nothing and this is why I thought it worked. 

Seth

silver2row, 2017-10-12:
Hello @pdp7 ,

Seth here. I am starting to get that idea, the Flask application on Python with the BBB running Adafruit_BBIO for pin support, working. I just needed to setup GPIO after calling it from XXX as XXX and so on.

My software needs some fine tuning but for now, everything is back working.

Seth

P.S. I am using kernel 4.4.x and some software, that I changed sort of, from that Richardson 2014 book I mentioned earlier. The sensor works, magnetic switch, but I cannot get the LED to light up when the web address changes on the address bar. For instance, I should be able to type 192.168.7.2:5000/50 to get the LED at half its capacity. I cannot get the LED to light at all.

pdp7, 2017-10-12:
@silver2row yes, `GPIO.setup("P8_11", GPIO.IN)` is needed before `GPIO.input("P8_11")`.  I've added the `GPIO.setup()` before `app.run()`:
```
from flask import Flask, render_template
app = Flask(__name__)
import Adafruit_BBIO.GPIO as GPIO
import Adafruit_BBIO.PWM as PWM

PWM.start("P8_13", 0.0)

@app.route("/")
def hello():
    if GPIO.input("P8_11"):
        doorStatus = "open"
    else:
        doorStatus = "closed"
    templateData = {
        'doorStatus': doorStatus,
    }
    return render_template('main-door.html', **templateData)

@app.route('/ledLevel/<level>')
def pin_state(level):
    PWM.set_duty_cycle("P8_13", float(level))
    return "LED level set to " + "."

if __name__ == "__main__":
    GPIO.setup("P8_11", GPIO.IN)
    app.run(host='192.168.7.2', port=5000, debug=True)
```

Here is general example of GPIO input:
```
debian@beaglebone:~/forum$ python
Python 2.7.9 (default, Aug 13 2016, 17:56:53) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.GPIO as GPIO
>>> GPIO.setup("P8_11", GPIO.IN)
>>> GPIO.input("P8_11")
0
```

pdp7, 2017-10-12:
@silver2row using the flask program above, here are my results:
```
debian@beaglebone:~/forum$ python server.py
Traceback (most recent call last):
  File "server.py", line 6, in <module>
    PWM.start("P8_13", 0.0)
RuntimeError: Problem with a sysfs file
debian@beaglebone:~/forum$ sudo python server.py
 * Running on http://192.168.7.2:5000/
 * Restarting with reloader
```

pdp7, 2017-10-12:
@silver2row please try running these commands and  then start you flask server:
```
find /sys/devices/platform/ocp/ -type f |grep /sys/devices/platform/ocp/ |grep pwmchip |egrep '(duty_cycle|enable|period|polarity)$' |xargs sudo chgrp gpio 

find /sys/devices/platform/ocp/ -type f |grep /sys/devices/platform/ocp/ |grep pwmchip |egrep '(duty_cycle|enable|period|polarity)$' |xargs sudo chmod g+w

find /sys/devices/platform/ocp/ -type f |grep /sys/devices/platform/ocp/ |grep pwmchip |egrep '(duty_cycle|enable|period|polarity)$' | xargs ls -la
```

silver2row, 2017-10-12:
Alright @pdp7 ,

I will run those commands in a bit. I upgraded back to 4.9.x to test things. Without control to write, I would be up S@!&'s creek. Thank you.

Seth

P.S. I got the software to work with the GPIO.setup command.

RobertCNelson, 2017-10-12:
@pdp7 

permissions on:

```
pwmchip6/pwm1
```
Need: 

https://github.com/jadonk/bonescript/issues/156

kernel patch: (currently applied to v4.11.x+ (and ready to back port to v4.9.x/v4.4.x at any time))
https://github.com/RobertCNelson/linux-dev/blob/master/patches/drivers/pwm/0001-pwm-Create-device-class-for-pwm-channels.patch

Regards,

pdp7, 2017-10-12:
@silver2row I'd recommend not using Linux 4.9 kernel for now.  You should be able to get the flask web application running ok with Linux 4.4 kernel.

pdp7, 2017-10-12:
@robertcnelson It is good that there is a path forward.  

@silver2row for now, please stick with Linux 4.4 kernel.  Once I resolve #170, you should then be able to use Linux 4.9

silver2row, 2017-10-12:
Okay @pdp7 ,

I will "downgrade" back to kernel 4.4.x. 

Seth

P.S. The server works. I can change the brightness of the LED with address bar commands and find out if my door is open or closed and finally, receiving the e-mail after the alloted 25 seconds if the door is left open.

pdp7, 2017-10-17:
@silver2row have you tried with Linux 4.4.x kernel?  Can this issue be closed?

silver2row, 2017-10-18:
@pdp7 I tried 4.4.x and it works just fine.

Seth

closed 2017-10-18 by silver2row