adafruit/adafruit-beaglebone-io-python #169: Problems with my BeagleBone Black PWM.

tomek242, 2017-10-05:
Hello everyone.
I have a big problem with my BBB. I try to use my board to simple tracking robot with DAGU Rover 5 chassis and DAGU motor driver board but... whatever version of debian i try to use, I see the same errors:
RuntimeError: You must start() the PWM channel first; or
ehrpwm error - duty is larger than period; or
RuntimeError: Problem with /sysfs.
Of course i try to use [THIS](url) solutions but there is no results still that same errors. I checked that problems for Angstrom, Debian 7,2; 7,11; 8.5; 8.7; 9.1 with  [beaglebone-universal-io](url) and without it, but still no effect.
Maybe someone here had working Debian uSD image with properly configured cape, or help some in configuration this?
Greetings. Thanks.

silver2row, 2017-10-06:
Hello tomek242,

My name is Seth. I just published a concern with similar instances of PWM not working. I get the RuntimeError: Problem with sysfs, too.

I tried a handful of resources and asked around. I could not get the PWM to start in working order for my software.

My software is from #168. I get the sysfs entry error and I have tried bone_capemgr, overlays, and config-pin <my pin number> pwm for writing the software while in the current terminal. I have empty answers so far.

tomek242: If you figure something out, please let me know. I would love to get my book software to work.

Seth

P.S. Thank you for posting about this issue.

tomek242, 2017-10-06:
Hello Seth.

Unfortunately, until now I have not been able to solve these problems. That's why I wrote about my issue here. Perhaps one of the developers has done it and is working properly.
Greetings.

pdp7, 2017-10-08:
@tomek242 sorry for my delayed response.

I'd like to get some information about what is currently running on your BeagleBone. **Please paste the output of the following commands:**
```
cat /etc/dogtag 
cat /etc/debian_version 
cat /proc/cmdline 
cat /sys/devices/platform/bone_capemgr/slots
cat /boot/uEnv.txt 
config-pin -q p9.14
uname -a
```

_FYI: Adafruit_BBIO (this library) no longer supports the Angstrom images.  The official BeagleBoard.org image has been Debian-based since 2014.  For more info: [BeagleBoard.org Latest Images](http://beagleboard.org/latest-images)_

**I'd also like you to try this test.** Start the Python interpreter:
```
sudo python
```
Then type or paste these lines:
```
import Adafruit_BBIO.PWM as PWM
#PWM.start(channel, duty, freq=2000, polarity=0)
#duty values are valid 0 (off) to 100 (on)
PWM.start("P9_14", 50)
PWM.set_duty_cycle("P9_14", 25.5)
PWM.set_frequency("P9_14", 10)

PWM.stop("P9_14")
PWM.cleanup()

#set polarity to 1 on start:
PWM.start("P9_14", 50, 2000, 1)
```

Your results should be similar to:
```
debian@beaglebone:~$ sudo python
[sudo] password for debian: 
Python 2.7.9 (default, Aug 13 2016, 17:56:53) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> import Adafruit_BBIO.PWM as PWM
>>> #PWM.start(channel, duty, freq=2000, polarity=0)
... #duty values are valid 0 (off) to 100 (on)
... PWM.start("P9_14", 50)
>>> PWM.set_duty_cycle("P9_14", 25.5)
>>> PWM.set_frequency("P9_14", 10)
>>> 
>>> PWM.stop("P9_14")
>>> PWM.cleanup()
>>> 
>>> #set polarity to 1 on start:
... PWM.start("P9_14", 50, 2000, 1)
>>> exit
Use exit() or Ctrl-D (i.e. EOF) to exit
>>> exit()
```

In addition, I'd also like to know which pins on the BeagleBone that you are using for your project.  **Would it be possible for you to post a link to a photo of your project?**

_( @silver2row I've just replied in #168 )_

pdp7, 2017-10-08:
@tomek242 I just posted about this to @silver2row in #168.  I believe there is problem with using PWM with the Linux 4.9 kernel and Adafruit_BBIO.  I need to investigate and resolve that issue.

In the meantime, please ensure that you are running the Linux 4.4 kernel.

**Please try installing the Linux 4.4 kernel instead with these instructions:**
```
cd /opt/scripts/tools/
sudo git pull
sudo ./update_kernel.sh --lts-4_4
sudo reboot
```
The above presumes that you are using an [official BeagleBoard.org Debian image](http://beagleboard.org/latest-images)

tomek242, 2017-10-08:
Hello pdp7. I just worked with Debian 8.3
4.1.15-ti-rt-r43 #1 SMP PREEMPT RT Thu Jan 21 20:13:58 UTC 2016 armv7l GNU/Linux
taken from [THIS link](url).

At this time thath the only image when my test program realy work. When i tape:
`root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots`
I take this:
`0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universaln`

Besides:
`root@beaglebone:~# cat /etc/dogtag
BeagleBoard.org Debian Image 2016-01-24`

Debian ver. is 8.3 (as i wrote the only worked image) personally it could be better for me using 7.11 or 9.1.

`root@beaglebone:~# cat /proc/cmdline
console=tty0 console=ttyO0,115200n8 root=/dev/mmcblk0p1 rootfstype=ext4 rootwait coherent_pool=1M quiet cape_universal=enable`

And another think as I wrote:
`root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-universal`

`root@beaglebone:~# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.1.15-ti-rt-r43
#uuid=
#dtb=

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
#dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
#dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

cmdline=coherent_pool=1M quiet cape_universal=enable

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M quiet cape_universal=enable video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##Disable HDMI/eMMC (v3.8.x)
#cape_disable=capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN,BB-BONE-EMMC-2G

##Disable HDMI (v3.8.x)
#cape_disable=capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN

##Disable eMMC (v3.8.x)
#cape_disable=capemgr.disable_partno=BB-BONE-EMMC-2G

##Audio Cape (needs HDMI Audio disabled) (v3.8.x)
#cape_disable=capemgr.disable_partno=BB-BONELT-HDMI
#cape_enable=capemgr.enable_partno=BB-BONE-AUDI-02

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh` 

Another command:
`root@beaglebone:~# config-pin -q p9.14
P9_14 Mode: pwm`

PYthon:
`root@beaglebone:~# sudo python
Python 2.7.9 (default, Mar  1 2015, 13:48:22)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.`

And testing python code:
`root@beaglebone:~# sudo python
Python 2.7.9 (default, Mar  1 2015, 13:48:22)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.PWM as PWM
>>> #PWM.start(channel, duty, freq=2000, polarity=0)
... #duty values are valid 0 (off) to 100 (on)
... PWM.start("P9_14", 50)
>>> PWM.set_duty_cycle("P9_14", 25.5)
>>> PWM.set_frequency("P9_14", 10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Could not write to the frequency file
>>>
>>> PWM.stop("P9_14")
>>> PWM.cleanup()
>>>
>>> #set polarity to 1 on start:
... PWM.start("P9_14", 50, 2000, 1)`

I try some other kernels for example that whart You recomended for me, I mean: 
`sudo ./update_kernel.sh --lts-4_4`

but nothing result. 
I've been suffering with this for month. Unfortunately without any result.

tomek242, 2017-10-08:
Ok. I was install 4.4 lts kernel and this is result:

Code:
`import Adafruit_BBIO.PWM as PWM
myPWM="P9_16"
PWM.start(myPWM, 0, 1000)
for i in range(0,5):
    DC=input("What Duty Cycle Would You Like (0-100)? ")
    PWM.set_duty_cycle(myPWM, DC)
PWM.stop(myPWM)
PWM.cleanup()`

Takes effect:
`python test2.py
[  183.195393] ehrpwm 48302200.pwm: Period value conflicts with channel 0
Traceback (most recent call last):
  File "test2.py", line 5, in <module>
    PWM.start(myPWM, 0, 1000)
RuntimeError: Problem with a sysfs file`

And in this case of code:
`#!/usr/bin/python
import Adafruit_BBIO.PWM as PWM
import time
pin1 = "P9_14"
pin2 = "P9_16"
pin3 = "P8_19"
pin4 = "P8_13"
PWM.start(pin1,0)
PWM.start(pin2,0)
PWM.start(pin3,0)
PWM.start(pin4,0)
for i in range(0,100):  #loop 100 times
        PWM.set_duty_cycle(pin1,100)
        PWM.set_duty_cycle(pin2,100)
        PWM.set_duty_cycle(pin3,100)
        PWM.set_duty_cycle(pin4,100)
time.sleep(1)
PWM.stop(pin1)
PWM.stop(pin2)
PWM.stop(pin3)
PWM.stop(pin4)
PWM.cleanup()`

Everything seems work good :(

Uhhhh a just think that will be so easy:)

tomek242, 2017-10-09:
BTW question: wow to paste correctly program code here? Mine as You see is linear and didnt looks well.

tomek242, 2017-10-10:
Any suggestions folks? :)

silver2row, 2017-10-11:
Hello,

If I use the resistor to limit current, there is still current going through it at another spot on the breadboard. If I was to add additional requirements, normally open window sensor and other resistors, I could come across an issue with supply. 

Power may become an issue. 

Seth

P.S. I may need more voltage, i.e. 5v or higher with the correct digital equipment.

silver2row, 2017-10-11:
We may be seeing issues with power on our boards when attempting to write the software and have it produce the correct results.

Seth

silver2row, 2017-10-11:
Hello Again,

The Motor Driver probably needs to be powered separately from you BBB.

Seth

tomek242, 2017-10-11:
Hello Seth.

I think it does not matter beacause logic level cames from BBB to driver has maybe (cirka) 3,3V not more so I think thats not a problem.
Personally i powered my board and BBB separately each from another source (two DC - DC converters with power distribution boards one for 12V, for motors and the second for 5V for board logic). 
However The BBB use 5V logic (As I think remember). So if You are not sure about the logic states valuse look this:
[http://slideplayer.pl/slide/3624618/11/images/3/Poziomy+napięć+w+standardowym+układzie+TTL.jpg](url)
or this [http://ep.com.pl/cache/images/norm/3/0/6/c3JjPS9pbWFnZXMvbm9ybS8zLzAvNi8xNzMwNmtvbndlcnRlcnlfMS5qcGcmdz04ODgmaD03MzA=_srcc00682232e2d35c16ff54a61de9aa8df.jpg](url)

I think the real problem is software, I mean Debian and pins configurtion. not pin voltages :)
Tom

silver2row, 2017-10-11:
Okay...thank you for your input.

Seth

tomek242, 2017-10-11:
Hello everyone.
Today i was test Debian image:
Linux beaglebone 4.4.72-bone-rt-r17 #1 PREEMPT RT Fri Jun 16 16:48:24 UTC 2017 armv7l GNU/Linux

And this is result for code:
import Adafruit_BBIO.PWM as PWM
import time

pin1 = "P9_14"
pin2 = "P9_16"
pin3 = "P8_19"
pin4 = "P8_13"

PWM.start(pin1,0)
PWM.start(pin2,0)
PWM.start(pin3,0)
PWM.start(pin4,0)

for i in range(0,100):

        PWM.set_duty_cycle(pin1,100)
        PWM.set_duty_cycle(pin2,100)
        PWM.set_duty_cycle(pin3,100)
        PWM.set_duty_cycle(pin4,100)

time.sleep(.1)

PWM.stop(pin1)
PWM.stop(pin2)
PWM.stop(pin3)
PWM.stop(pin4)
PWM.cleanup()

Result:

Traceback (most recent call last):
  File "test1.py", line 11, in <module>
    PWM.start(pin2,0)
IOError: [Errno 2] No such file or directory: '/slots'

pdp7, 2017-10-11:
@tomek242 Please paste the output of ```cat /etc/dogtag``` so that I know which image you are using.

Also, please use the code formatting button so that the whitespace of python code remains in tact.

pdp7, 2017-10-11:
@tomek242 This test program ran OK for me. 
```
debian@beaglebone:~/forum$ sudo python issue169.py
What Duty Cycle Would You Like (0-100)? 55
What Duty Cycle Would You Like (0-100)? 56
What Duty Cycle Would You Like (0-100)? 58
What Duty Cycle Would You Like (0-100)? 90
What Duty Cycle Would You Like (0-100)? 95
```
Please try this test, too:
```
import Adafruit_BBIO.PWM as PWM
myPWM="P9_16"
PWM.start(myPWM, 0, 1000)
for i in range(0,5):
  DC=input("What Duty Cycle Would You Like (0-100)? ")
  PWM.set_duty_cycle(myPWM, DC)
PWM.stop(myPWM)
PWM.cleanup()
```

RobertCNelson, 2017-10-11:
@tomek242 

```
root@beaglebone:~# config-pin -q p9.16
P9_16 pinmux file not found!
Cannot read pinmux file: /sys/devices/platform/ocp/ocp*P9_16_pinmux/state
```
This menas you don't have cape universal loaded..

For "4.4.72-bone-rt-r17" this i pre-u-boot-overlays

ah, and "2016-06-15" was wheezy, so u-boot overlays is going to be hard to enable, just too much to hack to get up to date..

You'll have better luck just grabbing the latest image from:
https://beagleboard.org/latest-images

which is: Debian 9.2 2017-10-10 4GB SD IoT

Regards,

pdp7, 2017-10-12:
@RobertCNelson thanks

@tomek242 Please try the [current BeagleBoard.org image](https://beagleboard.org/latest-images):
**[bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz](https://debian.beagleboard.org/images/bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz)**

It should have Linux 4.4 kernel installed.  Do not upgrade to the Linux 4.9 kernel as there is a PWM bug (#170)

silver2row, 2017-10-12:
Hello Tomek,

I got the PWM and GPIO working on kernel 4.4.x. I am going to test the 4.9.x kernel now. 

Seth

P.S. Pastebin you software or send it on this board with spacing each line four times. This way it will be easier to read. I will get back to you on what happens with the 4.9.x kernel.

tomek242, 2017-10-12:
Hello All.
All right today i will check bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz
Is that debian has some GUI?

pdp7, 2017-10-12:
@tomek242 Yes, try this image: [Debian 9.1 2017-08-31 4GB SD LXQT](http://debian.beagleboard.org/images/bone-debian-9.1-lxqt-armhf-2017-08-31-4gb.img.xz)

tomek242, 2017-10-12:
Ok Thanks, i try it when i back from work. :)

tomek242, 2017-10-12:
Hello Everyone.
Last Debian Image short raport:

After install adafruit-beaglebone-io-python

root@beaglebone:~# uname -a
Linux beaglebone 4.4.91-ti-r133 #1 SMP Tue Oct 10 05:18:08 UTC 2017 armv7l GNU/Linux

Important (As I think) - both of this codes seems work properly once time always after reset :) Well, some little succes:) But not enough for series projects :)

First test result:

Code:
import Adafruit_BBIO.PWM as PWM
import time

pin1 = "P9_14"
pin2 = "P9_16"
pin3 = "P8_19"
pin4 = "P8_13"

PWM.start(pin1,0)
PWM.start(pin2,0)
PWM.start(pin3,0)
PWM.start(pin4,0)

for i in range(0,100):
        PWM.set_duty_cycle(pin1,100)
        PWM.set_duty_cycle(pin2,100)
        PWM.set_duty_cycle(pin3,100)
        PWM.set_duty_cycle(pin4,100)

time.sleep(.1)
PWM.stop(pin1)
PWM.stop(pin2)
PWM.stop(pin3)
PWM.stop(pin4)

PWM.cleanup()

Result:

[  604.223556] ehrpwm 48304200.pwm: Period value conflicts with channel 1
Traceback (most recent call last):
  File "test1.py", line 12, in <module>
    PWM.start(pin3,0)
RuntimeError: Problem with a sysfs file

Second code:

import Adafruit_BBIO.PWM as PWM
myPWM="P9_14"
PWM.start(myPWM, 0, 1000)
for i in range(0,5):
    DC=input("What Duty Cycle Would You Like (0-100)? ")
    PWM.set_duty_cycle(myPWM, DC)
PWM.stop(myPWM)
PWM.cleanup()

Result:

[  736.749310] ehrpwm 48302200.pwm: Period value conflicts with channel 1
Traceback (most recent call last):
  File "testPWM.py", line 3, in <module>
    PWM.start(myPWM, 0, 1000)
RuntimeError: Problem with a sysfs file

pdp7, 2017-10-13:
@tomek242 thanks for following up with additional information.
**Are you running the Python as root with sudo?**

silver2row, 2017-10-13:
Hey Everyone, 

I thought we had to do this and then functionality to PWM:

    import Adafruit_BBIO.GPIO as GPIO
    import Adafruit_BBIO.PWM as PWM
    import time

and then...the fancy stuff.

If I am wrong, sorry for butting in again.

Seth

tomek242, 2017-10-13:
pdp7 - Drew i used both method to typing commands:
from user account [user: debian / password: temppwd] with sudo command,
and from root account [user: root / password: root] (sudo its not neccesary in this case).

Seth In Python you must import library first (Adafruit) and then fancy stuff. Maybe other languages has diffirent solution.

pdp7, 2017-10-13:
@tomek242 I ran this code:
```
import Adafruit_BBIO.PWM as PWM
import time

pin1 = "P9_14"
pin2 = "P9_16"
pin3 = "P8_19"
pin4 = "P8_13"

PWM.start(pin1,0)
PWM.start(pin2,0)
PWM.start(pin3,0)
PWM.start(pin4,0)

for i in range(0,100):
  PWM.set_duty_cycle(pin1,100)
  PWM.set_duty_cycle(pin2,100)
  PWM.set_duty_cycle(pin3,100)
  PWM.set_duty_cycle(pin4,100)

time.sleep(.1)
PWM.stop(pin1)
PWM.stop(pin2)
PWM.stop(pin3)
PWM.stop(pin4)

PWM.cleanup()
```
and recieved the same error:
```
debian@beaglebone:~$ sudo python issue169test1.py
Traceback (most recent call last):
  File "issue169test1.py", line 9, in <module>
    PWM.start(pin1,0)
RuntimeError: Problem with a sysfs file
```

I then ran it with strace:
```
sudo strace -f -o /tmp/issue169test1.txt python issue169test1.py
```

These lines are of interest:
```
8280  open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/period", O_RDWR) = 4
8280  open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/duty_cycle", O_RDWR) = 5
8280  open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/polarity", O_RDWR) = 6
8280  open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/enable", O_RDWR) = 7
8280  lseek(7, 0, SEEK_SET)             = 0
8280  read(7, "1", 1)                   = 1
8280  lseek(7, 0, SEEK_SET)             = 0
8280  write(7, "0", 1)                  = 1
8280  lseek(6, 0, SEEK_SET)             = 0
8280  write(6, "normal", 6)             = 6
8280  lseek(7, 0, SEEK_SET)             = 0
8280  write(7, "1", 1)                  = 1
8280  lseek(4, 0, SEEK_SET)             = 0
8280  read(4, "1000000\n", 20)          = 8
8280  lseek(5, 0, SEEK_SET)             = 0
8280  read(5, "0\n", 20)                = 2
8280  lseek(5, 0, SEEK_SET)             = 0
8280  write(5, "0", 1)                  = 1
8280  lseek(4, 0, SEEK_SET)             = 0
8280  write(4, "500000", 6)             = -1 EINVAL (Invalid argument)
8280  gettimeofday({1507874940, 541710}, NULL) = 0
```

It appears the issue is writing `500000` to `/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/period` returns `-1 EINVAL (Invalid argument)`

@RobertCNelson  any ideas?

tomek242, 2017-10-13:
pdp7 my project was described here
[http://myroboticadventure.blogspot.com/2014/05/my-first-working-robot-its-alive.html](url)
but in my case i planned to use 4 drive chassis with four encoders and diffirent, other autonomous system to drive my project.
A ask author about my problem but He don't answer my question for some other reason. Anyway as You can see They use only simple stock Angstrom Linux selling with this board.
Of course i try my test codes with this soft but i had that same errors. So Im stack:)
Greetings.

pdp7, 2017-10-13:
@tomek242 ah, I just realized this error is in `dmesg` corresponding to the failed write to the `period` file:
> [164585.592970] ehrpwm 48302200.pwm: Period value conflicts with channel 1

@RobertCNelson have you seen that issue recently?

tomek242, 2017-10-13:
I think that every Debian image for that board has some errors. :( Using program for one PWM output is cool but unusable in practic projects.

tomek242, 2017-10-16:
Howdy Gentelmen:)
Do You hahe some information or ideas about errosr on BBB?
Greetings.
Thomas

RobertCNelson, 2017-10-17:
```
It appears the issue is writing 500000 to /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/period returns -1 EINVAL (Invalid argument)
```
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/pwm/pwm-tiehrpwm.c?h=v4.14-rc5#n56

PERIOD_MAX=0xFFFF = 65535

Regards,

RobertCNelson, 2017-10-17:
```
	/*
	 * Period values should be same for multiple PWM channels as IP uses
	 * same period register for multiple channels.
	 */
```

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/pwm/pwm-tiehrpwm.c?h=v4.14-rc5#n56

Regards,

pdp7, 2017-10-17:
@tomek242 please try running this command:
```sudo apt update ; sudo apt install bb-cape-overlays --reinstall```

pdp7, 2017-10-17:
@tomek242 if you still have issue after re-installing bb-cape-overlays, then please try this image that @RobertCNelson created on October 15th: [2017-10-15_-_Debian_9_.28Stretch.29_-_Weekly](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#2017-10-15_-_Debian_9_.28Stretch.29_-_Weekly)

And then install the Adafruit_BBIO from `master` branch in this repo.

pdp7, 2017-10-18:
@tomek242  Please try this new Debian image from @RobertCNelson:
https://rcn-ee.net/rootfs/bb.org/testing/2017-10-17/stretch-iot/bone-debian-9.2-iot-armhf-2017-10-17-4gb.img.xz

tomek242, 2017-10-18:
Ok. Results
root@beaglebone:~# uname -a
Linux beaglebone 4.4.91-ti-r133 #1 SMP Tue Oct 10 05:18:08 UTC 2017 armv7l GNU/Linux

Code:
> 
#!/usr/bin/python
import Adafruit_BBIO.PWM as PWM
import time
pin = "P8_14"
PWM.start(pin, 0)
for i in range(0, 100):
        PWM.set_duty_cycle(pin, 100)
time.sleep(.1)
PWM.stop(pin)
PWM.cleanup()

When I try for example P8_16 everything seems good but something is wrong in line:
for i in range(0, 100): - normally should rotate wheel of my robot for 10 seconds (If I think properly), but it isn't. wheel rotate only about second. So it's bizarre:)

Second result: when i try to change the pin on - for example P8_16 the same wheel is rotated the same way as before or - two wheels runs together and one of wheels is spinning and spinning and I have to do a reset. As if BBB was hanging.

Thirt result: After typing:
> sudo apt update ; sudo apt install bb-cape-overlays --reinstall

I have this after trying run my test program whatever pin I trying to use:
> root@beaglebone:~/my_python# python test2.py
Traceback (most recent call last):
  File "test2.py", line 7, in <module>
    PWM.start(pin1,0)
ValueError: Invalid PWM key or name.

RobertCNelson, 2017-10-18:
@tomek242 you'll need to reboot after updating bb-cape-overlays..  Can you please share "test2.py" and i'll double check the overlays..

Regards,

tomek242, 2017-10-18:
Hello Robert.
Code of "test2.py" it's above:)
>#!/usr/bin/python
import Adafruit_BBIO.PWM as PWM
import time
pin = "P8_14"
PWM.start(pin, 0)
for i in range(0, 100):
PWM.set_duty_cycle(pin, 100)
time.sleep(.1)
PWM.stop(pin)
PWM.cleanup()

RobertCNelson, 2017-10-18:
@tomek242 but P8_14 isn't a pwm pin..

```
typedef struct pins_t { 
    const char *name; 
    const char *key; 
    int gpio;
    int pwm_mux_mode;
    int ain;
} pins_t;

  { "EHRPWM2B", "P8_13", 23, 4, -1},
  { "GPIO0_26", "P8_14", 26, -1, -1},
```
it can be used as "ehrpwm2_tripzone_in"...

Regards,

tomek242, 2017-10-18:
Sorry my mistake, just Im tired:)
Of course I mean pin's P8_13, P8_19, P9_14, P9_16

tomek242, 2017-10-18:
Ok next test for Debian:
> root@beaglebone:~# uname -a
Linux beaglebone 4.9.56-ti-r70 #1 SMP PREEMPT Fri Oct 13 14:05:49 UTC 2017 armv7l GNU/Linux

This same code as above:) and result:
> root@beaglebone:~/my_python# python test2.py
Traceback (most recent call last):
  File "test2.py", line 7, in <module>
    PWM.start(pin1,0)
RuntimeError: Problem with a sysfs file

I hope to help my tests a little bit to sove this...
Greetings.

RobertCNelson, 2017-10-18:
My logic anayliser has decided to fail on me today, so i've verify the actual P8_13 pin.

```
debian@beaglebone:~$ uname -r
4.9.56-ti-r70
debian@beaglebone:~$ sudo python pwm_p8_13.py 
[sudo] password for debian: 
debian@beaglebone:~$ cat pwm_p8_13.py 
#!/usr/bin/python
import Adafruit_BBIO.PWM as PWM
import time
pin = "P8_13"
PWM.start(pin, 0)
for i in range(0, 100):
  PWM.set_duty_cycle(pin, 100)
time.sleep(.1)
PWM.stop(pin)
PWM.cleanup()
```
Looks to me, like it's something i already fixed and is in the master repo of this project..

RobertCNelson, 2017-10-18:
okay, it took 3 pc's but P8_13 is showing an output, with 4.9.56-ti-r70

https://rcn-ee.net/rootfs/bb.org/testing/2017-10-17/stretch-iot/

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[6c2688b0be448b7bb9ca18849b430d496a84acb4]
eeprom:[A335BNLT000C1727BBBG1071]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.9.56-ti-r70]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171013.2-0rcnee1~stretch+20171013]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

pdp7, 2017-10-19:
@RobertCNelson thanks

@tomek242 please paste the output of this:
`sudo /opt/scripts/tools/version.sh`

also please install the latest version of Adafruit_BBIO from this git repo:
```
cd /opt/source/adafruit-beaglebone-io-python/
git pull 
sudo make install
```

tomek242, 2017-10-19:
All right today a will check everything.

tomek242, 2017-10-19:
Meantime i check voltage of pins. When i run test code with all pins as I show above a had  3,27V on pins P9_14; P8_19; P8_13, on pin P9_16 I had 4,4mV so its low or unsteadiness state of signal.

When I try to fired up each pin separately I have:
P9_14 - 1,9 mV and sometimes I have 3,27V and sometimes hanging up on this pinout
P9_16 - 4,4 mV and nothing Hi state.
P8_13 - 3,27V and looks good except problem with declare of "for i in range(0, 100)" as I described above
P8_19 - 3,27V and rest like pin P8_13.

pdp7, 2017-10-20:
@tomek242 Could you please paste the python program with the "Insert Code" button.  I can not tell from above what is supposed to be inside your for loop as there is no indentation:
```
import Adafruit_BBIO.PWM as PWM
import time
pin1 = "P9_14"
pin2 = "P9_16"
pin3 = "P8_19"
pin4 = "P8_13"
PWM.start(pin1,0)
PWM.start(pin2,0)
PWM.start(pin3,0)
PWM.start(pin4,0)
for i in range(0, 100):
PWM.set_duty_cycle(pin1,100)
PWM.set_duty_cycle(pin2,100)
PWM.set_duty_cycle(pin3,100)
PWM.set_duty_cycle(pin4,100)
time.sleep(.1)
# what is supposed to be inside the loop?
PWM.stop(pin1)
PWM.stop(pin2)
PWM.stop(pin3)
PWM.stop(pin4)
PWM.cleanup()
```

pdp7, 2017-10-20:
@tomek242 in regards to your comment above:
> When I try for example P8_19 everything seems good but something is wrong in line:
> for i in range(0, 100): - normally should rotate wheel of my robot for 10 seconds (If I think properly), but it isn't. Wheel rotate only about second. So it's bizarre

I don't understand why you are trying to run that loop 100 times.  The `PWM.set_duty_cycle()` is set to `100` each time so I am not sure what the purpose of the loop is.  If you want to delay a for a longer periond, then just make `time.sleep()` longer.

pdp7, 2017-10-20:
@tomek242 Please also paste the output of this command:
`journalctl |grep python`
There is debugging output from Adafruit_BBIO that may be in the system log.

tomek242, 2017-10-20:
Hello Drew.
>
I don't understand why you are trying to run that loop 100 times. The PWM.set........

That is because only for test I wrote this little code. How to write in English? - For example I want turn wheel in my robot for 10 seconds so, (if I think propelry) that loop should be do this when i type
>
for i in range(0, 100):.

> PWM.set_duty_cycle(pin1, 100)
Thats another thing.
In this parameter i set only percent of duty that means 100(%) means 3,3V; 50(%) means 1,6V etc. - which translates into wheel speed.
 Im sorry, i do not know how to explain it to You because I'm missing words :(
 P.S. I corrected "code":)
Greetings
Tom

pdp7, 2017-10-20:
There should be no need for a `for` loop.  You should be able to set the duty cycle of the PWM pin and then have the Python program sleep for 10 seconds like this:
```
PWM.set_duty_cycle(pin1, 100)
sleep(10) # Time in seconds.
```

> Thats another thing. In this parameter i would like set only percent of duty - that means 100(%) means 3,3V; 50(%) means 1,6V etc.... - which translates into wheel speed.

That should be the result you get with code like:
```
PWM.set_duty_cycle(pin1, 100)
sleep(10) # Time in seconds.
PWM.set_duty_cycle(pin1, 50) 
sleep(10) # Time in seconds.
```

tomek242, 2017-10-20:
All right it's clear for me:) But the problem with the pins remains.

tomek242, 2017-10-20:
Hello All
I was take another test of equipment and simple software tests:

I was try this code for example:
```
#!/usr/bin/python
import Adafruit_BBIO.PWM as PWM
import time

pin1 = "P9_14"
pin2 = "P9_16"
pin3 = "P8_19"
pin4 = "P8_13"

PWM.start(pin1, 0)
PWM.start(pin2, 0)
PWM.start(pin3, 0)
PWM.start(pin4, 0)

PWM.set_duty_cycle(pin1, 100)
PWM.set_duty_cycle(pin2, 100)
PWM.set_duty_cycle(pin3, 100)
PWM.set_duty_cycle(pin4, 100)

time.sleep(20)

PWM.stop(pin1)
PWM.stop(pin2)
PWM.stop(pin3)
PWM.stop(pin4)

PWM.cleanup()
```
All pins P8_13, P8_19, P9_14 seems work properly and takes Hi state for 20 seconds, except P9_16 - nothing HI signal.

Really strange things start to happen when i try another code:
Code:
```
import Adafruit_BBIO.PWM as PWM
import time
pin = "P9_16"
PWM.start(pin, 0)

PWM.set_duty_cycle(pin, 100)
time.sleep(20)
PWM.stop(pin)
PWM.cleanup()
```
When I seting up pin on P9_16 and run program, i had HI state on P9_14 but low on pin which I declared in code above - i mean (P9_16) :)

So next I change code like this:
```
import Adafruit_BBIO.PWM as PWM
import time
pin = "P9_14"
PWM.start(pin, 0)

PWM.set_duty_cycle(pin, 100)
time.sleep(20)
PWM.stop(pin)
PWM.cleanup()
```
And for pin P9_14 i had Hi state on this pin and sometimes hang out and permanently HI state on this pin (3,27V).

Also i check P8_19 pin.
Code:
```
import Adafruit_BBIO.PWM as PWM
import time
pin = "P8_19"
PWM.start(pin, 0)

PWM.set_duty_cycle(pin, 100)
time.sleep(20)
PWM.stop(pin)
PWM.cleanup()
```
Result - HI on P8_19 for 20 seconds but also HI state on pin P8_13 and hang out this pin in Hi state -
Or hanging two of P8 header PWM pins.

When I try  this code for P8_13 pin everything seems work correct.
Ok thats all for today.
Thank you for helping.
Greetings.

tomek242, 2017-10-23:
Hello. Any suggestinons guys??

pdp7, 2017-10-24:
> All pins P8_13, P8_19, P9_14 seems work properly and takes Hi state for 20 seconds, except P9_16 - nothing HI signal.

@tomek242 please run this command which will print out the pinmux state of all pins:
```
for i in `find /sys/devices/platform/ocp/ -type f |grep pinmux |grep state$|sort -n`; do echo $i: `cat $i`; done
```

@RobertCNelson any thoughts on why P9_16 might be an issue?

tomek242, 2017-10-24:
Hello Drew. Thats my 'output'
```
/sys/devices/platform/ocp/ocp:P8_07_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_08_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_09_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_10_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_11_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_12_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_13_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_14_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_15_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_16_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_17_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_18_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_19_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_26_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_11_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_12_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_13_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_14_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_15_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_16_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_17_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_18_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_21_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_22_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_23_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_24_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_26_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_27_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_30_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_41_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_42_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_91_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_92_pinmux/state: default
```

pdp7, 2017-10-24:
@tomek242  it doesn't look like any are set to pwm.

Please run this:
```
config-pin -q p9.14
config-pin p9.14 pwm
config-pin -q p9.14

config-pin -q p9.16
config-pin p9.16 pwm
config-pin -q p9.16

config-pin -q p8.19
config-pin p8.19 pwm
config-pin -q p8.19

config-pin -q p8.13
config-pin p8.13 pwm
config-pin -q p8.13
```

tomek242, 2017-10-27:
Hello. Sorry for late.
Thats my output.
```
root@beaglebone:~# config-pin -q p9.14
P9_14 Mode: pwm
root@beaglebone:~# config-pin p9.14 pwm
root@beaglebone:~# config-pin -q p9.14
P9_14 Mode: pwm
root@beaglebone:~# config-pin -q p9.16
P9_16 Mode: pwm
root@beaglebone:~# config-pin p9.16 pwm
root@beaglebone:~# config-pin -q p9.16
P9_16 Mode: pwm
root@beaglebone:~# config-pin -q p8.19
P8_19 Mode: pwm
root@beaglebone:~# config-pin p8.19 pwm
root@beaglebone:~# config-pin -q p8.19
P8_19 Mode: pwm
root@beaglebone:~# config-pin -q p8.13
P8_13 Mode: pwm
root@beaglebone:~# config-pin p8.13 pwm
root@beaglebone:~# config-pin -q p8.13
P8_13 Mode: pwm
```

pdp7, 2017-10-28:
@tomek242 good to hear you've made progress
> Sometimes pin's hangs out, besides: At this time all pins seems work fine. :)

Could you please paste the code you are running?

I'm interested to see if I can replicate the "hanging" scenerio.  NOTE: Please use the `Insert code` button so that the whitespace is preserved.

[8 more replies]

closed 2017-10-30 by pdp7