adafruit/adafruit-beaglebone-io-python #170: Adafruit_BBIO PWM does not work Linux 4.9 kernel

pdp7, 2017-10-08:
## Issue
**Adafruit_BBIO PWM does not work Linux 4.9 kernel.**

I will update with more information about the issue and my progress as I work towards resolution.  I wanted to quickly log an issue so that the problem is visible to everyone.

## Workaround
**Use the Linux 4.4 kernel**
```
cd /opt/scripts/tools/
sudo git pull
sudo ./update_kernel.sh --lts-4_4
sudo reboot
```

silver2row, 2017-10-08:
Hello Sir,

Thank you for adopting this issue. I will stay patient. 

Seth

pdp7, 2017-10-08:
@silver2row please try switching to the Linux 4.4 kernel.  The issue with Flask and GPIO described in #168 may be unrelated but I'd like to eliminate this Linux 4.9 kernel issue from your situation.

silver2row, 2017-10-09:
Hello Sir,

Seth here. I tried to run that software on kernel 4.4.x and I received all sorts of issues. 

Seth

P.S. I am currently dealing with the Flask people on GitHub.com.

silver2row, 2017-10-09:
Hello,

I set up a question on Stack Overflow: https://stackoverflow.com/questions/46654985/re-how-can-i-get-flask-to-use-gpio-pins-to-change-the-led-brightness-on-linux-w.

Seth

P.S. If you are interested in looking at what has happened so far, go ahead. The link is listed above.

pdp7, 2017-10-10:
@silver2row 
> Seth here. I tried to run that software on kernel 4.4.x and I received all sorts of issues.

Please paste the any errors or warning here so that I can them.

silver2row, 2017-10-11:
Hello...power constraints @pdp7 .

Sorry. I feel bad now. I was trying to power everything, one sensor and an LED, with the 3.3v. I think your Adafruit_BBIO works and actually, I know it does.

I am foolish at times and forgot about power issues.

Seth

P.S. I am very sorry. Please forgive me.

pdp7, 2017-10-11:
@silver2row Thank you for making me take a closer look at PWM.  There is a real bug with PWM in Adafruit_BBIO on Linux 4.9 kernel.  I am able to reproduce it and will post more details here soon.

pdp7, 2017-10-17:
Workaround by @RobertCNelson in https://github.com/adafruit/adafruit-beaglebone-io-python/commit/c35e4cb98a1f14c85aca7259132bcc97e93d78f8

pdp7, 2017-10-18:
@RobertCNelson pytest looking pretty good on the new `BeagleBoard.org Debian Image 2017-10-17` image

## pytest
note: install pytest with `sudo pip install pytest`
```
debian@beaglebone:/opt/source/adafruit-beaglebone-io-python$ sudo pytest test/*
====================================================== test session starts =======================================================
platform linux2 -- Python 2.7.13, pytest-3.2.3, py-1.4.34, pluggy-0.4.0
rootdir: /opt/source/adafruit-beaglebone-io-python, inifile:
collected 58 items                                                                                                                

test/test_adc.py ......
test/test_gpio_input.py ..
test/test_gpio_output.py ......
test/test_gpio_setup.py ..........
test/test_led.py ..
test/test_pwm_setup.py ..F........................
test/test_spi.py ..
test/test_uart.py ...

============================================================ FAILURES ============================================================
_________________________________________ TestPwmSetup.test_start_pwm_with_polarity_one __________________________________________

self = <test_pwm_setup.TestPwmSetup instance at 0xb640da08>

    def test_start_pwm_with_polarity_one(self):
        PWM.cleanup()
        PWM.start("P9_14", 0, 2000, 1)
    
        pwm_dir = get_pwm_dir()
    
        assert os.path.exists(pwm_dir)
        if kernel >= '4.1.0':
            duty = open(pwm_dir + '/duty_cycle').read()
        else:
            duty = open(pwm_dir + '/duty').read()
        period = open(pwm_dir + '/period').read()
        polarity = open(pwm_dir + '/polarity').read()
        assert int(duty) == 0
        assert int(period) == 500000
    
        if kernel >= '4.1.0':
>           assert polarity == "inversed\n"
E           AssertionError: assert 'normal\n' == 'inversed\n'
E             - normal
E             + inversed

test/test_pwm_setup.py:100: AssertionError
============================================== 1 failed, 57 passed in 12.84 seconds ==============================================
```

## Version:
```
debian@beaglebone:/opt/source/adafruit-beaglebone-io-python$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[6c2688b0be448b7bb9ca18849b430d496a84acb4]
eeprom:[A335BNLT000C3014BBBK1316]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-rc2-00002-g84a7f2]
kernel:[4.4.91-ti-r133]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171013.2-0rcnee1~stretch+20171013]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

pdp7, 2017-10-18:
pytest completes OK now after commit d9be5f9610972493a39cc03b3cd94691c28c46ad

```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo pytest
====================================================== test session starts =======================================================
platform linux2 -- Python 2.7.13, pytest-3.2.3, py-1.4.34, pluggy-0.4.0
rootdir: /home/debian/ssh/adafruit-beaglebone-io-python, inifile:
collected 58 items                                                                                                                

test/test_adc.py ......
test/test_gpio_input.py ..
test/test_gpio_output.py ......
test/test_gpio_setup.py ..........
test/test_led.py ..
test/test_pwm_setup.py ...........................
test/test_spi.py ..
test/test_uart.py ...

=================================================== 58 passed in 11.88 seconds ===================================================
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ 
````

pdp7, 2017-10-20:
Without @RobertCNelson "hack" fix, odd sequence of events:
```
debian@beaglebone:~$ sudo python ~/pwm-polarity.py
usleep(1000)
sleep
sleep
pwm_set_polarity: write failed
Traceback (most recent call last):
  File "/home/debian/pwm-polarity.py", line 4, in <module>
    PWM.start("P9_14", 50)
RuntimeError: Problem with a sysfs file

debian@beaglebone:~$ sudo  ~/bash.sh 

debian@beaglebone:~$ sudo python ~/pwm-polarity.py
usleep(1000)
sleep
sleep
usleep(1000)
sleep
sleep
```

## pwm-polarity.py
```
import Adafruit_BBIO.PWM as PWM 
#PWM.start(channel, duty, freq=2000, polarity=0) 
#duty values are valid 0 (off) to 100 (on) 
PWM.start("P9_14", 50)
PWM.set_duty_cycle("P9_14", 25.5) 
PWM.set_frequency("P9_14", 10)
PWM.stop("P9_14")
PWM.cleanup()
    
#set polarity to 1 on start:
PWM.start("P9_14", 50, 2000, 1)
```

## bash.sh
```
cd /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0
echo 20000 > period 
echo 10000 > duty_cycle
echo normal > polarity 
#echo normal > inversed 
echo 1 > enable 
```

silver2row, 2017-10-21:
Hello,

I am glad you two are still grinding gears over here. I got caught up w/ the Motor Bridge Cape. Let me know if you or the other fellow need anything. I have time and actually lots of it.

Seth

pdp7, 2017-10-24:
I'm going to follow up with PWM maintainer Boris from Free Eletrons.  Robert suggested I contact him as this does seem to be some sort of regression
> it smells like a race condition, v4.9.x was the first to bring in "atomic" pwm 
https://lwn.net/Articles/683893/

silver2row, 2017-10-27:
@pdp7 , 

Hello Sir...I read over the info. and I think maybe bringing in a second person on this situation is better, esp. since the link you gave has a gitHub.com site that is non-existent now. 

Seth

P.S. Are we in trouble for discussing the subject matter of this PWM condition on this site? I saw a message in the GitHub.com issues section. Someone is complaining about our concerns w/ this issue.

pdp7, 2017-10-31:
I've created branch [pwm-issue170](https://github.com/adafruit/adafruit-beaglebone-io-python/compare/pwm-issue170) to remove @RobertCNelson's polarity "hack" fix and experiment with a solution to reliably setting polarity in the Linux 4.9

pdp7, 2017-10-31:
For comparison, this is how bash interacts with the /sys polarity file and it runs OK.
### test.sh
```
cd /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0
echo 20000 > period 
echo 10000 > duty_cycle
echo normal > polarity 
echo 1 > enable 
```
### strace output
```
chdir("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0") = 0
open("period", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666) = 3
fcntl64(1, F_GETFD)                     = 0
fcntl64(1, F_DUPFD, 10)                 = 10
fcntl64(1, F_GETFD)                     = 0
fcntl64(10, F_SETFD, FD_CLOEXEC)        = 0
dup2(3, 1)                              = 1
close(3)                                = 0
write(1, "20000\n", 6)                  = 6
dup2(10, 1)                             = 1
fcntl64(10, F_GETFD)                    = 0x1 (flags FD_CLOEXEC)
close(10)                               = 0
open("duty_cycle", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666) = 3
fcntl64(1, F_GETFD)                     = 0
fcntl64(1, F_DUPFD, 10)                 = 10
fcntl64(1, F_GETFD)                     = 0
fcntl64(10, F_SETFD, FD_CLOEXEC)        = 0
dup2(3, 1)                              = 1
close(3)                                = 0
write(1, "10000\n", 6)                  = 6
dup2(10, 1)                             = 1
fcntl64(10, F_GETFD)                    = 0x1 (flags FD_CLOEXEC)
close(10)                               = 0
open("polarity", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666) = 3
fcntl64(1, F_GETFD)                     = 0
fcntl64(1, F_DUPFD, 10)                 = 10
fcntl64(1, F_GETFD)                     = 0
fcntl64(10, F_SETFD, FD_CLOEXEC)        = 0
dup2(3, 1)                              = 1
close(3)                                = 0
write(1, "normal\n", 7)                 = 7
dup2(10, 1)                             = 1
fcntl64(10, F_GETFD)                    = 0x1 (flags FD_CLOEXEC)
close(10)                               = 0
open("enable", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666) = 3
fcntl64(1, F_GETFD)                     = 0
fcntl64(1, F_DUPFD, 10)                 = 10
fcntl64(1, F_GETFD)                     = 0
fcntl64(10, F_SETFD, FD_CLOEXEC)        = 0
dup2(3, 1)                              = 1
close(3)                                = 0
write(1, "1\n", 2)                      = 2
dup2(10, 1)                             = 1
fcntl64(10, F_GETFD)                    = 0x1 (flags FD_CLOEXEC)
close(10)  
```

pdp7, 2017-10-31:
Commit 8e08a40f1411378e60e942b4efeb24984203dd50 moved `pwm_set_polarity()` after `pwm_set_frequency()` and `pwm_set_duty_cycle()`.  `PWM.start()` now runs OK after reboot:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat ~/pwm-polarity.py
import Adafruit_BBIO.PWM as PWM 
PWM.start("P9_14", 50, 2000, 1)
PWM.cleanup()

debian@beaglebone:~$ sudo strace -o /tmp/y python ~/pwm-polarity.py 
[sudo] password for debian: 
pwm_set_frequency()
pwm_set_duty_cycle()
pwm_set_polarity()
write 1 to pwm->enable_fd
```

strace:
```
open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/period", O_RDWR) = 4
open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/duty_cycle", O_RDWR) = 5
open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/polarity", O_RDWR) = 6
open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0/enable", O_RDWR) = 7
lseek(4, 0, SEEK_SET)                   = 0
read(4, "0\n", 20)                      = 2
lseek(5, 0, SEEK_SET)                   = 0
read(5, "0\n", 20)                      = 2
lseek(4, 0, SEEK_SET)                   = 0
write(4, "500000", 6)                   = 6
lseek(5, 0, SEEK_SET)                   = 0
write(5, "250000", 6)                   = 6
lseek(5, 0, SEEK_SET)                   = 0
write(5, "250000", 6)                   = 6
lseek(7, 0, SEEK_SET)                   = 0
read(7, "0", 1)                         = 1
lseek(6, 0, SEEK_SET)                   = 0
write(6, "inversed", 8)                 = 8
lseek(7, 0, SEEK_SET)                   = 0
write(7, "1", 1)                        = 1
lseek(7, 0, SEEK_SET)                   = 0
write(7, "1", 1)                        = 1
lseek(7, 0, SEEK_SET)                   = 0
write(7, "0", 1)                        = 1
close(4)                                = 0
close(5)                                = 0
close(6)                                = 0
```

pdp7, 2017-10-31:
#184 has been merged.  `PWM.start()` now runs OK.   `pwm_set_polarity()` has been moved after `pwm_set_frequency()` and `pwm_set_duty_cycle()`.

pytest runs OK:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python/test$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[6c2688b0be448b7bb9ca18849b430d496a84acb4]
eeprom:[A335BNLT000C3014BBBK1316]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-rc2-00002-g84a7f2]
kernel:[4.9.57-ti-r71]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171013.2-0rcnee1~stretch+20171013]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

debian@beaglebone:~/ssh/adafruit-beaglebone-io-python/test$ sudo python -mpytest 
================================================== test session starts ==================================================
platform linux2 -- Python 2.7.13, pytest-3.2.3, py-1.4.34, pluggy-0.4.0
rootdir: /home/debian/ssh/adafruit-beaglebone-io-python, inifile:
collected 58 items                                                                                                       

test_adc.py ......
test_gpio_input.py ..
test_gpio_output.py ......
test_gpio_setup.py ..........
test_led.py ..
test_pwm_setup.py ...........................
test_spi.py ..
test_uart.py ...

============================================== 58 passed in 13.53 seconds ===============================================
```

closed 2017-10-31 by pdp7