adafruit/adafruit-beaglebone-io-python #171: Adafruit_BBIO PWM does not work P9_22 and P9_42. (ECAPPWM0, ECAPPWM2) How can i solve that? Help me pls.

Altantsooj, 2017-10-09:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

Altantsooj, 2017-10-09:
I need 8 PWM port need. But just 6 PWM working...

silver2row, 2017-10-09:
Hello,

Do you have kernel 4.9.x?

Seth

P.S. See here: #170.

pdp7, 2017-10-10:
@Altantsooj I'd like to get some information about what is currently running on your BeagleBone. 

**Please paste the output of the following commands:**
```
cat /etc/dogtag 
cat /etc/debian_version 
cat /proc/cmdline 
cat /sys/devices/platform/bone_capemgr/slots
cat /boot/uEnv.txt 
config-pin -q p9.22
config-pin -q p9.42
uname -a
```

Altantsooj, 2017-10-10:
root@beaglebone:~# cat /etc/dogtag
BeagleBoard.org Debian Image 2017-09-21

root@beaglebone:~# cat /etc/debian_version
9.1
root@beaglebone:~# cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet

root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory

root@beaglebone:~# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.88-ti-r125
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

root@beaglebone:~# config-pin -q p9.28
P9_28 pinmux file not found!
Cannot read pinmux file: /sys/devices/platform/ocp/ocp*P9_28_pinmux/state

root@beaglebone:~# config-pin -q p9.42
P9_42 Mode: pwm

root@beaglebone:~# uname -a
Linux beaglebone 4.4.88-ti-r125 #1 SMP Thu Sep 21 19:23:24 UTC 2017 armv7l GNU/Linux

Altantsooj, 2017-10-10:
Sorry my mistake. ECAPPWM2 on P9_28. NOT P9_22.

RobertCNelson, 2017-10-10:
@Altantsooj 

P9.28 is hdmi-audio, in /boot/uEnv.txt add:

disable_uboot_overlay_audio=1

then reboot and try again.

Regards,

silver2row, 2017-10-11:
Hello Alt,

Seth here. Hey, you may need more power. I do not know what set-up you have but I think my set-up does not work due to power constraints. 

Seth

P.S. I tried the set-up and software to the "T" and changed things over and over. Power must be the issue. Try supplying 5v instead of 3.3v. This may work.

silver2row, 2017-10-12:
Forget the Power Constraints,

I got the software to work on kernel 4.4.x. I am updating to kernel 4.9.x and I will return service after checking if the software works. I got GPIO and PWM to work.

Seth

silver2row, 2017-10-13:
Hello Again,

I got put back to kernel 4.4.x. I got Adafruit_BBIO to work. I had to change up the software some but in the end, it worked for this specific kernel.

Seth

P.S. If you need help at all, drop me a line. Later for now.

pdp7, 2017-10-13:
@Altantsooj were you able to try @RobertCNelson's instructions?
> P9.28 is hdmi-audio, in /boot/uEnv.txt add:
> ```disable_uboot_overlay_audio=1```
> then reboot and try again.

pdp7, 2017-10-17:
@Altantsooj have you tried adding `disable_uboot_overlay_audio=1` to `/boot/uEnv.txt`?

pdp7, 2017-10-18:
@Altantsooj I am going to close this issue but please re-open if still a problem

closed 2017-10-18 by pdp7