adafruit/adafruit-beaglebone-io-python #172: BeagleBone Pocket pin names not known to source/common.c lookup_gpio_by_key()

randyrossi, 2017-10-12:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  BeagleBone Pocket

- Python version (run `python -version` or `python3 -version`):  N/A

- Error message you are receiving, including any Python exception traces: N/A

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): The beaglebone pocket uses pin conventions like P1_## or P2_## (as opposed to P8_## or P9_##) which are key names not recognized by the table that maps pin names to kernel gpio numbers in source/common.c  (specifically lookup_gpio_by_key() function).  On the pocket beagle, /usr/bin/config-pin does recognize these and provides the mapping info. But in order to program via python, this table needs to be updated for the pocket beagle.  I patched my local adafruit-beaglebone-io-python with my own mappings for the pins I wanted to use and it resolved my issue. So I think its just a matter of adding the P1_## P2_## entries and the pocket would be supported.

pdp7, 2017-10-12:
@randyrossi Thanks for opening this issue.  Support for the PocketBeagle has not been added yet.  I'm not able to give a good estimate right now as to when I will be able to implement it.  However, I would be happy to review any pull requests.

pdp7, 2017-10-17:
Initial support for PocketBeagle added by @RobertCNelson in #173

RobertCNelson, 2017-10-17:
@randyrossi 

first image build with this enablement..

https://rcn-ee.net/rootfs/bb.org/testing/2017-10-17/stretch-iot/bone-debian-9.2-iot-armhf-2017-10-17-4gb.img.xz

pdp7, 2017-10-24:
@randyrossi have you been able to try the image that @RobertCNelson linked above?

randyrossi, 2017-10-28:
Yes I did.  Just tried a simple gpio pin out and seems to work. Thx!

closed 2017-10-28 by randyrossi