adafruit/adafruit-beaglebone-io-python #174: I suspect GPO_3 is a typo and should be GP0_3 (numerical 0 instead of capital O)

randyrossi, 2017-10-13:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/e981c59c6b080781c493cbb72dae63d330243890/source/common.c#L172

RobertCNelson, 2017-10-13:
Correct that should be the number "0"...

silver2row, 2017-10-13:
Hello,

Nice catch. 

Seth

P.S. Is the Adafruit_BBIO having issues still?

RobertCNelson, 2017-10-13:
@silver2row which issue?

v4.9.x/v4.14.x kernels currently need this:

https://github.com/RobertCNelson/adafruit-beaglebone-io-python/commit/c35e4cb98a1f14c85aca7259132bcc97e93d78f8

In some situations u-boot overlays is getting stuck here:

https://github.com/RobertCNelson/adafruit-beaglebone-io-python/commit/4e180c54cfa0193da0b453f31f897d753dd9a4ea

Regards,

silver2row, 2017-10-14:
Oh...I was discussing 4.9.x. Okay, understood.

Seth

P.S. @RobertCNelson , I will follow those protocols to get things "cracking." Thank you.

closed 2017-10-17 by pdp7