adafruit/adafruit-beaglebone-io-python #178: version 1.0.4 and 1.0.5 are not fully compatible with Beaglebone Black

ramgu, 2017-10-24:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **Beaglebone Black/Debian 8.9**

- Python version (run `python -version` or `python3 -version`):  **python 2.7.9**

- Error message you are receiving, including any Python exception traces:  **Traceback (most recent call last):
  File "kk.py", line 3, in <module>
    GPIO.setup("P8_7", GPIO.OUT)
ValueError: Set gpio value failed, missing file or invalid permissions.***
syslog message:
"beaglebone python[2434]: gpio_set_value: 66 couldn't open '/sys/class/leds/red/brightness': 2-No such file or directory
"

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **run following python snippet:

  1 import Adafruit_BBIO.GPIO as GPIO
  2 
  3 GPIO.setup("P8_7", GPIO.OUT)
  4 GPIO.cleanup()
**

pdp7, 2017-10-24:
Thanks for opening.  This regression from the BeagleBone Blue support and I need to fix the logic.  The solution will be to identify which board Adafruit_BBIO is running on and then handle the pins appropriately.

pdp7, 2017-10-24:
I will need to add in logic to check for BeagleBone Blue and PocketBeagle:

**TI AM335x BeagleBone Blue**
```
grep -c 'TI AM335x BeagleBone Blue' /proc/device-tree/model (edited)
```
**TI AM335x PocketBeagle**
```
grep -c 'TI AM335x PocketBeagle' /proc/device-tree/model
```

pdp7, 2017-10-28:
I've created branch fix-issue-178 (5e2e785b519dab876f735724a63eef68a88f2ec0)

pdp7, 2017-10-28:
with commit becbc95861412d3b9016d1ba9f1cfb8007dabe4d in branch `fix-issue-178`:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat ~/issue178.py
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup("P8_7", GPIO.OUT)
GPIO.cleanup()

debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo python ~/issue178.py
gpio_set_value: uboot_overlay_enabled() is true
gpio_set_value: pocketbeagle() is FASLE
gpio_set_value: beaglebone_blue() is FALSE
gpio_set_value: default gpio path
gpio_set_value: filename=/sys/class/gpio/gpio66/value
```

pdp7, 2017-10-28:
@MarkAYoder could you test https://github.com/adafruit/adafruit-beaglebone-io-python/commit/becbc95861412d3b9016d1ba9f1cfb8007dabe4d on the BeagleBone Blue?

@RobertCNelson what considerations should I have for the PocketBeagle?

thanks!

Xperlack, 2017-10-30:
Hello.

I'm totally stocked on this. With two little differences:

- I get the error when trying to set a pin as input (e.g. `GPIO.setup("P9_12",GPIO.IN))` and

- it does work event if it's done as root.

I already read:

- [ ] 1. Issue #36 : [https://github.com/adafruit/adafruit-beaglebone-io-python/issues/36](url) and the merged issue #64 : [https://github.com/adafruit/adafruit-beaglebone-io-python/pull/64](url)

- [ ] 2. The issue_36's posting user relates it to: [https://stackoverflow.com/questions/18926979/beaglebone-black-adafruit-io-python-library-gpio-user-permissions](url).

- [ ] 3. And issue #165 : [https://github.com/adafruit/adafruit-beaglebone-io-python/issues/165](url)

So I've jumped into almost every file commented on this posts to figure out how the proposed solutions will match my problem. Just because I'm catious. But I didn't find the exact line to put this so referred delay (issue 36). 

So then I checked the *.patch option. Before patching, I just took a look on the code and what it does to  'py_gpio.c'. Then compare it with the actual code into 'py_gpio.c' file on my beaglebone. Surprise!!! - the code seems to be already handling the returns from export, setting directions and whatever. Then, I'm not sure the patch will solve something. So I DID NOT PATCH. Besides, I didn't like the huge counters polling the state of the /Sys directories.

Then I thought: ¬¬ man you just installed the OS and python libraries. This issues may be already solved on the last versions. So I came here and found out that yes, it was solved. Though is not working.

So to be sure what I installed the last updates I consulted again the tutorial [https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/installation-on-ubuntu](url) and made sure that the manual installation clones the files from this very exact repository. Although I did the 'apt-get' package manager installation.

On some point in the process I made the gpio setting directly on the /sys/class/gpio directories. Using `echo`, `export`, and all that sort of things. It worked fine, everything: output, input. Just wonderful. But of course I feel more like using the BBIO library facilities.

Then I met with the issue #165. It seems to be the newest!. I ran the solution proposed by some: RobertCNelson guy, which is: 

> `sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=100`

Nothing. It didn't work.

I'm stucked guys. Please help me.

pdp7, 2017-10-30:
@Xperlack please open up a new issue regarding the problem you are experiencing.  this issue is about regression effecting certain GPIO pins due to support added for non-standard LEDs on the BeagleBone Blue.

Xperlack, 2017-10-30:
Thanks pdp7. I'm sorry I got confused.

Before doing so I'll check an idea. May be it solves the problem.

pdp7, 2017-10-31:
These are all the LEDs defined in [source/event_gpio.h](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/event_gpio.h):
```
#define USR_LED_GPIO_MIN 53
#define USR_LED_GPIO_MAX 56
#define USR_LED_RED      66
#define USR_LED_GREEN    67
#define BAT25   27
#define BAT50   11
#define BAT75   61
#define BAT100  10000       // Placeholder until I find the real number
#define WIFI    10001       // Ditto
```
which map to entries in `pins_t table[]` in [source/common.c](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L72)

* 66
  * BeagleBone (not Blue): `{ "TIMER4", "P8_7", 66, -1, -1},`
  * BeagleBone Blue: `{ "RED_LED", "RED", 66, -1, -1}, // LEDs`

* 67
  * BeagleBone (not Blue):  `{ "TIMER7", "P8_8", 67, -1, -1},`
  * BeagleBone Blue:  `{ "GREEN_LED", "GREEN", 67, -1, -1},`

* 27
  * BeagleBone (not Blue): `{ "GPIO0_27", "P8_17", 27, -1, -1},`
  * BeagleBone Blue: `{ "BAT25", "BAT25", 27, -1, -1},`
  * PocketBeagle: `{ "GPIO0_27", "P2_19", 27, -1, -1},`

* 11
  * BeagleBone (not Blue): `{ "UART5_RTSN", "P8_32", 11, -1, -1},`
  * BeagleBone Blue: `{ "BAT50", "BAT50", 11, -1, -1},`

* 61
  * BeagleBone (not Blue): ` { "GPIO1_29", "P8_26", 61, -1, -1},`
  * BeagleBone Blue: ` { "BAT75", "BAT75", 61, -1, -1},`

NOTE: header reference [BBBP8Header.pdf](http://exploringbeaglebone.com/wp-content/uploads/2014/10/BBBP8Header.pdf) and [BBBP9Header.pdf](http://exploringbeaglebone.com/wp-content/uploads/2014/10/BBBP9Header.pdf)

pdp7, 2017-10-31:
Success with pytest:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python/test$ sudo python -mpytest 
================================================ test session starts =================================================
platform linux2 -- Python 2.7.13, pytest-3.2.3, py-1.4.34, pluggy-0.4.0
rootdir: /home/debian/ssh/adafruit-beaglebone-io-python, inifile:
collected 58 items                                                                                                    

test_adc.py ......
test_gpio_input.py ..
test_gpio_output.py ......
test_gpio_setup.py ..........
test_led.py ..
test_pwm_setup.py ...........................
test_spi.py ..
test_uart.py ...

============================================= 58 passed in 14.40 seconds =============================================
```

pdp7, 2017-10-31:
@RobertCNelson @MarkAYoder please try running `test/issue178.py` (from d75e0c8)

pdp7, 2017-10-31:
Merged #183 

@RobertCNelson @MarkAYoder please try running [test/issue178.py](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/test/issue178.py)

pdp7, 2017-10-31:
@ramgu `P8_7` is now working OK on BeagleBone Black:
```
debian@beaglebone:~$ python
Python 2.7.13 (default, Jan 19 2017, 14:48:08) 
[GCC 6.3.0 20170118] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.GPIO as GPIO
>>> GPIO.setup("P8_7", GPIO.OUT)
>>> GPIO.cleanup()
>>> 
debian@beaglebone:~$ cat /proc/device-tree/model 
TI AM335x BeagleBone Black
```

pdp7, 2017-11-01:
I am closing this issue as #183 has been merged.  Please re-open if any problems.

closed 2017-11-01 by pdp7

mvhall, 2018-02-08:
Has this issue now been marked as solved? I have a 9.1 LXQT version and tried the script to set outputs and inputs. Outputs work, but inputs not.
Also directly exporting the gpio's with the code below gives me an error 
OSError: [Errno 16] Device or resource busy
Actually the gpio pin is visible in /sys/class/gpio bus with user root and group root. I guess this is the problem causing the file.close() method to give a busy error. 
Am I the only one with this problem? Any solution?

pdp7, 2018-02-08:
@mvhall It would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):
```
sudo apt-get install strace
```
Then run strace, replace test.py with the name of your test program:
```
strace -f -o strace.log python test.py
```
Paste the contents of strace.log into a GitHub Gist and post the link in a reply here.

mvhall, 2018-02-09:
Will do so after my holliday. What I forgot to say is that the NHD7 (7" LCD from Newhaven) causes this, without the screen connected it works normal. 
The cape overlay seems to be the problem, but not 100% sure.

RobertCNelson, 2018-02-19:
The gpio subsystem has always required privileged access.  It's just recently (in the last year) we've gone thru each subsystem to make sure you don't have to be "root" to access them.. for the gpio, we have a gpio user that udev ties it to on bootup..

Regards,

mvhall, 2018-02-19:
Thanks Robert, this was not clear to me.
But what is the explanation that the adafruit library works on 9.1 LXQT without the  7" NHD LCD Cape, but not when connected?   

Any idea's looking at the gist I've just posted?

RobertCNelson, 2018-02-19:
@mvhall 

That combination doesn't work, as 1, i can find my NHD at my desk, and 2 i've been working in it for the last week or so.

If you want to test something:

Step 1, install this version of u-boot
```
sudo /opt/scripts/tools/developers/update_bootloader.sh --beta
```
Step 2: install this kernel:
```
sudo /opt/scripts/tools/update_kernel.sh --bone-channel --exp
```
Step 3: In progress (give me a moment)

RobertCNelson, 2018-02-19:
If you get:
```
debian@beaglebone:~$ dmesg | grep pin
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.217939] pinctrl core: initialized pinctrl subsystem
[    0.491998] omap_i2c 44e0b000.i2c: could not find pctldev for node /ocp/l4_wkup@44c00000/scm@210000/pinmux@800/pinmux_i2c0_pins, deferring probe
[    0.903922] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.216144] pinctrl-single 44e10800.pinmux: pin PIN15 already requested by panel; cannot claim for ocp:P8_15_pinmux
[    1.226715] pinctrl-single 44e10800.pinmux: pin-15 (ocp:P8_15_pinmux) status -22
[    1.234156] pinctrl-single 44e10800.pinmux: could not request pin 15 (PIN15) from group pinmux_P8_15_default_pin  on device pinctrl-single
[    1.246653] bone-pinmux-helper ocp:P8_15_pinmux: Error applying setting, reverse things back
[    1.255169] bone-pinmux-helper: probe of ocp:P8_15_pinmux failed with error -22
```
Rerun:

Step 3: Update overlays
```
cd /opt/source/bb.org-overlays/
git pull
make
sudo make install
```
Step 4: Reboot
```
sudo reboot
```

mvhall, 2018-02-19:
Hi thanks for your quick reply. I will test it tomorrow morning, first creating a new image of 9.1 LXQT.
Then I test it with NH7 LCD I guess?

RobertCNelson, 2018-02-19:
If your creating a new image anyways, might as well upgrade to the latest (9.3 lxqt)

https://beagleboard.org/latest-images

Regards,

mvhall, 2018-02-19:
Sweet, that wasn't there a few weeks ago. Ok will do so, with the NH7 Cape?

RobertCNelson, 2018-02-19:
yeah do everything with the cape plugged in.

mvhall, 2018-02-20:
Hi Robert,

I followed your steps with image 9.3LXQT. Needed to remove the LCD as it kept rebooting. After I installed the Bootloader and Overlays, I did a shutdown, installed the LCD Cape (NH7C) and booted. 
The GPIOxx ports in /sys/class/gpio were exported. 
Manually these (well I tested P9_12) can be set. Config-pin worked as well.
In python3 doing the following worked (got an error before). Note as you explained before, you need to use sudo python3.
>>>import sys
>>> f=open("/sys/class/gpio/export","w")
>>> f.write("%d" % (60))
>>> f.close()

Installed Adafruit_BBIO , and the GPIO.setup("P9_12", GPIO.IN) and GPIO.setup("P9_12", GPIO.OUT) worked, which I checked in /sys/class/gpio/gpio60 by using $cat direction. 
No need to add user debian to the gpio group.

So not fully tested, but my initial test which failed before now works. 
Oh not to forget: the LCD is working as well :-)
How does updating the information at Adafruit works? Can someone write a clear "test-case/instructions" for fellow developers? I will mail the folks at NewHaven myself to tell them the problem seems to be solved with your instructions above.  
Any gist needed here in this thread? Happy to provide the info, got it all documented. 
Thank you very much.

RobertCNelson, 2018-02-20:
Actually, 9.3LXQT has debian in the gpio group, that's why that worked. ;)

This is really just a new feature, something that's been on the list for awhile.  Something we figured out on the PocketBeagle capes.  Now it's just a matter of converting all to the "universal rewrite":

https://github.com/beagleboard/bb.org-overlays/tree/master/src/arm
https://github.com/beagleboard/bb.org-overlays/commits/master

it'll be the default when all capes are fully tested.

Regards,