adafruit/adafruit-beaglebone-io-python #180: Fixed length pin string names

PeteLawler, 2017-10-24:
So I'm of the habit that when I'm dealing with numbers or strings of generally a specified length, that I include leading 0's if needs be.

Thus in a numbering scheme where we go from 0 to 99, I'll tend to use 00, 01, 02, 03... etc. up to 99. This can make, if I need to, string manipulation easier.

After a month or two of tracking down a 'bug', I've finally nailed it. Can you see where I'm going?

Bug:
 SystemError: <built-in function output> returned NULL without setting an error

Cause:
GPIO.setup("P8_07", OUT )
Fix:
GPIO.setup("P8_7", OUT )

I'm wondering if, apart from within source, this should be noted on the README, or whether the source could be updated. I'm not sure about the latter because I don't quite know about external compatibilies you may be wishing to maintain. I'm happy to do any diff, just eager for any guidance....

I can't believe... ok I can... that it took this long to track down.

RobertCNelson, 2017-10-24:
@PeteLawler i know @jadonk wants to see these values converted to a mixed case option, with leading/no-leading 0's and mixed symbols: '.','_', etc..

https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L73-L261

and @pdp7 is working on splitting that single table up into a per board table.

Regards,

PeteLawler, 2017-10-24:
@RobertCNelson 
Fair enough.
I'll see about whether I can come up with some wording for the README or some error handling, or whether I'd prefer just to cower in the corner from guilt and shame and pretend none of the past month of bug hunting ever happened 😭

pdp7, 2017-10-24:
@PeteLawler that does seem awkward to have `P8_7` in the table instead of `P8_07`.
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L84
I suppose one way to prevent this issue is to accept both `P8_7` and `P8_07`.
For reference, `P8_07` does seem like the best choice as that is what pinmux helper driver is using:
`/sys/devices/platform/ocp/ocp:P8_07_pinmux`

RobertCNelson, 2017-10-24:
@pdp7 and config-pin already does this internal pin name fixup:

https://github.com/beagleboard/bb.org-overlays/blob/master/tools/beaglebone-universal-io/config-pin#L1127-L1164

Probably steal some of that syntax. ;)

Regards,

PeteLawler, 2017-10-24:
Ah great I was pretty sure I'd seen that nomenclature somewhere but after this adventure I really wondered. Had a brief squizz at the SRM, didn't see anything, then went back to coffee.
I can understand where @jadonk is coming from, though. Sticking with SRM nomenclature throughout would be best, no? Or maybe the SRM is wrong and should use fixed length strings.😁

RobertCNelson, 2017-10-25:
Considering the audience, i don't mind, P8_01/P8_1/P8.01/P8.1.. it's just implementing that in python.. ;)  bash is easy..

pdp7, 2017-11-28:
I've added documentation to the README that there is no '0' prefix for the pin numbers.

I'm going to close this issue but would be happy to review any pull requests that allow different naming schemes.

closed 2017-11-28 by pdp7