adafruit/adafruit-beaglebone-io-python #182: Problems with Cloud9 in reference to Issue #169

tomek242, 2017-10-30:
#169 
When I try same code which I present in Issue #169 I have the same error like bmcilw1:
Your code is running at http://localhost.
```
Important: use os.getenv(PORT, 8080) as the port and os.getenv(IP, 0.0.0.0) as the host in your scripts!

Traceback (most recent call last):
  File "/var/lib/cloud9/my_python/test1.py", line 10, in <module>
    PWM.start(pin1,0)
RuntimeError: Problem with a sysfs file
```

Question: Why the same code in Python work when I wrote it in simply nano editor and does not work in IDE Cloud9?

pdp7, 2017-10-30:
> Second thin is: how to set direction of PWM pins?

PWM signal is output only.  The pins do have other modes but PWM mode is always an output.

pdp7, 2017-10-30:
> Question: Why the same code in Python work when I wrote it in simply nano editor and does not work in IDE Cloud9?

Are you using this image: **[bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz](https://debian.beagleboard.org/images/bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz)**?

@RobertCNelson should there be any /sys permission problems with Cloud9 on the current image?

pdp7, 2017-10-31:
@tomek242 What is in `/var/lib/cloud9/my_python/`?  Did you create that directory?  I don't believe it should exist on a fresh image.

tomek242, 2017-10-31:
Hello Drew:)
Of course `/var/lib/cloud9/my_python/` its only my "files structure" :)

> @tomek242 Are you using this image: bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz?

Yes thats the same image, and i was try this same sample script :)
Btw. Robert C. Nelson take us some new images on his repository i mean images from 2017-10-29 - are they images ok? Or something is wrong with them? May I test it in my project without any risk?

> Second thin is: how to set direction of PWM pins?
> PWM signal is output only. The pins do have other modes but PWM mode is always an output.

Yes I know. Again I asked a bad question :-(  
I mean, how to configure the states on GPIO pins after system boot. I will use this pins to control wheels on my project.

bmcilw1, 2017-10-31:
Update: I've tried the image you guys suggested to me: https://github.com/adafruit/adafruit-beaglebone-io-python/issues/169#issuecomment-340249126.

When I use this image, PWM works 100% in BoneScript :). However, it does not work in Python via iCloud9 or when I run it from command line directly on the board (same error).

pdp7, 2017-10-31:
@tomek242 Sorry, I just realized that I edited your comment instead of replying to it.  Here are my responses.

> Yes thats the same image, and i was try this same sample script :)

Please paste the contents of `/var/lib/cloud9/my_python/test1.py` here with the `Insert code` button.  I will try running it in the Cloud9 IDE.

> Btw. Robert C. Nelson take us some new images on his repository i mean images from 2017-10-29 - are they images ok? Or something is wrong with them? May I test it in my project without any risk?

The best place to look for testing images is on this eLinux wiki page:
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Image_Testing_Snapshots

The entries on that page link to downloads on @RobertCNelson's server.

> I mean, how to configure the states on GPIO pins after system boot. I will use this pins to control wheels rotate on my project.

Which pins are you using?

Are you wishing to use those pins in GPIO mode?

If so, the `config-pin` utility can be used to set the pin mode.

pdp7, 2017-10-31:
@bmcilw1 Please paste the Python code you are running (use the `Insert code` button).

Also, I have just merged several fixes ( #183 and #184 )  so please install the latest version of Adafruit_BBIO from this repo.

bmcilw1, 2017-10-31:
I figured out what was going on with the sysfs file error. See my issue.

pdp7, 2017-10-31:
> How do I get the latest image from this repo @pdp7 ?

Please see https://github.com/adafruit/adafruit-beaglebone-io-python/issues/185#issuecomment-340865380

pdp7, 2017-11-02:
@tomek242 Linux 4.14 kernel should resolve the issue of non-root access to the PWM outputs.  Please try installing 4.14 kernel build by @RobertCNelson with:
```
sudo apt-get update
sudo apt-get upgrade
sudo /opt/scripts/tools/update_kernel.sh --lts-4_14
sudo reboot
```

pdp7, 2017-11-02:
@tomek242 please build and install the `master` branch of this repo:
```
sudo apt-get update
sudo apt-get install build-essential python-dev python-pip -y
git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
sudo python setup.py install
```
I have pushed commit b65cbf8e41b444bad7c4ef6cfd4f88a30210fd78 which causes `pwm_start()` to sleep 100 ms after export to avoid udev race condition (refer to #185).

pdp7, 2017-11-03:
@tomek242 I've published [release 1.0.6](https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.0.6) which should resolve this issue when using the Debian Stretch image with the 4.14.x kernel.

tomek242, 2017-11-04:
Hello. Sorry for the late reply. At now - config-pin utility solved my problrm with GPIO pis :)
Actually I use to my test:
```
root@beaglebone:~# uname -a
Linux beaglebone 4.9.59-ti-r73 #1 SMP PREEMPT Sat Oct 28 18:53:20 UTC 2017 armv7l GNU/Linux

root@beaglebone:~# uname -rv
4.9.59-ti-r73 #1 SMP PREEMPT Sat Oct 28 18:53:20 UTC 2017
```
I taken it from Robert C Nelson:
[https://rcn-ee.net/rootfs/bb.org/testing/](https://rcn-ee.net/rootfs/bb.org/testing/)
My test programs seems work good.
Drew You suggested that I should try your solution for kernel 4.14. Honestly, for now I just dont have much time to install another kernel and testing it:) I test Your solution Monday:)

tomek242, 2017-11-07:
Hello Drew.
I check all Images of our BBB since [Debian Stretch 9.2 image (2017-10-17)](https://rcn-ee.net/rootfs/bb.org/testing/2017-10-17/stretch-iot/) from @RobertCNelson's server.
In all cases everythings works good for me. Cloud9 also didn't have any problems. :)
All right, so far so good:) Big Thank You Drew and Robert:)
Greetings Tom:)

pdp7, 2017-11-08:
@tomek242 good to hear.  thanks for following up.

closed 2017-11-08 by pdp7