adafruit/adafruit-beaglebone-io-python #185: Problems with Adafruit_BBIO.PWM (reference issue #169)

bmcilw1, 2017-10-31:
I've had severe problems with PWM on my BBB. I've just re-flashed my board with the latest image, then ran apt-get update. My python code gets the same output stated above. Here's the code:

```
import Adafruit_BBIO.PWM as PWM
 
servo_pin = "P8_13"
duty_min = 3
duty_max = 14.5
duty_span = duty_max - duty_min
 
#PWM.start(servo_pin, 50, 1000, 1)
#PWM.start(servo_pin, 50)
PWM.start(servo_pin, (100-duty_min), 60.0)
 
while True:
    angle = raw_input("Angle (0 to 180 x to exit):")
    if angle == 'x':
        PWM.stop(servo_pin)
        PWM.cleanup()
        break
    angle_f = float(angle)
    duty = 100 - ((angle_f / 180) * duty_span + duty_min) 
    PWM.set_duty_cycle(servo_pin, duty)
```

The output is
```
Your code is running at http://localhost.
Important: use os.getenv(PORT, 8080) as the port and os.getenv(IP, 0.0.0.0) as the host in your scripts!

Traceback (most recent call last):
  File "/var/lib/cloud9/senior_design/pwm.py", line 10, in <module>
    PWM.start(servo_pin, (100-duty_min), 60.0)
RuntimeError: Problem with a sysfs file

Process exited with code: 0
```

I next tried to test PWM using the BoneScript library with the following code. The first time I run the BoneScript version, it does work. However, if I re-run the code with no change or by changing frequency, duty cycle, or port PWM does not output again, and the running PWM stops. It does not work again until I restart the board. UPDATE - It will output again after I run the pin PWM config given above (and pasted below)

```
var b = require('bonescript');
b.pinMode('P9_14', b.OUTPUT);
b.analogWrite('P9_14', 0.7, 2000.0, printStatus);
function printStatus(x) {
    console.log('x.err = ' + x.err);
}
```

I've been reading through this thread and to help out here's additional stuff I've run.

```
debian@beaglebone:/var/lib/cloud9$ for i in `find /sys/devices/platform/ocp/ -type f |grep pinmux |grep state$|sort -n`; do echo $i: `cat $i`; done
/sys/devices/platform/ocp/ocp:P8_07_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_08_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_09_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_10_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_11_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_12_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_13_pinmux/state: pwm
/sys/devices/platform/ocp/ocp:P8_14_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_15_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_16_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_17_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_18_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_19_pinmux/state: default
/sys/devices/platform/ocp/ocp:P8_26_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_11_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_12_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_13_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_14_pinmux/state: gpio
/sys/devices/platform/ocp/ocp:P9_15_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_16_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_17_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_18_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_21_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_22_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_23_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_24_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_26_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_27_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_30_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_41_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_42_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_91_pinmux/state: default
/sys/devices/platform/ocp/ocp:P9_92_pinmux/state: default
```

Next, I ran the suggested code above. I found that this will restore PWM to the bonescript code, but I still get the same `Problem with a sysfs file` error from the Python code.
```
debian@beaglebone:/var/lib/cloud9$ config-pin -q p9.14
P9_14 Mode: gpio Direction: out Value: 0
debian@beaglebone:/var/lib/cloud9$ config-pin p9.14 pwm
debian@beaglebone:/var/lib/cloud9$ config-pin -q p9.14
P9_14 Mode: pwm
debian@beaglebone:/var/lib/cloud9$
debian@beaglebone:/var/lib/cloud9$ config-pin -q p9.16
P9_16 Mode: default Direction: in Value: 0
debian@beaglebone:/var/lib/cloud9$ config-pin p9.16 pwm
debian@beaglebone:/var/lib/cloud9$ config-pin -q p9.16
P9_16 Mode: pwm
debian@beaglebone:/var/lib/cloud9$
debian@beaglebone:/var/lib/cloud9$ config-pin -q p8.19
P8_19 Mode: default Direction: in Value: 0
debian@beaglebone:/var/lib/cloud9$ config-pin p8.19 pwm
debian@beaglebone:/var/lib/cloud9$ config-pin -q p8.19
P8_19 Mode: pwm
debian@beaglebone:/var/lib/cloud9$
debian@beaglebone:/var/lib/cloud9$ config-pin -q p8.13
P8_13 Mode: pwm
debian@beaglebone:/var/lib/cloud9$ config-pin p8.13 pwm
debian@beaglebone:/var/lib/cloud9$ config-pin -q p8.13
P8_13 Mode: pwm
```

bmcilw1, 2017-10-31:
I've tried the image suggested to me, and it DID fix the BoneScript PWM issue. https://github.com/adafruit/adafruit-beaglebone-io-python/issues/169#issuecomment-340249126

Python PWM still gives the same error

bmcilw1, 2017-10-31:
UPDATE: I finally figured out what was going on. I needed to run the python commands as SUDO or ROOT.

I also simplified the test code:
```
import Adafruit_BBIO.PWM as PWM
import time

pin1 = "P9_14"

PWM.start(pin1, 50, 4000)

time.sleep(5)

PWM.stop(pin1)
PWM.cleanup()
```

Should this require sudo?
#169 
#182

pdp7, 2017-10-31:
I have just merged several fixes ( #183 and #184 ) so please install the latest version of Adafruit_BBIO from this repo:
```
sudo apt-get update
sudo apt-get install build-essential python-dev python-pip -y
git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
sudo python setup.py install
```

pdp7, 2017-10-31:
@bmcilw1 wrote:
> UPDATE: I finally figured out what was going on. I needed to run the python commands as SUDO or ROOT.
Yes, Python needs to be run as root to control the PWM outputs (for kernels older than Linux 4.14).

For Linux 4.14 kernel, the PWM paths in /sys are now correctly setup by udev for non-root access.  @RobertCNelson has added support for this in [c_pwm.c](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/c_pwm.c#L433)

pdp7, 2017-10-31:
@bmcilw1 The Linux 4.14 kernel should resolve the issue of non-root access to the PWM outputs.  Please try installing 4.14 kernel build by @RobertCNelson with:
```
sudo apt-get update
sudo apt-get upgrade
sudo /opt/scripts/tools/update_kernel.sh --lts-4_14
sudo reboot
```

silver2row, 2017-11-01:
Hey...are we using the 4.14.x kernel now? I am trying to stay up-to-date and have each of my "items" working w/ updated ideas and software.

Seth

pdp7, 2017-11-01:
@silver2row It is not the default yet.  However, @RobertCNelson and I have been testing 4.14.x kernel and it seems to be working ok.  If you'd like, please give it a try with the [instructions above](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/185#issuecomment-340870632).

bmcilw1, 2017-11-02:
Nope. @pdp7 I've installed the new kernel, but still sudo is required.

```
debian@beaglebone:/var/lib/cloud9/senior_design$ python pwm.py
Traceback (most recent call last):
  File "pwm.py", line 6, in <module>
    PWM.start(pin1, 30, 4000)
RuntimeError: Problem with a sysfs file
debian@beaglebone:/var/lib/cloud9/senior_design$ sudo python pwm.py
[sudo] password for debian:
debian@beaglebone:/var/lib/cloud9/senior_design$
```

Here's the state of my board

```
debian@beaglebone:/var/lib/cloud9/senior_design$ apt-get -v
apt 1.4.8 (armhf)
Supported modules:
*Ver: Standard .deb
*Pkg:  Debian dpkg interface (Priority 30)
 Pkg:  Debian APT solver interface (Priority -1000)
 Pkg:  Debian APT planner interface (Priority -1000)
 S.L: 'deb' Debian binary tree
 S.L: 'deb-src' Debian source tree
 Idx: Debian Source Index
 Idx: Debian Package Index
 Idx: Debian Translation Index
 Idx: Debian dpkg status file
 Idx: Debian deb file
 Idx: Debian dsc file
 Idx: Debian control file
 Idx: EDSP scenario file
 Idx: EIPP scenario file
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /etc/dogtag
cat /sys/devices/platform/bone_capemgr/slots
cat /boot/uEnv.txt
config-pin -q p9.14
uname -aBeagleBoard.org Debian Image 2017-10-17
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /etc/debian_version
9.2
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/bone_capemgr/slots
cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.0-rc7-ti-r6
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin -q p9.14
P9_14 Mode: pwm
debian@beaglebone:/var/lib/cloud9/senior_design$ uname -a
Linux beaglebone 4.14.0-rc7-ti-r6 #1 SMP Mon Oct 30 17:10:41 UTC 2017 armv7l GNU/Linux
```

RobertCNelson, 2017-11-02:
@bmcilw1 as a quick test just run this ass "debian" user, if you have an led on P9.14 it should blink..

```
debian@bborg-stretch-v8-18:~$ config-pin P9.14 pwm
debian@bborg-stretch-v8-18:~$ cd /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip*/

debian@bborg-stretch-v8-18:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ ls -lha
total 0
drwxrwxr-x 3 root pwm     0 Nov  2 02:09 .
drwxr-xr-x 3 root root    0 Nov  2 02:10 ..
lrwxrwxrwx 1 root pwm     0 Nov  2 02:09 device -> ../../../48302200.pwm
-rw-rw---- 1 root pwm  4.0K Nov  2 02:09 export
-rw-rw-r-- 1 root pwm  4.0K Nov  2 02:09 npwm
drwxrwxr-x 2 root pwm     0 Nov  2 02:09 power
lrwxrwxrwx 1 root pwm     0 Nov  2 02:09 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm  4.0K Nov  2 02:09 uevent
-rw-rw---- 1 root pwm  4.0K Nov  2 02:09 unexport

debian@bborg-stretch-v8-18:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 0 > export

debian@bborg-stretch-v8-18:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ ls -lha
total 0
drwxrwxr-x 4 root pwm     0 Nov  2 02:12 .
drwxr-xr-x 3 root root    0 Nov  2 02:10 ..
lrwxrwxrwx 1 root pwm     0 Nov  2 02:09 device -> ../../../48302200.pwm
-rw-rw---- 1 root pwm  4.0K Nov  2 02:12 export
-rw-rw-r-- 1 root pwm  4.0K Nov  2 02:09 npwm
drwxrwxr-x 2 root pwm     0 Nov  2 02:09 power
drwxrwxr-x 3 root pwm     0 Nov  2 02:12 pwm-3:0
lrwxrwxrwx 1 root pwm     0 Nov  2 02:09 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm  4.0K Nov  2 02:09 uevent
-rw-rw---- 1 root pwm  4.0K Nov  2 02:09 unexport

debian@bborg-stretch-v8-18:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 20000 > pwm-*\:0/period 
debian@bborg-stretch-v8-18:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 10000 > pwm-*\:0/duty_cycle
debian@bborg-stretch-v8-18:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 1 > pwm-*\:0/enable 
```

bmcilw1, 2017-11-02:
Haha @RobertCNelson I do believe you have a typo ^^ :)

RobertCNelson, 2017-11-02:
Just a few. ;)

pdp7, 2017-11-02:
@bmcilw1 I believe there is a race condition with udev setting the permissions correctly.  I'm adding some debugging output to investigate further.

bmcilw1, 2017-11-02:
@RobertCNelson I've lit up an LED, however it does not noticeably blink. Here's my output. Most noticeably was that 0 was busy...

```
debian@beaglebone:/var/lib/cloud9/senior_design$ cd
debian@beaglebone:~$ clear
debian@beaglebone:~$ config-pin P9.14 pwm
debian@beaglebone:~$ cd /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip*/
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ ls -lha
total 0
drwxrwxr-x 4 root pwm     0 Nov  2 01:49 .
drwxr-xr-x 3 root root    0 Nov  2 01:48 ..
lrwxrwxrwx 1 root pwm     0 Nov  2 01:48 device -> ../../../48302200.pwm
-rw-rw---- 1 root pwm  4.0K Nov  2 01:49 export
-rw-rw-r-- 1 root pwm  4.0K Nov  2 01:48 npwm
drwxrwxr-x 2 root pwm     0 Nov  2 01:48 power
drwxrwxr-x 3 root pwm     0 Nov  2 01:49 pwm-3:0
lrwxrwxrwx 1 root pwm     0 Nov  2 01:48 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm  4.0K Nov  2 01:48 uevent
-rw-rw---- 1 root pwm  4.0K Nov  2 01:48 unexport
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 0 > export
-bash: echo: write error: Device or resource busy
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 0 > export
-bash: echo: write error: Device or resource busy
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ ls -lha
total 0
drwxrwxr-x 4 root pwm     0 Nov  2 01:49 .
drwxr-xr-x 3 root root    0 Nov  2 02:53 ..
lrwxrwxrwx 1 root pwm     0 Nov  2 01:48 device -> ../../../48302200.pwm
-rw-rw---- 1 root pwm  4.0K Nov  2 02:53 export
-rw-rw-r-- 1 root pwm  4.0K Nov  2 01:48 npwm
drwxrwxr-x 2 root pwm     0 Nov  2 01:48 power
drwxrwxr-x 3 root pwm     0 Nov  2 01:49 pwm-3:0
lrwxrwxrwx 1 root pwm     0 Nov  2 01:48 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm  4.0K Nov  2 01:48 uevent
-rw-rw---- 1 root pwm  4.0K Nov  2 01:48 unexport
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 20000 > pwm-*\:0/period
-bash: echo: write error: Invalid argument
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$  echo 10000 > pwm-*\:0/duty_cycle
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 1 > pwm-*\:0/enable
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 20000 > pwm-*\:0/period
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 200 > pwm-*\:0/period
-bash: echo: write error: Invalid argument
debian@beaglebone:/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3$ echo 20000 > pwm-*\:0/period

```

bmcilw1, 2017-11-02:
@pdp7 I'm correct in assuming that you can reproduce this issue on your end?

pdp7, 2017-11-02:
Regarding the udev race condition, I had a bash loop logging `ls -a` every 100 ms and observed that the `pwm-6:1` is created and later udev sets the group ownership and permission.

Initial:
`ls: cannot access '/sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1/enable': No such file or directory`
100 ms elapsed:
`-rw-r--r-- 1 root root 4096 Nov  2 03:22 /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1/enable`
100 ms elapsed:
`-rw-rw-r-- 1 root pwm 4096 Nov  2 03:22 /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm-6:1/enable`

pdp7, 2017-11-02:
@bmcilw1 I have pushed commit b65cbf8.  `pwm_start()` will now sleep 100 ms after export to avoid race condition as udev needs the opportunity to set group ownership and permission.

Please build and install the `master` branch from this repo and test again.

pdp7, 2017-11-02:
@bmcilw1 I tested the program from the top of this issue and it does run OK after the fix in b65cbf8

pdp7, 2017-11-03:
@bmcilw1  I've published [release 1.0.6](https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.0.6) which should resolve this issue when using the Debian Stretch image with the 4.14.x kernel.

pdp7, 2017-11-08:
@bmcilw1 have you had a chance to try [release 1.0.6](https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.0.6)?

It should resolve this issue when using the Debian Stretch image with the 4.14.x kernel.

pdp7, 2017-11-08:
@bmcilw1 please re-open if you are still having this issue

closed 2017-11-08 by pdp7