adafruit/adafruit-beaglebone-io-python #186: Problems with UART

bmcilw1, 2017-11-02:
I am unable to set up UART with my board. My code runs with no errors, but no signal shows on the oscilloscope - it's 100% a voltage low. This time, invoking the script with sudo does not help. I've also tried to open a terminal with minicom to view the output for the tty. Here's my test code:

```
import Adafruit_BBIO.UART as UART
import serial
 
UART.setup("UART1")
 
ser = serial.Serial(port = "/dev/ttyO1", baudrate=9600)
ser.close()
ser.open()
if ser.isOpen():
	print "Serial is open!"
ser.write("Hello World!")
ser.close()
```

I've also tried the following experiment.

```
#first terminal window:
minicom -b 9600 -D /dev/ttyO1
 
#second terminal window:
minicom -b 9600 -D /dev/ttyO2

#now type in one terminal and press enter. Expect to view characters in second window.
```

No matter what I do, no data is sent over UART. Any suggestions or ideas? Here's the state of my board.

```
debian@beaglebone:/var/lib/cloud9/senior_design$ apt-get -v
apt 1.4.8 (armhf)
Supported modules:
*Ver: Standard .deb
*Pkg:  Debian dpkg interface (Priority 30)
 Pkg:  Debian APT solver interface (Priority -1000)
 Pkg:  Debian APT planner interface (Priority -1000)
 S.L: 'deb' Debian binary tree
 S.L: 'deb-src' Debian source tree
 Idx: Debian Source Index
 Idx: Debian Package Index
 Idx: Debian Translation Index
 Idx: Debian dpkg status file
 Idx: Debian deb file
 Idx: Debian dsc file
 Idx: Debian control file
 Idx: EDSP scenario file
 Idx: EIPP scenario file
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /etc/dogtag
cat /sys/devices/platform/bone_capemgr/slots
cat /boot/uEnv.txt
config-pin -q p9.14
uname -aBeagleBoard.org Debian Image 2017-10-17
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /etc/debian_version
9.2
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/bone_capemgr/slots
cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.0-rc7-ti-r6
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin -q p9.14
P9_14 Mode: pwm
debian@beaglebone:/var/lib/cloud9/senior_design$ uname -a
Linux beaglebone 4.14.0-rc7-ti-r6 #1 SMP Mon Oct 30 17:10:41 UTC 2017 armv7l GNU/Linux
```

bmcilw1, 2017-11-02:
For reference, I've connected my BBB according the this tutorial which I got the code from: https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/uart

RobertCNelson, 2017-11-02:
for the usarts you need to set the pinmux:

```
config-pin P9.26 uart
config-pin P9.24 uart
```

But now that i know how the pwm mux setup in adafruit-io, it wouldn't be to hard to change that in the library too.

pdp7, 2017-11-02:
@RobertCNelson thanks for the suggestion

@bmcilw1 Please use config-pin as Robert mentioned above to configure P9.24 and P9.26 for UART1

Verify the pin mode is set with:
```
cat /sys/devices/platform/ocp/ocp:P9_24_pinmux/state
cat /sys/devices/platform/ocp/ocp:P9_26_pinmux/state
```

bmcilw1, 2017-11-02:
Interesting, I now see some different behavior. Here's the verification for my uart setup

```
debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin P9.26 uart
debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin P9.24 uart
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/ocp/ocp:P9_24_pinmux/state
uart
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/ocp/ocp:P9_26_pinmux/state
uart
```

Now, I can type in window 1, but it still doesn't show up in window 2.

[image]

To verify that I'm not crazy, here's my pin connections of my board: 
[image]

bmcilw1, 2017-11-02:
And... I figured it out.

1. I also needed to configure pin 21 and 22.
1. I was connected to the wrong pins :) oops

```
debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin P9.26 uart          debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin P9.24 uart          debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin P9.21 uart
debian@beaglebone:/var/lib/cloud9/senior_design$ config-pin P9.22 uart
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/ocp/ocp:P9_24_pinmux/state
uart
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/ocp/ocp:P9_26_pinmux/state
uart
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/ocp/ocp:P9_21_pinmux/state
uart
debian@beaglebone:/var/lib/cloud9/senior_design$ cat /sys/devices/platform/ocp/ocp:P9_22_pinmux/state
uart

```

[image]

Yes!!

bmcilw1, 2017-11-02:
Why do I need to configure these pins? Shouldn't these be the default settings? `config-pin P9.26 uart ` should be uart by default to me because the pinout diagram marks it as such...

pdp7, 2017-11-02:
> I also needed to configure pin 21 and 22.

Ah ok, that makes sense as UART2 is P9.21 and P9.22.

So for the loopback test, this is needed:
```
config-pin P9.21 uart  # UART2_TXD
config-pin P9.22 uart  # UART2_RXD
config-pin P9.24 uart  # UART1_TXD
config-pin P9.26 uart  # UART1_RXD
```

bmcilw1, 2017-11-02:
Correct. Is that expected behavior?

pdp7, 2017-11-02:
@bmcilw1 Yes, config-pin is used on the current image to control pin mode.  I intend to add the ability for Adafruit_BBIO set pin mode with config-pin.  Refer to issue #158

RobertCNelson, 2017-11-02:
Default state has been gpio_pd with input_en for a long time..

bmcilw1, 2017-11-02:
I'm ready to close if no one has any additional stuff here. However, I think that the need to configure pins as well as an example should probably be mentioned in the README since many of the examples won't work without it.

closed 2017-11-02 by bmcilw1

pdp7, 2017-11-02:
@bmcilw1 thanks, I've updated the [README](READM.md) to document the instructions for this issue: b4767a35f60e0766663c89318a17b2125e267ed0

hulkco, 2019-03-30:
hello everyone, I read this thread, and many notes on the Internet about how to enable UART ports in the Beaglebone, without success so far, attach images, use the latest image Stretch LXQT and my badge is the BBGW, I would appreciate any help
[image]
[image]
[image]

pdp7, 2019-03-30:
@hulkco could you please paste the output of this script:
`sudo /opt/scripts/tools/version.sh`
and paste the contents of:
`/boot/uEnv.txt`

pdp7, 2019-03-30:
@hulkco in addition, could you please open up a new issue?  

I find it easier to manage when there is not a long history of older comments.  We can still reference #186 in the new issue.

In the new issue, please include the following:

- paste the output of this script:
  - `sudo /opt/scripts/tools/version.sh`
- paste the contents of:
  - `/boot/uEnv.txt`
- the commands you entered (like `config-pin`)
- the source code for any test programs you are trying to run

thanks!

hulkco, 2019-03-30:
Thast ok @pdp7 i open a new  and thanks for the help!!!