adafruit/adafruit-beaglebone-io-python #189: PWM.set_duty_cycle segmentation fault - but only using python3.6

MrBillium, 2017-11-10:
New clean install using official stretch 2017-10-10 IoT image which includes Python 3.53. Calling set_duty_cycle throws Segment fault if run using Python3.60 or 3.63 venvs but no error using included Python3 or python 3.53 venv
I have narrowed it down to the following simple code.  Any ideas?

Bill

Code  file pwmtest.py

`import Adafruit_BBIO.PWM as PWM

pin = 'P8_13'
PWM.start(pin, 50, 2000, 1)
PWM.set_duty_cycle(pin, 30)`

 results for venv's
venvs/py360/bin/python /tmp/pwmtest.py               -  Segment fault
venvs/py363/bin/python /tmp/pwmtest.py               -  Segment fault
venvs/py353/bin/python /tmp/pwmtest.py               -  No error

pdp7, 2017-11-15:
@MrBillium thanks for opening this issue.  I'll attempt to reproduce and investigate the cause.

pdp7, 2017-11-19:
@MrBillium  which kernel is running when you get the seg fault?

pdp7, 2017-11-19:
@MrBillium who did you install Python 3.6?  I don't see it the repo.  Just Python 3.5.

MrBillium, 2017-11-19:
Python 3.6 was built from source and works fine otherwise - I built as follows
./configure --enable-optimizations
make
make altinstall

Using kernal 4.49 

root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[d36fe9a7be9ebfc872b10a470e904ab4c61c4516]
eeprom:[A335BNLTBBG115081628]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.4.91-ti-r133]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171009.0-0rcnee1~stretch+20171009]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
root@beaglebone:~#

pdp7, 2017-11-19:
@MrBillium Is this the release that I should compile? https://www.python.org/downloads/release/python-363/

MrBillium, 2017-11-19:
Yes 3.63 is the latest – I had the same problem with 3.60.

 

Bill

MrBillium, 2017-11-20:
Note that if you compile Python on a BB it will take 6-7 hours!

 

Bill

pdp7, 2017-11-20:
@RobertCNelson would you be able to package Python 3.6.3 for BeagleBone?

RobertCNelson, 2017-11-20:
@pdp7 so Python 3.6.3 is in Buster:
https://packages.debian.org/buster/python3
Grab this image for testing:

https://rcn-ee.net/rootfs/bb.org/testing/2017-11-19/buster-iot/

Regards,

pdp7, 2017-11-20:
@RobertCNelson thanks

@MrBillium I'm running this Debian Buster image linked above with 4.14.0-ti-r9 kernel.  I get a seg fault when trying to run `PWM.set_duty_cycle()` on Python 2.7.14 and Python 3.6.3.

I'll investigate further to see why that occurs.

fyi - here are the full results:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python ~/issue189.py
Segmentation fault
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3 ~/issue189.py
Segmentation fault
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python --version     
Python 2.7.14
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3 --version
Python 3.6.3
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python ~/issue189-no-set_duty_cycle.py
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3 ~/issue189-no-set_duty_cycle.py
```

pdp7, 2017-11-21:
I have compiled Python 3.6.3 from source and installed in Debian 9.2 (2017-10-17 image). `PWM.set_duty_cycle()` does cause a segmentation fault with Python 3.6.3.
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2017-10-17
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat /etc/debian_version 
9.2
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python --version
Python 2.7.13
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3 --version
Python 3.5.3
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3.6 --version
Python 3.6.3
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python2 ~/issue189.py 
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3 ~/issue189.py 
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3.6 ~/issue189.py 
Segmentation fault
```

pdp7, 2017-11-21:
I'm investigating if a change in Python 3.6 versus 3.5 triggered this seg fault issue for the call to `PyArg_ParseTupleAndKeywords()` in `py_set_duty_cycle()`.  

This could be relevant:
**[What’s New In Python 3.6: Build and C API Changes](https://docs.python.org/3/whatsnew/3.6.html)**
> PyArg_ParseTupleAndKeywords() now supports positional-only parameters. Positional-only parameters are defined by empty names. (Contributed by Serhiy Storchaka in bpo-26282).

pdp7, 2017-11-24:
I've not been able to determine thus far what is causing the issue for Python 3.6.  Here are the functions that call `PyArg_ParseTupleAndKeywords()`.  I'm going to investigate them further:

### source/py_gpio.c
```
// python function setup(channel, direction, pull_up_down=PUD_OFF, initial=None, delay=0)
static PyObject *py_setup_channel(__attribute__ ((unused)) PyObject *self, PyObject *args, PyObject *kwargs)
   if (!PyArg_ParseTupleAndKeywords(args, kwargs, "si|iii", kwlist, &channel, &direction, &pud, &initial, &delay))
```

```
// python function add_event_callback(gpio, callback, bouncetime=0)
static PyObject *py_add_event_callback(__attribute__ ((unused)) PyObject *self, PyObject *args, PyObject *kwargs)
   if (!PyArg_ParseTupleAndKeywords(args, kwargs, "sO|i", kwlist, &channel, &cb_func, &bouncetime))
```

```
// python function add_event_detect(gpio, edge, callback=None, bouncetime=0
static PyObject *py_add_event_detect(__attribute__ ((unused)) PyObject *self, PyObject *args, PyObject *kwargs)
   if (!PyArg_ParseTupleAndKeywords(args, kwargs, "si|Oi", kwlist, &channel, &edge, &cb_func, &bouncetime))
```

### source/py_pwm.c
```
// python function start(channel, duty_cycle, freq)
static PyObject *py_start_channel(__attribute__ ((unused)) PyObject *self, PyObject *args, PyObject *kwargs)
    if (!PyArg_ParseTupleAndKeywords(args, kwargs, "s|ffi", kwlist, &channel, &duty_cycle, &frequency, &polarity)) {
```

```
// python method PWM.set_duty_cycle(channel, duty_cycle)
static PyObject *py_set_duty_cycle(__attribute__ ((unused)) PyObject *self, PyObject *args, PyObject *kwargs)
    if (!PyArg_ParseTupleAndKeywords(args, kwargs, "s|f", kwlist, &channel, &duty_cycle))
```

```
// python method PWM.set_frequency(channel, frequency)
static PyObject *py_set_frequency(__attribute__ ((unused)) PyObject *self, PyObject *args, PyObject *kwargs)
    if (!PyArg_ParseTupleAndKeywords(args, kwargs, "s|f", kwlist, &channel, &frequency))
```

pdp7, 2017-11-24:
`PWM.set_frequency()` also causes a seg fault with Python 3.6:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat ~/pwm-test.py
import Adafruit_BBIO.PWM as PWM 
PWM.start("P9_14", 50)
PWM.set_frequency("P9_14", 10)

debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo python3.6 ~/pwm-test.py
Segmentation fault
```

pdp7, 2017-11-25:
It does not appear that `PyArg_ParseTupleAndKeywords()` is a problem for functions in `source/py_gpio.c` with Python 3.6: 
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3.6 ~/edge.py
py_setup_channel
py_add_event_detect
py_add_event_callback
py_add_event_callback
```
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat ~/edge.py
import Adafruit_BBIO.GPIO as GPIO

def hello(channel):  
    print("Hello")

def goodbye(channel):  
    print("Goodbye")

GPIO.setup("P8_14", GPIO.IN)
GPIO.add_event_detect("P8_14", GPIO.RISING)
GPIO.add_event_callback("P8_14", hello)
GPIO.add_event_callback("P8_14", goodbye)
```
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3.6 ~/issue42.py 
py_setup_channel
py_add_event_detect
```
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ cat ~/issue42.py 
import Adafruit_BBIO.GPIO as GPIO

counterX = 0

def updateEncoderX(channel):
    global counterX
    counterX = counterX + 1
    print("Encoder counter updated: %d") % counterX

GPIO.setup("P8_14", GPIO.IN)
GPIO.add_event_detect("P8_14", GPIO.RISING, callback=updateEncoderX)

try:
    while 1:
        pass

except KeyboardInterrupt:
    GPIO.cleanup()
GPIO.cleanup()
```

pdp7, 2017-11-25:
I recompiled Python 3.6 with Py_DEBUG.

gdb is now able to produce a backtrace with symbols:
```
(gdb) run ~/test-pwm-duty.py
Starting program: /usr/local/bin/python3.6 ~/test-pwm-duty.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
call PWM.start()
py_start_channel
call PWM.set_duty_cycle()
enter py_set_duty_cycle
call PyArg_ParseTupleAndKeywords

Program received signal SIGSEGV, Segmentation fault.
0x005233a0 in PyArg_ParseTupleAndKeywords (args=args@entry=0xb6b3b878, keywords=keywords@entry=0xb6b3b88c, 
    format=0xb6af59b4 "s|f", kwlist=0xb6b081f0 <kwlist>) at Python/getargs.c:1354
1354	        (keywords != NULL && !PyDict_Check(keywords)) ||
(gdb) 
```

pdp7, 2017-11-25:
### Python-3.6.3/Python/getargs.c
```
1343 /* Return false (0) for error, else true. */
1344 int
1345 PyArg_ParseTupleAndKeywords(PyObject *args,
1346                             PyObject *keywords,
1347                             const char *format,
1348                             char **kwlist, ...)
1349 {
1350     int retval;
1351     va_list va;
1352 
1353     if ((args == NULL || !PyTuple_Check(args)) ||
1354         (keywords != NULL && !PyDict_Check(keywords)) ||
1355         format == NULL ||
1356         kwlist == NULL)
1357     {
1358         PyErr_BadInternalCall();
1359         return 0;
1360     }
1361 
1362     va_start(va, kwlist);
1363     retval = vgetargskeywords(args, keywords, format, kwlist, &va, 0);
1364     va_end(va);
1365     return retval;
1366 }
```

pdp7, 2017-11-25:
The solution turned out to be speciftying `METH_VARARGS | METH_KEYWORDS` instead of just `METH_VARARGS` for `set_duty_cycle` and `set_frequency`:
```diff
-    { "set_duty_cycle", (PyCFunction)py_set_duty_cycle, METH_VARARGS, "Change the duty cycle\ndutycycle - between 0.0 and 100.0" },
+    { "set_duty_cycle", (PyCFunction)py_set_duty_cycle, METH_VARARGS | METH_KEYWORDS, "Change the duty cycle\ndutycycle - between 0.0 and 100.0" },
```
```diff
-    { "set_frequency", (PyCFunction)py_set_frequency, METH_VARARGS, "Change the frequency\nfrequency - frequency in Hz (freq > 0.0)" },
+    { "set_frequency", (PyCFunction)py_set_frequency, METH_VARARGS | METH_KEYWORDS, "Change the frequency\nfrequency - frequency in Hz (freq > 0.0)" },
```

closed 2017-11-25 by pdp7