adafruit/adafruit-beaglebone-io-python #190: GPIO.output still extremely slow in V1.07 on Stretech

MrBillium, 2017-11-12:
I am still struggling to get this library  working well on Stretch.
 With a brand new load of latest Stretch  IOT image  ( 11-4-17) and no changes except your prerequisites and installation, I am getting a 900mS run time for setting a pin 100 times.  The same code take 3mS running on Jessie. This problem is independent of Python version and I have tried both BBB and BBG.

Bill

`import Adafruit_BBIO.GPIO as GPIO
import timeit

pin = 'P9_14'

GPIO.setup(pin, GPIO.OUT)

def setoutput():
    for x in range(1, 10):
        GPIO.output(pin, True)
        GPIO.output(pin, False)

# run 5 times, 10 x 2 x 5 = 100 outputs
result = timeit.Timer(setoutput).timeit(5)
print('time to complete 100 sets = ', result)
GPIO.cleanup()`

pdp7, 2017-11-15:
@MrBillium thanks for the feedback.  I'll look into this issue.

pdp7, 2017-11-17:
@MrBillium please paste the output of:
```
sudo /opt/scripts/tools/version.sh 
```

Here is the Python test I've run:
```
import Adafruit_BBIO.GPIO as GPIO
import timeit

pin = 'P9_14'

GPIO.setup(pin, GPIO.OUT)

def setoutput():
  for x in range(1, 10):
    GPIO.output(pin, True)
    GPIO.output(pin, False)
    #run 5 times, 10 x 2 x 5 = 100 outputs

result = timeit.Timer(setoutput).timeit(5)
print('time to complete 100 sets = ', result)
GPIO.cleanup()
```

Here is the output for Debian 9.2 with `4.14.0-rc7-ti-r6` kernel:
```
debian@beaglebone:~$ sudo python time-test.py
('time to complete 100 sets = ', 1.4771080017089844)
```

output of `sudo /opt/scripts/tools/version.sh`:
```
git:/opt/scripts/:[6c2688b0be448b7bb9ca18849b430d496a84acb4]
eeprom:[A335BNLT000C3014BBBK1316]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-rc2-00002-g84a7f2]
kernel:[4.14.0-rc7-ti-r6]
nodejs:[v6.11.5]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171027.1-0rcnee1~stretch+20171027]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

pdp7, 2017-11-17:
> With a brand new load of latest Stretch IOT image ( 11-4-17) and no changes except your prerequisites and installation, I am getting a 900mS run time for setting a pin 100 times. The same code take 3mS running on Jessie

@MrBillium:
What kernel is running on the Jessie image that you are using?   
Are you running the same version of Adafruit_BBIO on both Jessie and Stretch?

pdp7, 2017-11-17:
One possible inefficiency is this check for board type inside `gpio_set_value()`:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/event_gpio.c#L405

The function `beaglebone_blue()` executes `popen()` to run grep: 
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L602

This should be cached so that it only needs to be done once when Adafruit_BBIO is initialized.

MrBillium, 2017-11-17:
Just to be clear, when I say BBB I mean BB Black not Blue. Also for both test cases the setting of the pin does actually work in that I see the output switching.

First a BBB with brand new 11-4 image and no changes except Adafruit install and requirements:  Runs in 1.3 seconds.
root@beaglebone:/home# pip3 show adafruit_bbio
Name: Adafruit-BBIO
Version: 1.0.7
Summary: A module to control BeagleBone IO channels
Home-page: https://github.com/adafruit/adafruit-beaglebone-io-python/
Author: Justin Cooper
Author-email: justin@adafruit.com
License: MIT
Location: /usr/local/lib/python3.5/dist-packages
Requires:

root@beaglebone:/home# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[a5d5f437baeaa7f8e14d5ae037268abf7a67e88e]
eeprom:[A335BNLT00C05114BBBK195C]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-11-05]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.4.91-ti-r136]
nodejs:[v6.11.5]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171101.0-0rcnee1~stretch+20171101]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

root@beaglebone:/home# sudo python3 timetest.py
time to complete 100 sets =  1.387603264000063
root@beaglebone:/home#

Next a BBG with my production  load of Jessie with overlay changes . BBIO library installed in Python 3.6 Venv. Runs in 0.05 seconds!

(py360) root@beaglebone:/home# pip3 show adafruit_bbio
Name: Adafruit-BBIO
Version: 1.0.3
Summary: A module to control BeagleBone IO channels
Home-page: https://github.com/adafruit/adafruit-beaglebone-io-python/
Author: Justin Cooper
Author-email: justin@adafruit.com
License: MIT
Location: /home/venvs/py360/lib/python3.6/site-packages
Requires:

(py360) root@beaglebone:/home# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[e869b11af4400bfba2a333f00f0f06e1d145d1d2]
eeprom:[A335BNLTBBG1BBG217014819]
dogtag:[BeagleBoard.org Debian Image 2017-03-26]
bootloader:[/dev/mmcblk1]:[U-Boot 2017.03-00002-gd12b1519b4]
kernel:[4.4.55-ti-r94]
nodejs:[v4.8.1]
(py360) root@beaglebone:/home# python3 timetest.py
time to complete 100 sets =  0.005994903000100749
(py360) root@beaglebone:/home#

I have several BB's on my bench so can run any tests you think may help.

Bill

pdp7, 2017-11-18:
@MrBillium It appears that Adafruit_BBIO version 1.0.3 is on your Jessie setup.  Could you try the latest Adafruit_BBIO version (v1.0.7) on Jessie and see if the performance regression occurs?

pdp7, 2017-11-19:
Timing test shows large performance improvement **after** commit 32814797ffeafccdc07491243deca5760759ab54
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3 ~/time-test.py 
time to complete 100 sets =  0.006628266011830419

debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python ~/time-test.py 
('time to complete 100 sets = ', 0.0067021846771240234)
```
Results **before** commit 32814797ffeafccdc07491243deca5760759ab54 with performance regression:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python ~/time-test.py 
('time to complete 100 sets = ', 1.5183091163635254)

debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ python3 ~/time-test.py 
time to complete 100 sets =  1.3909540939785074
```

### time-test.py
```
import Adafruit_BBIO.GPIO as GPIO
import timeit

pin = 'P9_14'

GPIO.setup(pin, GPIO.OUT)

def setoutput():
  for x in range(1, 10):
    GPIO.output(pin, True)
    GPIO.output(pin, False)
    #run 5 times, 10 x 2 x 5 = 100 outputs

result = timeit.Timer(setoutput).timeit(5)
print('time to complete 100 sets = ', result)
GPIO.cleanup()
```

MrBillium, 2017-11-19:
On Stretch image, rolling BBIO back  to 1.0.3 fixes the issue
1.0.7 time to complete = 1.37
1.0.3 time to complete = 0.006

Bill

pdp7, 2017-11-19:
@MrBillium were you able to try with the latest `master`?  I think commit 32814797ffeafccdc07491243deca5760759ab54 should help.

pdp7, 2017-11-19:
@MrBillium please try version 1.0.8:
https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.0.8

MrBillium, 2017-11-19:
Yes  it works!

 

I am getting 0.006S run time using 1.0.8 and my LCD  which uses Adafruit_charlcd is now writing whole screen at a time instead of one character at a time – which is why I went down this rabbit hole.

 

Thanks again

 

Bill

pdp7, 2017-11-20:
Good to hear!  Thanks  for pointing out this performance regression so that it could be fixed.

closed 2017-11-20 by pdp7