adafruit/adafruit-beaglebone-io-python #191: Turn off RotaryEncoder's debug output by default

dplanella, 2017-11-14:
I'm using a rotary encoder based on the KY-040 module, with the `Adafruit_BBIO.Encoder.RotaryEncoder` class. I'm running the latest **IoT Debian image** on a **Beaglebone Black**:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.2 (stretch)
Release:	9.2
Codename:	stretch
$ uname -r
4.4.91-ti-r133
$ python --version
Python 2.7.13
```

To enable the QEP module on the hardware, I added these lines to `/boot/uEnv.txt` as instructed in the `Encoder` module source code and rebooted:

```
cmdline=coherent_pool=1M quiet cape_universal=enable
cape_enable=bone_capemgr.enable_partno=cape-universala
```

I then configured the `equep2` pins on the command line:

```
$ config-pin P8.11 qep 
$ config-pin P8.12 qep 
```

And with these inputs connected to the encoder, I ran this simple code via an SSH connection: 

```
import time
import Adafruit_BBIO.Encoder as Encoder

encoder = Encoder.RotaryEncoder(Encoder.RotaryEncoder.EQEP2)

encoder.setAbsolute()
encoder.zero()

while True:
    print("Encoder position: " + encoder.getPosition())
    time.sleep(1)
```

- **Expected**: the position is printed on screen every second
- **Actual**: the position is printed on screen every second (yay!), but the output is obscured by a wealth of debug messages.

In summary, I would expect debug info not to be printed on a production library, unless explicitly enabled. Or in other words, can the debug output be turned off by default?

Thanks!

pdp7, 2017-11-15:
@dplanella thanks for the feedback, I'll eliminate that debug logging

dplanella, 2017-11-17:
@pdp7 no worries, I went ahead and submitted pull #195 to fix this. I'd appreciate a review and hopefully we can then close the issue.

pdp7, 2017-11-17:
@dplanella thanks!

pdp7, 2017-11-17:
I've merged #195

closed 2017-11-17 by pdp7

dplanella, 2017-11-17:
Excellent, thanks!